YoyPa/isw #1: Want support for MSI GF63 RD..

Xieyt, 2019-02-03:
hello,I need help to make this script work in my gf 63 ? can u plzz guide me ?

YoyPa, 2019-03-21:
Sorry for late response, I did not checked the web page for a long time and did not received a mail notification (It's github fault :innocent: ).
I can help but I need to know what is blocking you and then the output of ```./isw -c```.
I'm going to rewrite isw to support multiple laptop (a directory containing each known model config file), it currently need to be adapted for each machine.

Cu3blu, 2019-04-21:
Hello YoyPa!
Thank you a lot for your project! It works nice with GF63 8RD. Config for GL63_8RD fits.
Best wishes,
Alex

YoyPa, 2019-04-23:
Can you post output of ```isw -c``` anyway to complete the pokedex ? :innocent:

Cu3blu, 2019-04-24:
Hello YoyPa!
Sure!

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 08 0b  >................<
000030 03 09 00 0d 50 0a 05 00 94 11 88 2c c2 01 e0 00  >....P......,....<
000040 00 00 5d 00 b6 11 00 00 65 10 68 31 b8 0b fa 32  >..].....e.h1...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 1f 00 28 32 3c 41 46 4b  >..........(2<AFK<
000070 64 1e 1e 2d 3c 50 64 7d 96 00 08 03 03 03 03 03  >d..-<Pd}........<
000080 00 00 37 3d 43 49 4f 55 63 23 23 26 2a 2d 32 37  >..7=CIOUc##&*-27<
000090 3e 00 08 03 03 03 03 03 02 00 64 02 00 6e 00 00  >>.........d..n..<
0000a0 31 36 52 31 45 4d 53 31 2e 31 30 35 31 30 32 33  >16R1EMS1.1051023<
0000b0 32 30 31 38 31 36 3a 33 30 3a 32 37 b6 11 00 40  >201816:30:27...@<
0000c0 00 07 25 00 00 00 00 00 00 36 00 00 00 e4 14 00  >..%......6......<
0000d0 00 00 00 00 7f 00 d1 00 00 00 00 81 01 00 00 00  >................<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 80 00 0c 0f 05 00 00 00 00 00 00 00 00 00  >................<
000100
```

Thanks again!

YoyPa, 2019-04-24:
Added with https://github.com/YoyPa/isw/commit/5310e714845a83b3e1c92db9b02d96fa40ec0d6b

YoyPa, 2019-07-14:
Hello @Cu3blu, I'm checking profile directly from EC update file (msi website) and there is a mismatch between cpu values of your dump and cpu values of update file:

you:
```
# CPU
cpu_temp_0 = 40
cpu_temp_1 = 50
cpu_temp_2 = 60
cpu_temp_3 = 65
cpu_temp_4 = 70
cpu_temp_5 = 75
cpu_fan_speed_0 = 30
cpu_fan_speed_1 = 45
cpu_fan_speed_2 = 60
cpu_fan_speed_3 = 80
cpu_fan_speed_4 = 100
cpu_fan_speed_5 = 125
cpu_fan_speed_6 = 150
```
update file:
```
-----------CPU---------------
Value       set @ address
0x37(55°C)  0xf801(byte63489)
0x40(64°C)  0xf802(byte63490)
0x49(73°C)  0xf803(byte63491)
0x4c(76°C)  0xf804(byte63492)
0x52(82°C)  0xf805(byte63493)
0x58(88°C)  0xf806(byte63494)

0x23(35%)   0xf80b(byte63499)
0x26(38%)   0xf80c(byte63500)
0x2a(42%)   0xf80d(byte63501)
0x2d(45%)   0xf80e(byte63502)
0x32(50%)   0xf80f(byte63503) 
0x37(55%)   0xf810(byte63504)
0x3e(62%)   0xf811(byte63505)
```
You made the dump before or after altering EC with isw ? :thinking:

Cu3blu, 2019-07-19:
Hello YoyPa! I made it after altering as Iremember. But I modified the file before altering EC... Sorry for messing you up :-(

YoyPa, 2019-08-12:
Ok, thanks for confirmation :).

closed 2019-08-12 by YoyPa