YoyPa/isw #2: Support for MSI GS65

EKami, 2019-03-27:
Hello,
Here is the EC dump for the MSI GS65:
```
$ sudo isw -c

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 5b 0b  >..............[.<
000030 03 09 00 0d 50 0a 05 00 a0 14 60 3b 71 02 e0 00  >....P.....`;q...<
000040 00 00 64 00 1e 15 00 00 1e 15 a6 43 e5 0b 00 00  >..d........C....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2b 00 37 40 46 4c 52 58  >........+.7@FLRX<
000070 64 2d 2d 32 3c 46 4b 50 50 00 08 03 03 03 03 03  >d--2<FKPP.......<
000080 28 00 37 3d 41 47 4d 56 5e 00 00 32 3c 46 50 5a  >(.7=AGMV^..2<FPZ<
000090 5a 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >Z.........d..n..<
0000a0 31 36 51 34 45 4d 53 31 2e 31 30 36 30 31 32 33  >16Q4EMS1.1060123<
0000b0 32 30 31 39 31 31 3a 31 33 3a 31 39 1e 15 00 00  >201911:13:19....<
0000c0 00 01 24 00 00 9e c1 00 00 82 00 00 00 b1 14 00  >..$.............<
0000d0 00 00 00 00 70 80 00 00 00 00 00 00 00 00 00 00  >....p...........<
0000e0 e2 02 00 03 10 00 00 00 00 00 00 00 00 9a 00 e4  >................<
0000f0 00 00 80 00 0c 8f 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100
```

Could you add the support for this machine? Thank you.

YoyPa, 2019-03-27:
Seems it use same addresses than GS40, I added GS65_DEFAULT profile: https://github.com/YoyPa/isw/commit/c228d92537ae514e9dc52024f17b0dffba2c4b3b
Please confirm that isw -r GS65_DEFAULT correctly read temps and fan speeds.
isw -w should also work but I don't know if you want to play with it. :innocent:

EKami, 2019-03-27:
@YoyPa Thanks! Looks like it's reading the temps and the fans correctly :)

```
$ sudo isw -r GS65_DEFAULT
CPU
└─temp     (°C): 47
└─fan speed (%): 45
GPU
└─temp     (°C): 46
└─fan speed (%): 0
```

YoyPa, 2019-03-27:
Nice :), one question I was asking myself, does your gpu was turned on when you dumped EC and with your ```isw -r``` ?

EKami, 2019-03-27:
> Nice :), one question I was asking myself, does your gpu was turned on when you dumped EC and with your isw -r ?

Probably yeah, I think it's always on

YoyPa, 2019-03-27:
That explain why gpu temp is accessible.

reikdas, 2019-03-29:
Hi, 
I noticed that the EC dump for my MSI GS65 8rf varies slightly.
```
[reik@reik-msi ~]$ sudo isw -c

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 02 00 b0 e0 06 49 0b  >..............I.<
000030 02 05 00 0d 30 0a 05 00 a0 14 60 3b 71 02 c0 00  >....0.....`;q...<
000040 50 0a 30 00 2f 13 52 f7 18 09 96 39 1a 0c f8 43  >P.0./.R....9...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 31 00 37 40 46 4c 52 58  >........1.7@FLRX<
000070 64 32 1e 32 3c 4b 4b 4b 4b 00 08 03 03 03 03 03  >d2.2<KKKK.......<
000080 34 00 37 3d 41 47 4d 57 61 00 00 32 37 3c 41 44  >4.7=AGMWa..27<AD<
000090 4e 00 08 03 03 03 03 03 02 00 64 02 00 6e 00 00  >N.........d..n..<
0000a0 31 36 51 32 45 4d 53 31 2e 31 30 35 30 35 30 32  >16Q2EMS1.1050502<
0000b0 32 30 31 38 30 39 3a 35 35 3a 35 33 2f 13 00 00  >201809:55:53/...<
0000c0 00 07 23 00 00 9e 40 00 00 01 00 00 00 e2 14 00  >..#...@.........<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 01 00 00 00  >....p...........<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 80 00 0c 8f 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100
```
It works though - 
```
[reik@reik-msi ~]$ sudo isw -r GS65_DEFAULT
CPU
└─temp     (°C): 49
└─fan speed (%): 50
GPU
└─temp     (°C): 50
└─fan speed (%): 50
```

I am not sure if the different EC dump is cause for worry, but I decided to post it just in case.

YoyPa, 2019-03-30:
So I checked differences, it cause no trouble for now but it will if I support changing fan speed in the future.
Details:
```
000060 00 00 00 00 00 00 00 00 2b 00 37 40 46 4c 52 58
000060 00 00 00 00 00 00 00 00 31 00 37 40 46 4c 52 58
# CPU temp: Only realtime change, normal.

000070 64 2d 2d 32 3c 46 4b 50 50 00 08 03 03 03 03 03
000070 64 32 1e 32 3c 4b 4b 4b 4b 00 08 03 03 03 03 03
# CPU fan speed: I will have to split GS65 if I support changing fan speed
# fs0, fs3, fs5, fs6 change

000080 28 00 37 3d 41 47 4d 56 5e 00 00 32 3c 46 50 5a
000080 34 00 37 3d 41 47 4d 57 61 00 00 32 37 3c 41 44

000090 5a
000090 4e
# GPU temp: I will have to split GS65 if I support GPU temp
# t5, t6 change
# GPU fan speed: I will have to split GS65 if I support GPU fan speed
# fs2, fs3, fs4, fs5, fs6 change
```
PS: please open new issue in the future, I can easily miss comment on closed issue.

A little poll to @reikdas  and @EKami:
should I use commercial denomination: 8SG vs 8RF
or BIOS related info (from EC): 16Q4EMS1 vs 16Q2EMS1
to differentiate profiles.

YoyPa, 2019-03-30:
I now support changing fan speed so I made two GS65 config:
GS65_8SG_DEFAULT
GS65_8RF_DEFAULT

closed 2019-03-30 by YoyPa