YoyPa/isw #3: Possibility for fan speed beyond 100%? [question]

ZZZXXX0110, 2019-03-28:
Thank you for creating this great tool!

With the method of writing to EC, would it be possible to achieve fan speed beyond 100%, like you can do in MSI's Dragon Center with the "Cooler Boost" setting?

Thanks!

ZZZXXX0110, 2019-03-28:
In case if it helps, below is the EC dump when "Cooler Boost" mode is activated with Dragon Center, extracted with RW-Everything:
```
Type:Embedded Controller: 0066,0062
Width:01
00=00 01=80 02=00 03=00 04=00 05=00 06=00 07=00 
08=00 09=00 0A=00 0B=00 0C=00 0D=00 0E=00 0F=00 
10=00 11=00 12=00 13=00 14=00 15=00 16=00 17=00 
18=00 19=00 1A=00 1B=00 1C=00 1D=00 1E=00 1F=00 
20=00 21=00 22=00 23=00 24=00 25=00 26=00 27=00 
28=00 29=00 2A=00 2B=B0 2C=E0 2D=07 2E=09 2F=0B 
30=02 31=15 32=01 33=0D 34=50 35=C3 36=05 37=00 
38=A0 39=14 3A=60 3B=3B 3C=71 3D=02 3E=C0 3F=00 
40=50 41=0A 42=39 43=00 44=3B 45=14 46=A9 47=FB 
48=73 49=0B 4A=3F 4B=3B 4C=BE 4D=0B 4E=F8 4F=43 
50=00 51=00 52=00 53=00 54=00 55=00 56=00 57=00 
58=00 59=00 5A=00 5B=00 5C=00 5D=00 5E=00 5F=00 
60=00 61=00 62=00 63=00 64=00 65=00 66=00 67=00 
68=29 69=00 6A=37 6B=40 6C=46 6D=4C 6E=52 6F=58 
70=64 71=00 72=00 73=32 74=3D 75=54 76=5C 77=64 
78=4B 79=00 7A=08 7B=03 7C=03 7D=03 7E=03 7F=03 
80=00 81=00 82=37 83=3D 84=41 85=47 86=4D 87=57 
88=61 89=00 8A=00 8B=32 8C=37 8D=4B 8E=56 8F=64 
90=4E 91=00 92=08 93=03 94=03 95=03 96=03 97=03 
98=82 99=00 9A=64 9B=02 9C=00 9D=6E 9E=00 9F=00 
A0=31 A1=36 A2=51 A3=32 A4=45 A5=4D A6=53 A7=31 
A8=2E A9=31 AA=30 AB=35 AC=30 AD=35 AE=30 AF=32 
B0=32 B1=30 B2=31 B3=38 B4=30 B5=39 B6=3A B7=35 
B8=35 B9=3A BA=35 BB=33 BC=3B BD=14 BE=40 BF=00 
C0=00 C1=01 C2=24 C3=00 C4=00 C5=9E C6=40 C7=00 
C8=00 C9=00 CA=00 CB=49 CC=00 CD=53 CE=14 CF=00 
D0=00 D1=00 D2=00 D3=00 D4=70 D5=00 D6=00 D7=00 
D8=00 D9=00 DA=00 DB=80 DC=81 DD=00 DE=00 DF=00 
E0=E2 E1=02 E2=00 E3=00 E4=10 E5=00 E6=00 E7=00 
E8=00 E9=00 EA=00 EB=00 EC=00 ED=81 EE=00 EF=00 
F0=00 F1=00 F2=C3 F3=00 F4=8C F5=8F F6=05 F7=00 
F8=00 F9=52 FA=32 FB=00 FC=00 FD=00 FE=00 FF=00 
```

And below is the EC dump when fan speed is at 0%:
```
Type:Embedded Controller: 0066,0062
Width:01
00=00 01=80 02=00 03=00 04=00 05=00 06=00 07=00 
08=00 09=00 0A=00 0B=00 0C=00 0D=00 0E=00 0F=00 
10=00 11=00 12=00 13=00 14=00 15=00 16=00 17=00 
18=00 19=00 1A=00 1B=00 1C=00 1D=00 1E=00 1F=00 
20=00 21=00 22=00 23=00 24=00 25=00 26=00 27=00 
28=00 29=00 2A=00 2B=B0 2C=E0 2D=07 2E=49 2F=0B 
30=02 31=15 32=00 33=05 34=50 35=C3 36=05 37=00 
38=A0 39=14 3A=60 3B=3B 3C=71 3D=02 3E=C0 3F=00 
40=50 41=0A 42=3A 43=00 44=3B 45=14 46=97 47=F9 
48=9C 49=0B 4A=85 4B=3A 4C=BE 4D=0B 4E=F8 4F=43 
50=00 51=00 52=00 53=00 54=00 55=00 56=00 57=00 
58=00 59=00 5A=00 5B=00 5C=00 5D=00 5E=00 5F=00 
60=00 61=00 62=00 63=00 64=00 65=00 66=00 67=00 
68=2E 69=00 6A=37 6B=40 6C=46 6D=4C 6E=52 6F=58 
70=64 71=00 72=00 73=32 74=3C 75=4B 76=4B 77=4B 
78=4B 79=00 7A=08 7B=03 7C=03 7D=03 7E=03 7F=03 
80=28 81=00 82=37 83=3D 84=41 85=47 86=4D 87=57 
88=61 89=00 8A=00 8B=32 8C=37 8D=3C 8E=41 8F=44 
90=4E 91=00 92=08 93=03 94=03 95=03 96=03 97=03 
98=02 99=00 9A=64 9B=02 9C=00 9D=6E 9E=00 9F=00 
A0=31 A1=36 A2=51 A3=32 A4=45 A5=4D A6=53 A7=31 
A8=2E A9=31 AA=30 AB=35 AC=30 AD=35 AE=30 AF=32 
B0=32 B1=30 B2=31 B3=38 B4=30 B5=39 B6=3A B7=35 
B8=35 B9=3A BA=35 BB=33 BC=3B BD=14 BE=40 BF=00 
C0=00 C1=01 C2=24 C3=00 C4=00 C5=9E C6=40 C7=00 
C8=00 C9=00 CA=00 CB=00 CC=00 CD=00 CE=14 CF=00 
D0=00 D1=00 D2=00 D3=00 D4=70 D5=00 D6=00 D7=00 
D8=00 D9=00 DA=00 DB=80 DC=81 DD=00 DE=00 DF=00 
E0=E2 E1=02 E2=00 E3=00 E4=10 E5=00 E6=00 E7=00 
E8=00 E9=00 EA=00 EB=00 EC=00 ED=D4 EE=00 EF=00 
F0=00 F1=00 F2=C1 F3=00 F4=8C F5=8F F6=05 F7=00 
F8=00 F9=00 FA=32 FB=00 FC=00 FD=00 FE=00 FF=00 
```

ZZZXXX0110, 2019-03-29:
It's probably easier to see in these screenshots:
In "Cooler Boost" mode:
[image]

Fans at 0%:
[image]

YoyPa, 2019-03-30:
This is a hard one, are you sure you did not inverted screen cap ?
Cause in the second one I see GPU fan profile at 00 90 90 90 90 90 90 (from address 8A to 90)
0x90= 144%.
Also if you look at the text part of the second capture, the bios version and time are messed up (16Q2EMS1[...]), I don't know what happened but it look ugly :stuck_out_tongue_closed_eyes: .

YoyPa, 2019-03-30:
You can probe how high your cpu fan can go by altering value at address 0x72 IF cpu temp is below 0x37(55°C)

YoyPa, 2019-03-30:
I tested 0x90 as a fan speed value, and I disabled it when I seen RPM going over 6000RPM, so it seems possible to go super high (max is 0xff(255%)) but I will not test, too risky to my taste :see_no_evil:.

edward-p, 2019-05-16:
For me, `addr:0x98` controls the `Cooler Boost` behavior:

|value|cooler boost|
|------|--------------|
|0x02|off|
|0x82|on|

DadouLapous, 2019-05-30:
For me (GL63-8RD), the same address ```0x98``` controls the ```Cooler Boost``` behavior, but with different values: ```0x86``` and ```0x06```
```
sudo isw -s 0x98 134

Value       set @ address
0x86(134)   0x98(byte152)
```
enables ```Cooler Boost```
and 
```
 sudo isw -s 0x98 6  

Value       set @ address
0x6(6)      0x98(byte152)
```
disables it.

edward-p, 2019-05-30:
Different model may have different maximum fan speed. My Dragon Center, allow me to set fan speed up to 150%.

YoyPa, 2019-06-13:
I tested cooler boost and it seems to trigger from any value >=128 (GS40-6QE), can you test ? @edward-p @DadouLapous

edward-p, 2019-06-13:
> I tested cooler boost and it seems to trigger from any value >=128 (GS40-6QE), can you test ? @edward-p @DadouLapous

Yes, you're right. I just tested. Any value >=128 can trigger the cooler boost.

DadouLapous, 2019-06-13:
Same here, 
> Any value >=128 can trigger the cooler boost

YoyPa, 2019-06-14:
I added option -b for coolerboost, https://github.com/YoyPa/isw/commit/0d57eec958f9780e76b428be6477a046cb05c835.
It will use 128 for ON (can be changed in isw.conf) and 0 for OFF (not configurable)

closed 2019-06-24 by YoyPa