YoyPa/isw #4: Support for changing fan speed?

ZZZXXX0110, 2019-03-28:
Thank you for making this amazing tool, again!

Any chance changing fan speed could be supported? I notice you mentioned certain speed is ignored, but perhaps we could at least make valid fan speed to be triggered at a lower temperatures? 

My GS65's fan is only spinning at 80% even when the CPU temperature has already reached 90 degree, which is the CPU's throttling threshold :(

I tried to modify the values corresponding to the different "temp_x", but that does not seem to make any difference...

ZZZXXX0110, 2019-03-28:
Will be happy to do testing for you, by the way!

YoyPa, 2019-03-29:
I will look at it this week end, my main concern is that I'm not sure if the EC show the set fan value or the true fan speed. So I don't know if the fan speed actually change. I Need to boot windows and test with RWeverything+MSI tool which show me fan speed in RPM.

ZZZXXX0110, 2019-03-29:
> I will look at it this week end, my main concern is that I'm not sure if the EC show the set fan value or the true fan speed. So I don't know if the fan speed actually change. I Need to boot windows and test with RWeverything+MSI tool which show me fan speed in RPM.

Using another tool written by someone else in Reddit, I was able to change the fan speed to 100% (and I can hear the fan speeding up significantly), then isw shows that EC does indeed correctly report the true fan speed.

ZZZXXX0110, 2019-03-29:
Also, I think the EC addresses in the GS65 profile might be incorrect, I tried to change fan speed by directly inputing EC values, but it did not work.

I got it to work however, with the EC address from here: https://gist.github.com/falmear/fba37bc63dd93346dc1e9962dda9d922#file-gistfile1-txt-L19

YoyPa, 2019-03-30:
> Also, I think the EC addresses in the GS65 profile might be incorrect, I tried to change fan speed by directly inputing EC values, but it did not work.
> 
> I got it to work however, with the EC address from here: https://gist.github.com/falmear/fba37bc63dd93346dc1e9962dda9d922#file-gistfile1-txt-L19

Nice file, it's comforting me on what I do and think, if you look at wiki png: 
https://github.com/YoyPa/isw/blob/master/wiki/ec.png
we are using exactly same addresses.
I only disagree with 0x69 and 0x81, I tested and it change nothing.
And I agree on 0x70 and 0x88 they probably are useless as there are no corresponding fan speed. (That's why I only alter 6 value instead of 7 with isw).

YoyPa, 2019-03-30:
Support added with https://github.com/YoyPa/isw/commit/828fa40179a80124bc7af8c63454796244ab7cca
Please tell me if it's ok.

closed 2019-04-24 by YoyPa