YoyPa/isw #5: Need help for figuring out whats wrong

kamyd, 2019-04-12:
Hi and Thanks a lot for your great work and for helping people like me
i use xubuntu 19.04 and i used your guide for other distros to copy the files in right directories 
but when i use command sudo isw -r it ask for profile name and i cant figure out what it is
and this is my isw -c:
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F

I have GE62 6QE Apache pro and i really need to be able to control my fan speed 
Thank you in advance 
i will gladly provide any other file that is needed

YoyPa, 2019-04-13:
Hmmm,
Does /sys/kernel/debug/ec/ec0/io exist ?
```
ls -la /sys/kernel/debug/ec/ec0/io
```
is od available ?
```
which od
```
If both response are yes, can you try:
```
sudo od -A x -t x1 /sys/kernel/debug/ec/ec0/io
```

kamyd, 2019-04-13:
First thanks a lot
please check if i done this right 
i copied all the files according to line 21 to 25 of the PKGBUILD
then ran modprobe ec_sys write_support=1
is this right?

and now:
yes both exist 
[image]

when i run sudo od -A x -t x1 /sys/kernel/debug/ec/ec0/io
it gives operation not permitted
[image]

kamyd, 2019-04-13:
and then i ran sudo isw -r GS65_8RF_DEFAULT and this is the result:
[image]

YoyPa, 2019-04-13:
For option -c and -r configuring ec_sys is not necessary, it's only for -s and -w.
And according to your ls -la, ec_sys --write_support=1 is correctly set.
It's clearly a permission problem, and I guess it's related to sudo and /etc/sudoers.
You can try as root (su root) and if it work you know where to look.

kamyd, 2019-04-13:
wow man thank you so much
your advice gave me a hint and that not permitted error was because of my secure-boot i disabled it and its fixed now:
[image]
 
now what should i do to change my fan speeds? ( I know you wrote how but i didnt quit understand how)
i should change one of those profiles fan speed to my need and run isw -w profile name?
and if i restart the system these changes will be gone right?

kamyd, 2019-04-13:
and this is my isw -c i dont know why but i think you need it to guide me
[image]

Thank you again

YoyPa, 2019-04-13:
Nice to know that secure boot can prevent access to EC.
I will add GE62_6QE_DEFAULT in isw.conf, need around 15min.

kamyd, 2019-04-13:
> Nice to know that secure boot can prevent access to EC.
> I will add GE62_6QE_DEFAULT in isw.conf, need around 15min.

Cool and Thank you again 
I didnt expect that either , once on debian 9 i got the same error for something completely different and later i figured it was caused by secure boot and since its based on debian i though it can help.

There is just one thing that i dont understand:
after copying all the files from cloned folder i get 'encountered system error' after each boot and give me a option to report , i dont understand whats causing this , and i dont get that error before copying files.
do you have any idea?

YoyPa, 2019-04-13:
> now what should i do to change my fan speeds? ( I know you wrote how but i didnt quit understand how)
> i should change one of those profiles fan speed to my need and run isw -w profile name?

Edit isw.conf, I made GE62_6QE_DEFAULT profile. You can either copy paste it and rename it GE62_6QE_KAMYAR then edit values or directly edit GE62_6QE_DEFAULT.
```
# cpu fan will spin at fan_speed_0:
# 		if realtime_cpu_temp is above temp_0
#			fan_speed_1 is applied
# 		if realtime_cpu_temp is above temp_1
#			fan_speed_2 is applied
#		[...]
```
> and if i restart the system these changes will be gone right?

EC is reset on PowerSourceChange/Suspend/Hibernate/Reboot.
You can reapply on startup via systemd service (see bottom of readme).
> system program problem detected

I never encountered this, and need more info, maybe in journalctl or dmesg.

kamyd, 2019-04-13:
> [346b501](https://github.com/YoyPa/isw/commit/346b501b43aa643693721b70732e3fc0fb3a1fd0)
> 
> > now what should i do to change my fan speeds? ( I know you wrote how but i didnt quit understand how)
> > i should change one of those profiles fan speed to my need and run isw -w profile name?
> 
> Edit isw.conf, I made GE62_6QE_DEFAULT profile. You can either copy paste it and rename it GE62_6QE_KAMYAR then edit values or directly edit GE62_6QE_DEFAULT.
> temp are in °C and fan_speed in %.
> 
> ```
> # cpu fan will spin at fan_speed_0:
> # 		if realtime_cpu_temp is above temp_0
> #			fan_speed_1 is applied
> # 		if realtime_cpu_temp is above temp_1
> #			fan_speed_2 is applied
> #		[...]
> ```
> 
> > and if i restart the system these changes will be gone right?
> 
> EC is reset on PowerSourceChange/Suspend/Hibernate/Reboot.
> You can reapply on startup via systemd service (see bottom of readme).
> 
> > system program problem detected
> 
> I never encountered this, and need more info, maybe in journalctl or dmesg.

You just made my day Thank you so so much 

its working perfectly

and my friend has another msi notebook the last time i checked our ec firmware name was the same i will test my profile on his system and report back to add its name to supported devices 

and lastly thank you again for your great work

YoyPa, 2019-04-13:
:)

closed 2019-04-13 by YoyPa

KolbjornBlix, 2021-02-22:
Could you add a profile for the GP65 Leopard 9SD based on this:

EC dump
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 1b 0b  >................<
000030 02 05 00 05 51 0a 05 00 26 13 6c 2a ea 01 c7 00  >....Q...&.l*....<
000040 22 0b 49 00 8d 10 b1 fb 00 0c 7b 2b a3 0b 0c 30  >".I.......{+...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 27 00 32 39 40 47 4e 55  >........'.29@GNU<
000070 64 00 00 28 30 38 40 48 50 00 0a 08 08 08 08 08  >d..(08@HP.......<
000080 00 00 37 3c 41 46 4b 50 61 00 00 2c 34 3c 46 50  >..7<AFKPa..,4<FP<
000090 50 00 0a 07 07 07 07 07 02 16 7d 02 16 61 00 00  >P.........}..a..<
0000a0 31 36 55 31 45 4d 53 31 2e 31 30 31 30 35 32 30  >16U1EMS1.1010520<
0000b0 32 30 31 39 31 30 3a 30 34 3a 30 32 bf 10 40 00  >201910:04:02..@.<
0000c0 00 01 23 00 00 9e 40 00 00 00 00 00 00 00 14 00  >..#...@.........<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 27 10 00 00 00 00 00 01 00 00 d1 00 e4  >...'............<
0000f0 00 00 80 02 0c 00 05 80 00 00 00 00 00 00 00 00  >................<
000100