YoyPa/isw #6: Support for MSI GL63 8RD

DadouLapous, 2019-04-14:
Hello
Can you add support for GL63 8RD ? 
My EC dump :

`sudo isw -c                                          
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b  >..............I.<
000030 03 09 00 0d 51 0a 05 00 54 0f 30 2a 88 01 e7 00  >....Q...T.0*....<
000040 00 00 60 00 f7 0e 00 00 57 0e 5f 2f b1 0b 0c 30  >..`.....W._/...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2a 0a 32 39 40 47 4e 55  >........*.29@GNU<
000070 64 28 00 28 2f 36 3d 44 4b 5f 0a 08 08 08 08 08  >d(.(/6=DK_......<
000080 2c 36 37 3c 41 46 4b 50 63 00 00 35 3b 41 47 4d  >,67<AFKPc..5;AGM<
000090 53 00 0a 07 07 07 07 07 06 16 7d 02 16 63 00 00  >S.........}..c..<
0000a0 31 36 50 36 45 4d 53 31 2e 31 30 34 30 36 31 33  >16P6EMS1.1040613<
0000b0 32 30 31 38 31 34 3a 33 33 3a 35 38 f7 0e 00 40  >201814:33:58...@<
0000c0 00 01 23 00 00 00 00 00 00 39 00 00 00 c6 00 00  >..#......9......<
0000d0 00 00 00 00 7f 00 00 00 00 00 00 80 00 00 00 00  >................<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 b8 00 00  >................<
0000f0 00 00 80 01 0c 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

`
Thanks !

YoyPa, 2019-04-15:
Format with ```
```
sudo isw -c
EC dump
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b >..............I.<
000030 03 09 00 0d 51 0a 05 00 54 0f 30 2a 88 01 e7 00 >....Q...T.0*....<
000040 00 00 60 00 f7 0e 00 00 57 0e 5f 2f b1 0b 0c 30 >..`.....W./...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
000060 00 00 00 00 00 00 00 00 2a 0a 32 39 40 47 4e 55 >........*.29@GNU<
000070 64 28 00 28 2f 36 3d 44 4b 5f 0a 08 08 08 08 08 >d(.(/6=DK......<
000080 2c 36 37 3c 41 46 4b 50 63 00 00 35 3b 41 47 4d >,67<AFKPc..5;AGM<
000090 53 00 0a 07 07 07 07 07 06 16 7d 02 16 63 00 00 >S.........}..c..<
0000a0 31 36 50 36 45 4d 53 31 2e 31 30 34 30 36 31 33 >16P6EMS1.1040613<
0000b0 32 30 31 38 31 34 3a 33 33 3a 35 38 f7 0e 00 40 >201814:33:58...@<
0000c0 00 01 23 00 00 00 00 00 00 39 00 00 00 c6 00 00 >..#......9......<
0000d0 00 00 00 00 7f 00 00 00 00 00 00 80 00 00 00 00 >................<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 b8 00 00 >................<
0000f0 00 00 80 01 0c 00 05 00 00 00 00 00 00 00 00 00 >................<
000100
```

YoyPa, 2019-04-15:
Added it with https://github.com/YoyPa/isw/commit/74f657e2e476a123911b94c399ff3569f755f1b5
Just a thing on your dump, current profile may need a little change:
Your temp was 0x2a(address 0x68) (below 0x32(addr. 0x6A)).
So fan_speed should've been 0x00(addr. 0x72), but is instead set at 0x28(addr. 0x71 and addr. 0x73).
And there is an additional value at the end 0x5f(addr. 0x79).

Can you test if value @0x72 is used and on which temperature condition ?
Can you test if value @0x79 is used and on which temperature condition ?

DadouLapous, 2019-04-15:
My GPU temp is matching address 0x68
I may have found why all my fans were on, i can enable all my fans by pressing a hotkey,
and by playing with it and waiting a bit i had different results for 0x71.
0x72 and 0x79 seems to stay at 0x00 and 0x5f.

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b  >..............I.<
000030 03 03 00 05 51 0a 05 00 54 0f 30 2a 88 01 87 00  >....Q...T.0*....<
000040 98 08 63 00 c5 0e b3 00 91 0e fc 2f a8 0b 0c 30  >..c......../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2a 0a 32 39 40 47 4e 55  >........*.29@GNU<
000070 64 00 00 28 2f 36 3d 44 4b 5f 0a 08 08 08 08 08  >d..(/6=DK_......<
000080 28 36 37 3c 41 46 4b 50 63 00 00 35 3b 41 47 4d  >(67<AFKPc..5;AGM<
000090 53 00 0a 07 07 07 07 07 06 16 7d 02 16 63 00 00  >S.........}..c..<
0000a0 31 36 50 36 45 4d 53 31 2e 31 30 34 30 36 31 33  >16P6EMS1.1040613<
0000b0 32 30 31 38 31 34 3a 33 33 3a 35 38 c5 0e 00 40  >201814:33:58...@<
0000c0 00 07 25 00 00 00 00 00 00 41 00 00 00 00 00 00  >..%......A......<
0000d0 00 00 00 00 7f 00 00 00 00 00 00 80 00 00 00 00  >................<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 80 01 0c 00 05 00 00 00 00 00 00 00 00 00  >................<
000100
```

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b  >..............I.<
000030 03 09 00 0d 51 0a 05 00 54 0f 30 2a 88 01 a7 40  >....Q...T.0*...@<
000040 00 00 64 00 c3 0e 00 00 c3 0e ef 2f a8 0b 0c 30  >..d......../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2f 0a 32 39 40 47 4e 55  >......../.29@GNU<
000070 64 2f 00 28 2f 36 3d 44 4b 5f 0a 08 08 08 08 08  >d/.(/6=DK_......<
000080 2d 36 37 3c 41 46 4b 50 63 00 00 35 3b 41 47 4d  >-67<AFKPc..5;AGM<
000090 53 00 0a 07 07 07 07 07 06 16 7d 02 16 63 00 00  >S.........}..c..<
0000a0 31 36 50 36 45 4d 53 31 2e 31 30 34 30 36 31 33  >16P6EMS1.1040613<
0000b0 32 30 31 38 31 34 3a 33 33 3a 35 38 c5 0e 00 40  >201814:33:58...@<
0000c0 00 07 25 00 00 00 00 00 00 40 00 00 00 ca 00 00  >..%......@......<
0000d0 00 00 40 00 7f 00 00 00 00 00 00 80 00 00 00 00  >..@.............<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 80 01 0c 00 03 00 00 00 00 00 00 00 00 00  >................<
000100
```

YoyPa, 2019-04-15:
The first dump is working as expected.
On the second one, fan_speed doesnt correspond to temperature treshold.
```
          | realtime |
----------+----------+--------------------------------
temp      |       2f |<32 >32  >39  >40  >47  >4e  >55
----------+----------+--↓---↓----↓----↓----↓----↓----↓
fan_speed |       2f | 00  28   2f   36   3d   44   4b  
                   ↑
         sould be 00
```

DadouLapous, 2019-04-15:
I can't write into EC : 
```
sudo isw -w GL63_8RD_DEFAULT

cpu_temp:
Value 0x32(50°C) written at 0x6a(byte106)
Traceback (most recent call last):
  File "/usr/bin/isw", line 84, in ec_write
    file.seek(list_ta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 142, in <module>
    main()
  File "/usr/bin/isw", line 139, in main
    parser.parse_args()
  File "/usr/lib/python3.7/argparse.py", line 1749, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1781, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1987, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.7/argparse.py", line 1927, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.7/argparse.py", line 1839, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.7/argparse.py", line 2369, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.7/argparse.py", line 2402, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 91, in ec_write
    print('Value '+hex(list_fs[i])+'('+str(list_fs[i])+'°C)'+' written at '+hex(list_fsa[i])+'(byte'+str(list_fsa[i])+')')
OSError: [Errno 22] Invalid argument
```

YoyPa, 2019-04-15:
Probably ec_sys not having --write_support=1

DadouLapous, 2019-04-15:
I have installed is with AUR, and rebooted since, and also tried `modprobe ec_sys write_support=1`, is there another step to do ?

YoyPa, 2019-04-15:
From readme:
```
It need ec_sys module with option write_support=1, there are two scenario to set that:

    ec_sys is a builtin kernel module:
        add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
        then update your grub and reboot.
    ec_sys is not a builtin kernel module:
        copy both isw-ec_sys.conf files provided (/etc/mod[...]) with same path (Arch AUR package will do it for you).
        then reboot OR modprobe ec_sys write_support=1.
```
If ```modinfo ec_sys``` throw an error, ec_sys is builtin.
For exemple my grub look like that:
```
$ cat /etc/default/grub | grep -i CMDLINE
GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi=! acpi_osi=\"Windows 2009\" acpi_backlight=vendor video.report_key_events=0 ec_sys.write_support=1"
GRUB_CMDLINE_LINUX="cryptdevice=UUID=8e253eae-1b4b-4721-b93e-516da16f35ca:cryptroot"
```

DadouLapous, 2019-04-15:
Okay, thanks
I wrote to EC that : 
```
sudo isw -w GL63_8RD_2
cpu_temp:
Value 0xa(10°C) written at 0x6a(byte106)
Value 0xb(11°C) written at 0x6b(byte107)
Value 0xc(12°C) written at 0x6c(byte108)
Value 0xd(13°C) written at 0x6d(byte109)
Value 0xe(14°C) written at 0x6e(byte110)
Value 0xf(15°C) written at 0x6f(byte111)
cpu_fan_speed:
Value 0x0(0°C) written at 0x72(byte114)
Value 0x28(40°C) written at 0x73(byte115)
Value 0x2f(47°C) written at 0x74(byte116)
Value 0x36(54°C) written at 0x75(byte117)
Value 0x3d(61°C) written at 0x76(byte118)
Value 0x44(68°C) written at 0x77(byte119)
Value 0x4b(75°C) written at 0x78(byte120)
```
But i couldn't really hear a noise difference, and even with fan button enabled, fans stays at 75% (`sudo isw -r GL63_8RD_DEFAULT`)

YoyPa, 2019-04-15:
Ok, so 0x5f(addr. 0x79) is never used, the profile is correct, thanks for your testing.

PS: Just noticed fan speed are reported in °C :rofl:

x0wllaar, 2019-05-03:
There's definitely something more going on with this laptop. 
For me, realtime CPU and GPU fan speeds are at 0xCD and 0xCB respectively.

I've decompiled the DSTD and it doesn't mention those offsets at all. Also, while 0x68 is called CPUT in there, 0x80 is called SYST and not GPUT or something.

Do you now how to get the exact EC model number without disassembling the laptop so I can look up a datasheet?

YoyPa, 2019-05-04:
> For me, realtime CPU and GPU fan speeds are at 0xCD and 0xCB respectively.

Weird, it's a bunch of zero for me.

> I've decompiled the DSTD and it doesn't mention those offsets at all. Also, while 0x68 is called CPUT  in there, 0x80 is called SYST and not GPUT or something.

Don't know much about DSDT, never took the time to look into it, but it's the same on GS40
```
                Offset (0x68), 
                CPUT,   8, 
                Offset (0x7E), 
                RES1,   3, 
                CHET,   1, 
                Offset (0x80), 
                SYST,   8,
```
> Do you now how to get the exact EC model number without disassembling the laptop so I can look up a datasheet?

No :/

x0wllaar, 2019-05-06:
> No :/

I am almost sure that it's either ENE KB9028 or KB9032, which is bad because there's no datasheet for it :-(

I might try experimenting with it a little bit more

DadouLapous, 2019-07-10:
> The first dump is working as expected.
> On the second one, fan_speed doesn't correspond to temperature threshold.
> 
> ```
>           | realtime |
> ----------+----------+--------------------------------
> temp      |       2f |<32 >32  >39  >40  >47  >4e  >55
> ----------+----------+--↓---↓----↓----↓----↓----↓----↓
> fan_speed |       2f | 00  28   2f   36   3d   44   4b  
>                    ↑
>          sould be 00
> ```
> 
> GL63 hotkey seems to alter fan speed without touching profile.
> Wondering if 0x5f is ever used: If you set low temp threshold like 10 11 12 13 14 15 °C and play with GL63 hotkey does fan speed 0x5f is ever applied ?

I still have this strange issue, my fan speed doesn't go down to `gpu_fan_speed_0`  even if the cpu cooled down, 
somehow, i can force it to go down to 0 by forcing it to go to `gpu_fan_speed_6` (via a different profile), and reapplying the 1rst profile it goes down to `gpu_fan_speed_0` (but it weirdly gets stuck at `gpu_fan_speed_6`  for some minutes before finally start decreasing).
I don't have this issue with my gpu. 

I know that's not really linked to isw, but you might have an idea ? 
I'll try to install windows in some weeks to see whats happening in dragon center

YoyPa, 2019-07-11:
Maybe changing fan_mode can help:
```
fan_mode can be 140, 76 or 12, for Advanced, Basic or Auto. Advanced seems to work better with suspend/hibernate.
```

DadouLapous, 2019-07-11:
Well i don't see any kind of different behavior between fan modes, I will have to dig a bit into this.
I'll see later if windows can handle this...

YoyPa, 2019-07-15:
Hello, an other maybe useless clue:

I started looking into zip file provided by msi, they are very long, and seems to have multiple (seen 4max) profiles (no clue about the purpose). Most reported laptop use 1st profile, WS65_8SK use the 2nd.
Your EC version is 16P6EMS1.104:
```
wget download.msi.com/archive/frm_exe/nb/16P6EMS1.104.zip
```
you can look into it with:
```
od -A x -t x1z 16P6EMS1.104
```
Profiles are located around 0xf800, you can check them with:
```
isw -f 16P6EMS1.104
```

DadouLapous, 2019-08-23:
Hello,
I've finally found the problem :)
It seems that to apply `cpu_fan_speed_0`, is re-applied when  
```cpu_current_temp  <= cpu_temp_0  - 10```
fan goes back to  `cpu_fan_speed_1` as usual, when 
```cpu_current_temp  >= cpu_temp_0 ```
My `cpu_temp_0  - 10` was just a bit too low !

closed 2019-08-23 by DadouLapous