YoyPa/isw #7: Support for changing GPU fan speed?

iExalt, 2019-04-21:
I'm using ISW right now on a 16L13 and the CPU temp thresholds and fan speeds work perfectly fine. My GPU fan seems to stay at a constant fan speed however. If I understood the EC hexdump better I'd try my hand at writing it myself, but from my understanding 0x8A - 0x8F contain the GPU fan speeds and 0x82 - 0x88 contain the GPU fan temperature thresholds.

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 26 4b 0b  >.............&K.<
000030 03 09 00 05 50 0a 05 00 86 15 40 38 26 02 e0 40  >....P.....@8&..@<
000040 00 00 63 00 20 14 00 00 c7 13 cf 40 ec 0b a0 41  >..c. ......@...A<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 22 00 0a 14 1a 20 26 2c  >........".... &,<
000070 64 2d 14 1e 23 28 2d 37 4b 5f 07 09 08 06 03 03  >d-..#(-7K_......<
000080 2f 00 3d 43 49 4f 55 5a 5e 13 13 27 2b 2f 36 3e  >/.=CIOUZ^..'+/6><
000090 64 00 03 03 03 03 03 03 06 05 69 02 64 69 00 00  >d.........i.di..<
0000a0 31 36 4c 31 45 44 36 31 2e 31 30 37 30 35 30 35  >16L1ED61.1070505<
0000b0 32 30 31 36 31 33 3a 35 38 3a 30 31 20 14 00 30  >201613:58:01 ..0<
0000c0 00 01 22 00 00 9e c0 00 00 62 01 9a 00 ac 14 00  >.."......b......<
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00  >....p...........<
0000e0 e2 02 00 31 10 00 00 00 00 00 00 00 00 82 00 00  >...1............<
0000f0 00 00 00 00 0c 0f 05 00 00 00 00 00 00 01 00 00  >................<
000100
```

YoyPa, 2019-04-23:
> but from my understanding 0x8A - 0x8F contain the GPU fan speeds and 0x82 - 0x88 contain the GPU > fan temperature thresholds.

You are mostly right:
0x8A - 0x90*, the value 8A doesn't work for my GS40(black square): https://github.com/YoyPa/isw/blob/master/wiki/ec.png, maybe it's spread accross models, maybe it's just mine, maybe I'm wrong :D.

0x82 - 0x87*, I think 0x88 is useless:
```
 gpu fan will spin at 0x8A:
 		if realtime_gpu_temp is above 0x82
			0x8B is applied
 		if realtime_gpu_temp is above 0x83
			0x8C is applied
 		if realtime_gpu_temp is above 0x84
			0x8D is applied
 		if realtime_gpu_temp is above 0x85
			0x8E is applied
 		if realtime_gpu_temp is above 0x86
			0x8F is applied
 		if realtime_gpu_temp is above 0x87
			0x90 is applied
 		if realtime_gpu_temp is above 0x88
			??? no more available value ???
```
I'm currently adding gpu support, it take a bit of time as I need to edit every currently available laptop to add their gpu values.

YoyPa, 2019-04-23:
GPU fan speed change added with https://github.com/YoyPa/isw/commit/71d6d7d8e7890dd0ec738e389ce40077d35ffdbb https://github.com/YoyPa/isw/commit/601742a5e631e875ca52844497370add83caf886
MSI_16L13 added with https://github.com/YoyPa/isw/commit/75eb93b6ac28e8006cfb650381186366d954ee39

closed 2019-04-24 by YoyPa