YoyPa/isw #8: Support for MSI GT60 2PE

joematt18, 2019-04-24:
Hi
Here is the EC dump for the MSI GT60 2PE:

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
*
000020 00 00 00 00 00 00 00 00 00 00 00 80 80 06 09 0b  >................<
000030 03 09 00 05 51 0a 05 00 78 1e 5c 2b bc 02 e0 40  >....Q...x.\+...@<
000040 00 00 62 00 1d 0b 00 00 cb 0a 97 30 9e 0b 38 31  >..b........0..81<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2b 00 34 40 4c 58 5d 5f  >........+.4@LX]_<
000070 64 20 20 20 23 2d 32 42 4c 00 05 09 06 08 03 01  >d   #-2BL.......<
000080 00 00 34 40 4c 57 5a 5f 64 00 20 20 2a 37 42 4c  >..4@LWZ_d.  *7BL<
000090 64 00 02 02 02 02 02 02 02 00 7d 02 00 66 00 00  >d.........}..f..<
0000a0 31 36 46 34 45 4d 53 31 20 56 65 72 35 2e 31 37  >16F4EMS1 Ver5.17<
0000b0 2c 20 31 32 2f 30 34 2f 32 30 31 34 1d 0b 00 00  >, 12/04/2014....<
0000c0 00 01 25 00 0c 9e c0 00 00 1f 00 00 00 fb 14 00  >..%.............<
0000d0 00 00 00 00 60 00 01 02 00 00 00 80 01 00 00 00  >....`...........<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 b7 00 00  >................<
0000f0 00 04 14 1b a1 28 2e 3c 49 57 64 64 01 f4 04 08  >.....(.<IWdd....<
000100

Could you add support for this model? I would be grateful.

YoyPa, 2019-04-26:
Format with ```
```
EC dump
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
*
000020 00 00 00 00 00 00 00 00 00 00 00 80 80 06 09 0b >................<
000030 03 09 00 05 51 0a 05 00 78 1e 5c 2b bc 02 e0 40 >....Q...x.+...@<
000040 00 00 62 00 1d 0b 00 00 cb 0a 97 30 9e 0b 38 31 >..b........0..81<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
000060 00 00 00 00 00 00 00 00 2b 00 34 40 4c 58 5d 5f >........+.4@LX]_<
000070 64 20 20 20 23 2d 32 42 4c 00 05 09 06 08 03 01 >d #-2BL.......<
000080 00 00 34 40 4c 57 5a 5f 64 00 20 20 2a 37 42 4c >..4@LWZ_d. *7BL<
000090 64 00 02 02 02 02 02 02 02 00 7d 02 00 66 00 00 >d.........}..f..<
0000a0 31 36 46 34 45 4d 53 31 20 56 65 72 35 2e 31 37 >16F4EMS1 Ver5.17<
0000b0 2c 20 31 32 2f 30 34 2f 32 30 31 34 1d 0b 00 00 >, 12/04/2014....<
0000c0 00 01 25 00 0c 9e c0 00 00 1f 00 00 00 fb 14 00 >..%.............<
0000d0 00 00 00 00 60 00 01 02 00 00 00 80 01 00 00 00 >....`...........<
0000e0 e2 02 00 04 10 00 00 00 00 00 00 00 00 b7 00 00 >................<
0000f0 00 04 14 1b a1 28 2e 3c 49 57 64 64 01 f4 04 08 >.....(.<IWdd....<
000100
```

joematt18, 2019-04-26:
Sorry, but how I do that?
I'm extremely newbie, just installed Manjaro a week ago, so I'm kinda helpless.

> Format with ```
>
> ```
> EC dump
>        00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
> 000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 >................<
> *
> 000020 00 00 00 00 00 00 00 00 00 00 [...]

YoyPa, 2019-04-26:
Done, I know your laptop only have one fan, but the EC still contain cpu and gpu part like other laptops. Don't know if gpu part is useful (like GPU on/off modifier) but I added it anyway.

YoyPa, 2019-04-26:
> Sorry, but how I do that?
> I'm extremely newbie, just installed Manjaro a week ago, so I'm kinda helpless.

With those three characters: ```
(before and after the formatted text)

joematt18, 2019-04-26:
many thanks
I think it's probably useful, because both cpu and gpu share the same fan, but different heat pipes, in windows when only the gpu is hot the fan speeds up too.

> Done with [effec98](https://github.com/YoyPa/isw/commit/effec98e5eecb9f576747193837ea8933c0fbbe0), I know your laptop only have one fan, but the EC still contain cpu and gpu part like other laptops. Don't know if gpu part is useful (like GPU on/off modifier) but I added it anyway.

joematt18, 2019-04-26:
> > Sorry, but how I do that?
> > I'm extremely newbie, just installed Manjaro a week ago, so I'm kinda helpless.
> 
> With those three characters: ```
> (before and after the formatted text)

thanks again, it's nice to get help in the beginning

YoyPa, 2019-04-26:
By the way I will not make a release for one laptop, so no update will be shown by pacman/yay/trizen/[insert you package manager] so you have to manually replace your ```/etc/isw.conf``` with the one in this git.

joematt18, 2019-04-26:
I tried to use the profile and got this in response

```[castrojvm@castrojvm ~]$ sudo isw -w GT60_2PE_DEFAULT[sudo] password for castrojvm: 
Traceback (most recent call last):
  File "/usr/bin/isw", line 74, in ec_write
    file.seek(list_gta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 160, in <module>
    main()
  File "/usr/bin/isw", line 157, in main
    parser.parse_args()
  File "/usr/lib/python3.7/argparse.py", line 1749, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1781, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1987, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.7/argparse.py", line 1927, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.7/argparse.py", line 1839, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.7/argparse.py", line 2369, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.7/argparse.py", line 2402, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 101, in ec_write
    hex(list_gfsa[i])+'(byte'+str(list_gfsa[i])+')'
OSError: [Errno 22] Invalid argument
```

YoyPa, 2019-04-27:
I think module ```ec_sys``` doesn't have ```write_support```.
From the readme:
```
ec_sys is a builtin kernel module:
    add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
    then update your grub and reboot.
```

joematt18, 2019-04-27:
This is another thing that I'm in doubt, how do I do that?
I simply copy this line to the file? how I update the grub?

> I think module `ec_sys` doesn't have `write_support`.
> From the readme:
> 
> ```
> ec_sys is a builtin kernel module:
>     add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
>     then update your grub and reboot.
> ```

YoyPa, 2019-04-27:
You need to modify one specific line of /etc/default/grub, for example mine look like that:
(Notice ```ec_sys.write_support=1``` at the end)
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi=! acpi_osi=\"Windows 2009\" acpi_backlight=vendor video.report_key_events=0 ec_sys.write_support=1"
```
Then to update your grub (Apply the config file) you need to type:
```
sudo update-grub
```

joematt18, 2019-04-27:
I updated the GRUB and now it's working, many thanks, and congratulations, great job, now I have not just control of the fans, but more granular control than with the MSI Dragon center.

closed 2019-04-27 by YoyPa