YoyPa/isw #9: Set "Fan Mode" first

edward-p, 2019-05-14:
Finnally, I found out `0xf4` controls the Fan Mode:

|Value|Fan Mode|
|-------|--------------|
|0x0c|Auto|
|0x4c|Basic|
|0x8c|Advanced|

The default mode of My Laptop on start up is set to `0x0c` for `Auto Mode`.
I have to write `0x8c` into `0xf4` to enable `Avanced Mode` first.
Then I will be able to control fan speed by setting up `0x72-0x78`, other wise those data will be overwritten by EC itself under `Auto Mode` (It just acts like I don't have write access of `/sys/kernel/debug/ec/ec0/io` :rofl: ).
Now things are more clear. After you  waking the laptop from suspend mode,  `0xf4`  is set to `0x0c` for `Auto Mode`, I guess.

YoyPa, 2019-05-16:
Dump would be easier to read with ```od -A x -t x1 /sys/kernel/debug/ec/ec0/io``` or via ```isw -c```

Nice finding, this is echoing MSI's silent option behavior:
http://i68.tinypic.com/2l8jdpx.jpg

> The default mode of My Laptop on start up is set to 0x0c for Auto Mode.

Same here on GS40 6QE

>I have to write 0x8c into 0xf4 to enable Avanced Mode first. Then I will be able to control fan speed by setting up 0x72-0x78, other wise those data will be overwritten by EC itself under Auto Mode

That's different :thinking: . When is it overwritten ? after a timer/action ?
I will probably add an option like -m auto/basic/advanced

edward-p, 2019-05-16:
> That's different  . When is it overwritten ? after a timer/action ?

Sorry, that's my fault. It will be overwritten only when you write `v:0x0c` into `a:0xf4`.
It depends on how I modify the EC IO file. I used to dump, modify it and write the whole file back. So `v:0x0c` is written into `a:0xf4` again. Then you know what happens the next. :rofl:

With `sudo isw -w 'GE62_6QE_DEFAULT'` in command line, there's no problem. 

---

But I still have problem with the systemd service:

- I enabled `isw@GE62_6QE_DEFAULT.service`. 
- My systemd show it was successfully executed and exited.
```
● isw@GE62_6QE_DEFAULT.service - ISW fan control service
   Loaded: loaded (/usr/lib/systemd/system/isw@.service; enabled; vendor preset: disabled)
   Active: inactive (dead) since Thu 2019-05-16 17:11:40 CST; 8s ago
  Process: 6555 ExecStart=/usr/bin/isw -w GE62_6QE_DEFAULT (code=exited, status=0/SUCCESS)
 Main PID: 6555 (code=exited, status=0/SUCCESS)

May 16 17:11:40 leopard isw[6555]: [1B blob data]
May 16 17:11:40 leopard isw[6555]: 0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
May 16 17:11:40 leopard isw[6555]: 0x2f(47%)   0x73(byte115)    0x35(53%)   0x8b(byte139)
May 16 17:11:40 leopard isw[6555]: 0x38(56%)   0x74(byte116)    0x39(57%)   0x8c(byte140)
May 16 17:11:40 leopard isw[6555]: 0x42(66%)   0x75(byte117)    0x3d(61%)   0x8d(byte141)
May 16 17:11:40 leopard isw[6555]: 0x4c(76%)   0x76(byte118)    0x41(65%)   0x8e(byte142)
May 16 17:11:40 leopard isw[6555]: 0x56(86%)   0x77(byte119)    0x45(69%)   0x8f(byte143)
May 16 17:11:40 leopard isw[6555]: 0x56(86%)   0x78(byte120)    0x45(69%)   0x90(byte144)
May 16 17:11:40 leopard systemd[1]: isw@GE62_6QE_DEFAULT.service: Succeeded.
May 16 17:11:40 leopard systemd[1]: Started ISW fan control service.

```

But my fan speed is not set properly on Startup. :thinking:
The value of `a:0x72` should be `00`, but it is still `2f`.

```
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b
000030 03 09 00 0d 33 0a 05 00 54 0f 30 2a 88 01 e0 00
000040 00 00 61 00 7a 0e 00 00 f5 0d 73 2f c9 0b 0c 30
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000060 00 00 00 00 00 00 00 00 24 0a 37 3d 43 49 4f 4f
000070 64 2f 2f 32 38 42 4c 56 56 00 0a 08 08 08 08 08
000080 00 00 37 3c 41 46 4b 4b 5e 00 00 35 39 3d 41 45
000090 45 00 0a 07 07 07 07 07 02 16 64 02 0a 5e 00 00
0000a0 31 36 4a 35 45 4d 53 33 2e 31 31 30 30 36 31 35
0000b0 32 30 31 36 31 31 3a 30 35 3a 30 34 7a 0e 00 10
0000c0 00 06 23 00 00 9e c0 00 00 22 00 00 00 90 14 00
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 d3 00 00
0000f0 00 00 81 02 0c 0a 0f 00 00 00 00 00 00 00 00 00
000100
```

Then I perform `sudo isw -w 'GE62_6QE_DEFAULT'` again in my terminal, fan speeds are set properly.
The value of `a:0x72` is `00` now.

```
edward•~» sudo isw -w 'GE62_6QE_DEFAULT'                                             [17:15:44]

------------CPU----------    ----------GPU------------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
0x4f(79°C)  0x6f(byte111)    0x4b(75°C)  0x87(byte135)

0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
0x2f(47%)   0x73(byte115)    0x35(53%)   0x8b(byte139)
0x38(56%)   0x74(byte116)    0x39(57%)   0x8c(byte140)
0x42(66%)   0x75(byte117)    0x3d(61%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)    0x41(65%)   0x8e(byte142)
0x56(86%)   0x77(byte119)    0x45(69%)   0x8f(byte143)
0x56(86%)   0x78(byte120)    0x45(69%)   0x90(byte144)
edward•~» sudo od -A x -t x1 /sys/kernel/debug/ec/ec0/io                             [17:15:51]
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b
000030 03 09 00 05 33 0a 05 00 54 0f 30 2a 88 01 e0 00
000040 00 00 61 00 7a 0e 00 00 f5 0d 73 2f c7 0b 0c 30
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000060 00 00 00 00 00 00 00 00 2d 0a 37 3d 43 49 4f 4f
000070 64 00 00 2f 38 42 4c 56 56 00 0a 08 08 08 08 08
000080 00 00 37 3c 41 46 4b 4b 5e 00 00 35 39 3d 41 45
000090 45 00 0a 07 07 07 07 07 02 16 64 02 0a 5e 00 00
0000a0 31 36 4a 35 45 4d 53 33 2e 31 31 30 30 36 31 35
0000b0 32 30 31 36 31 31 3a 30 35 3a 30 34 7a 0e 00 10
0000c0 00 06 23 00 00 9e c0 00 00 2f 00 00 00 00 14 00
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 d3 00 00
0000f0 00 00 81 02 0c 0a 0f 00 00 00 00 00 00 00 00 00
000100
```
It seems like the systemd service is started too early.

---

Also, I want `isw@GE62_6QE_DEFAULT.service` to start on resume form suspend.
So I can apply my fan profile automatically on resume.
It may be a service file like this:
```
[Unit]
Description=ISW fan control service
After=suspend.target

[Service]
ExecStart=/usr/bin/isw -w %I
Type=oneshot

[Install]
WantedBy=multi-user.target suspend.target
```

YoyPa, 2019-05-16:
Ok, so there is no problem then, but the info on 0xf4 is still nice to have.

---

I will try to reproduce and fix the service issue.

---

I already did a second service file for sleep but I never pushed it (not tested/I'm lazy):
```
[Unit]
Description=ISW suspend/resume
After=sleep.target

[Service]
ExecStart=/usr/bin/systemctl start 'isw@*' --all
Type=oneshot

[Install]
WantedBy=sleep.target
```

YoyPa, 2019-05-16:
I was also affected and it seems adding a 2s sleep i enough. 1s is to low in my case.
It's kind of ugly but meh... it seems to work.
```
 [Service]
+ExecStartPre=/bin/sleep 2
 ExecStart=/usr/bin/isw -w %I
 Type=oneshot
```

YoyPa, 2019-05-16:
I'm hitting the same problem when resuming from suspend but using /bin/sleep doesn't work (Config is correctly applied 1/10).
And I don't understand the logic of what's happening:
CPU temp is not applied
GPU temp is applied
CPU fan speed is not applied
GPU fan speed is applied
ISW follow the order above when writing, so it look like CPU related values are reset after ISW finished. :face_with_thermometer:

edward-p, 2019-05-16:
What if we set fan mode first, then apply the profile.

To test it, a service file like this will be fine:
```
...
[Service]
ExecStart=/usr/bin/sh -c isw -s 0xf4 140 && isw -w %I
...
```
If it works. Will it still work, if we remove the 2s sleep?

YoyPa, 2019-05-16:
I used the execstartpre line instead:
```
[Service]
ExecStartPre=/usr/bin/isw -s 0xf4 140
ExecStart=/usr/bin/isw -w %I
```
Seems to work when booting (seems not rebooted enough) but not when resuming from suspend.
Don't know if it's relevant but :
```
#JOURNALCTL
mai 16 16:21:42 yo-gs isw[1752]: [1B blob data]
mai 16 16:21:42 yo-gs isw[1752]: Value       set @ address
mai 16 16:21:42 yo-gs isw[1752]: 0x8c(140)   0xf4(byte244)
mai 16 16:21:52 yo-gs isw[1754]: [1B blob data]
mai 16 16:21:52 yo-gs isw[1754]: -----------CPU-----------    -----------GPU-----------
mai 16 16:21:52 yo-gs isw[1754]: Value       set @ address    Value       set @ address
mai 16 16:21:52 yo-gs isw[1754]: 0x3f(63°C)  0x6a(byte106)    0x3f(63°C)  0x82(byte130)
mai 16 16:21:52 yo-gs isw[1754]: 0x44(68°C)  0x6b(byte107)    0x44(68°C)  0x83(byte131)
mai 16 16:21:52 yo-gs isw[1754]: 0x49(73°C)  0x6c(byte108)    0x49(73°C)  0x84(byte132)
mai 16 16:21:52 yo-gs isw[1754]: 0x4e(78°C)  0x6d(byte109)    0x4e(78°C)  0x85(byte133)
mai 16 16:21:52 yo-gs isw[1754]: 0x53(83°C)  0x6e(byte110)    0x53(83°C)  0x86(byte134)
mai 16 16:21:52 yo-gs isw[1754]: 0x58(88°C)  0x6f(byte111)    0x58(88°C)  0x87(byte135)
mai 16 16:21:52 yo-gs isw[1754]: [1B blob data]
mai 16 16:21:52 yo-gs isw[1754]: 0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
mai 16 16:21:52 yo-gs isw[1754]: 0x32(50%)   0x73(byte115)    0x32(50%)   0x8b(byte139)
mai 16 16:21:52 yo-gs isw[1754]: 0x3c(60%)   0x74(byte116)    0x3c(60%)   0x8c(byte140)
mai 16 16:21:52 yo-gs isw[1754]: 0x41(65%)   0x75(byte117)    0x41(65%)   0x8d(byte141)
mai 16 16:21:52 yo-gs isw[1754]: 0x46(70%)   0x76(byte118)    0x46(70%)   0x8e(byte142)
mai 16 16:21:52 yo-gs isw[1754]: 0x50(80%)   0x77(byte119)    0x50(80%)   0x8f(byte143)
mai 16 16:21:52 yo-gs isw[1754]: 0x5a(90%)   0x78(byte120)    0x5a(90%)   0x90(byte144)
mai 16 16:21:52 yo-gs systemd[1]: isw@GS40_6QE_YO.service: Succeeded.
mai 16 16:21:52 yo-gs systemd[1]: Started ISW fan control service.
mai 16 16:21:52 yo-gs systemd[1]: isw-sleep.service: Succeeded.
mai 16 16:21:52 yo-gs systemd[1]: Started ISW suspend/resume.

#DMESG
[jeu. mai 16 16:21:43 2019] ACPI: EC: EC started
[jeu. mai 16 16:21:43 2019] ACPI: EC: interrupt unblocked
[jeu. mai 16 16:21:43 2019] ACPI: EC: event unblocked
```
Considering dmesg is late by 8s (add 8s to dmesg time(according to my iptables log)) it's a 1s difference between EC start and isw write.

edward-p, 2019-05-16:
It works for me, with the `suspend.target` not the `sleep.target`.

```
[Unit]
Description=ISW fan control service
After=suspend.target

[Service]
ExecStartPre=/usr/bin/isw -s 0xf4 140
ExecStart=/usr/bin/isw -w %I
Type=oneshot

[Install]
WantedBy=multi-user.target suspend.target
```

---

```
#JOURNALCTL
May 16 16:50:32 leopard isw[15060]: [1B blob data]
May 16 16:50:32 leopard isw[15060]: Value       set @ address
May 16 16:50:32 leopard isw[15060]: 0x8c(140)   0xf4(byte244)
May 16 16:50:32 leopard isw[15075]: [1B blob data]
May 16 16:50:32 leopard isw[15075]: ------------CPU----------    ----------GPU------------
May 16 16:50:32 leopard isw[15075]: Value       set @ address    Value       set @ address
May 16 16:50:32 leopard isw[15075]: 0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
May 16 16:50:32 leopard isw[15075]: 0x3d(61°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
May 16 16:50:32 leopard isw[15075]: 0x43(67°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
May 16 16:50:32 leopard isw[15075]: 0x49(73°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
May 16 16:50:32 leopard isw[15075]: 0x4f(79°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
May 16 16:50:32 leopard isw[15075]: 0x4f(79°C)  0x6f(byte111)    0x4b(75°C)  0x87(byte135)
May 16 16:50:32 leopard isw[15075]: [1B blob data]
May 16 16:50:32 leopard isw[15075]: 0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
May 16 16:50:32 leopard isw[15075]: 0x2f(47%)   0x73(byte115)    0x35(53%)   0x8b(byte139)
May 16 16:50:32 leopard isw[15075]: 0x38(56%)   0x74(byte116)    0x39(57%)   0x8c(byte140)
May 16 16:50:32 leopard isw[15075]: 0x42(66%)   0x75(byte117)    0x3d(61%)   0x8d(byte141)
May 16 16:50:32 leopard isw[15075]: 0x4c(76%)   0x76(byte118)    0x41(65%)   0x8e(byte142)
May 16 16:50:32 leopard isw[15075]: 0x56(86%)   0x77(byte119)    0x45(69%)   0x8f(byte143)
May 16 16:50:32 leopard isw[15075]: 0x56(86%)   0x78(byte120)    0x45(69%)   0x90(byte144)
May 16 16:50:32 leopard systemd[1]: isw@GE62_6QE_DEFAULT.service: Succeeded.
May 16 16:50:32 leopard systemd[1]: Started ISW fan control service.
#DMESG
[Thu May 16 16:38:52 2019] ACPI: EC: EC started
[Thu May 16 16:38:52 2019] ACPI: EC: interrupt unblocked
[Thu May 16 16:38:52 2019] ACPI: EC: event unblocked
```
That's weird. :thinking:

YoyPa, 2019-06-13:
Sry for being super late :innocent: .
I added the Fan mode (advanced) which seems to solve the cpu config not being applied on suspend, it is configurable inside isw.conf.
I also added 2s sleep into the systemd service which seems to solve the 2/3 first values being reset.
I also switched from using 2 files to 1 like you did but still use sleep.target (suspend+hibernate).

And it seems to work so far :crossed_fingers: .

edward-p, 2019-06-14:
So far so good now. Close.

closed 2019-06-14 by edward-p