YoyPa/isw #10: Support for MSI GE62 7RE

BruynoogheMatthias, 2019-05-15:
So here's the dump made on the laptop

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 4b  >..............IK<
000030 03 09 00 0d 50 0a 05 00 26 13 6c 2a ea 01 e0 00  >....P...&.l*....<
000040 00 00 64 00 e5 0f 00 00 e5 0f dd 2f a0 0b 0c 30  >..d......../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2c 0a 37 3d 43 49 4f 4f  >........,.7=CIOO<
000070 64 30 28 30 38 42 4c 56 56 00 0a 08 08 08 08 08  >d0(08BLVV.......<
000080 2a 36 37 3c 41 46 4b 4b 63 00 00 36 3e 46 4e 56  >*67<AFKKc..6>FNV<
000090 56 00 0a 07 07 07 07 07 02 16 7d 02 16 63 00 00  >V.........}..c..<
0000a0 31 36 4a 39 45 4d 53 31 2e 31 30 39 30 35 32 32  >16J9EMS1.1090522<
0000b0 32 30 31 37 31 36 3a 30 36 3a 32 36 e5 0f 00 00  >201716:06:26....<
0000c0 00 07 25 00 00 9e c0 00 00 5e 00 00 00 c0 14 00  >..%......^......<
0000d0 00 00 00 00 73 00 01 00 00 00 00 80 81 00 00 00  >....s...........<
0000e0 e2 02 00 81 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 83 01 0c 00 0c 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
0x4f(79°C)  0x6f(byte111)    0x4b(75°C)  0x87(byte135)

0x28(40%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x30(48%)   0x73(byte115)    0x36(54%)   0x8b(byte139)
0x38(56%)   0x74(byte116)    0x3e(62%)   0x8c(byte140)
0x42(66%)   0x75(byte117)    0x46(70%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)    0x4e(78%)   0x8e(byte142)
0x56(86%)   0x77(byte119)    0x56(86%)   0x8f(byte143)
0x56(86%)   0x78(byte120)    0x56(86%)   0x90(byte144)
```

BruynoogheMatthias, 2019-05-15:
If I'm correct this is basically the same as the others?

YoyPa, 2019-05-16:
Profile added with https://github.com/YoyPa/isw/commit/ecc52770f7bcc1d2968805649b7b2b4141636675.
Yep, they seems the all work the same way.