YoyPa/isw #11: Temperature shown is different from i3blocks temperature(GS65 8RF)

reikdas, 2019-06-09:
The CPU temperature shown by `sudo isw -r GS65_8RF_DEFAULT` and by i3blocks (https://github.com/vivien/i3blocks-contrib/tree/master/temperature) differs inconsistently. 

By inconsistently, I mean the temperature difference is not constant.. as in there isn't always a temperature difference of for eg - 10 degrees. The difference can be any amount. Also, the temperature shown by isw is usually lesser than the temperature shown by i3blocks' script.

YoyPa, 2019-06-13:
Is it the same via ```sensors``` and ```i7z``` ?
Mine are pretty consistent:
```
#i7z
Core [core-id]  :Actual Freq (Mult.)      C0%   Halt(C1)%  C3 %   C6 %   C7 %  Temp      VCore
Core 1 [0]:       897.35 (8.96x)        3.41    4.05       0       1    93.6    46      0.6013
Core 2 [1]:       897.23 (8.96x)        3.58    4.41       1    1.31    91.9    49      0.5963
Core 3 [2]:       897.39 (8.96x)        2.57    2.02       1       1      95    46      0.6013
Core 4 [3]:       897.64 (8.96x)        3.89    2.62       1    1.02    93.8    45      0.6013

#sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +45.0°C  (high = +100.0°C, crit = +100.0°C)

#isw 
$ sudo isw -r GS40_6QE_DEFAULT
CPU
└─temp     (°C): 49
└─fan speed (%): 0
GPU
└─temp     (°C): 0
└─fan speed (%): 0

reikdas, 2019-06-15:
It seems like the i3blocks script is inconsistent. `sensors` and `isw` matches :)

closed 2019-06-15 by reikdas