YoyPa/isw #12: Support for MSI P65 8RD

mibofra, 2019-06-22:
Here is the requested data:

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 0b  >..............I.<
000030 02 05 01 0d 50 0a 05 00 a0 14 60 3b 71 02 c0 00  >....P.....`;q...<
000040 50 0a 46 00 9f 14 f5 f8 50 0e de 3c 24 0c f8 43  >P.F.....P..<$..C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 3b 00 37 40 46 4c 52 58  >........;.7@FLRX<
000070 64 32 1e 32 3c 4b 4b 4b 4b 00 08 03 03 03 03 03  >d2.2<KKKK.......<
000080 34 00 37 3d 41 47 4d 57 63 32 00 32 37 3c 41 44  >4.7=AGMWc2.27<AD<
000090 4e 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >N.........d..n..<
0000a0 31 36 51 33 45 4d 53 31 2e 31 30 33 30 39 30 36  >16Q3EMS1.1030906<
0000b0 32 30 31 38 30 39 3a 30 32 3a 34 36 93 14 00 40  >201809:02:46...@<
0000c0 00 01 25 00 00 00 00 00 00 02 00 a1 00 de 00 00  >..%.............<
0000d0 00 00 01 00 7f 00 d5 00 00 00 00 80 01 00 00 00  >................<
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 83 00 00  >................<
0000f0 00 00 80 00 0c 00 06 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)    0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)    0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)    0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)    0x57(87°C)  0x87(byte135)

0x1e(30%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)    0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)    0x37(55%)   0x8c(byte140)
0x4b(75%)   0x75(byte117)    0x3c(60%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)    0x41(65%)   0x8e(byte142)
0x4b(75%)   0x77(byte119)    0x44(68%)   0x8f(byte143)
0x4b(75%)   0x78(byte120)    0x4e(78%)   0x90(byte144)
```
As soon as possible, I'll make a request for anothe MSI laptop, as I'm the owner of a MSI WS65 too.

Cheers :)

YoyPa, 2019-06-23:
Done with commit https://github.com/YoyPa/isw/commit/dc81e09f5534c24dbccbbe8afe6833bea22a6581
:)

closed 2019-06-23 by YoyPa

mibofra, 2019-06-25:
Thanks! I'm gonna upgrade ubuntu on the other one, and send the profile