YoyPa/isw #13: Support for MSI GS75

jirislav, 2019-06-24:
Hi, here is the dump of MSI GS75 :) Thank you for the awesome program!
```
# isw -cp MSI_ADDRESS_DEFAULT

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e0 06 7a 0f  >..............z.<
000030 03 01 01 0d 50 0a 05 00 a0 14 60 3b 71 02 c0 00  >....P.....`;q...<
000040 50 0a 50 00 72 15 00 00 15 11 27 40 fd 0b f8 43  >P.P.r.....'@...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 38 00 37 40 46 4c 52 58  >........8.7@FLRX<
000070 64 3c 2d 32 3c 46 46 46 4b 00 08 03 03 03 03 03  >d<-2<FFFK.......<
000080 3c 00 37 3d 41 47 4d 57 5e 3c 00 32 3c 46 46 50  ><.7=AGMW^<.2<FFP<
000090 6e 00 08 03 03 03 03 05 02 00 7d 02 00 64 00 00  >n.........}..d..<
0000a0 31 37 47 31 45 4d 53 31 2e 31 30 34 30 33 31 34  >17G1EMS1.1040314<
0000b0 32 30 31 39 31 33 3a 35 36 3a 32 35 72 15 00 00  >201913:56:25r...<
0000c0 00 06 22 00 00 9e c0 00 00 ca 00 74 00 8e 14 00  >.."........t....<
0000d0 00 00 00 00 70 80 00 00 00 00 00 80 00 13 06 18  >....p...........<
0000e0 e2 02 00 24 10 00 00 00 00 00 00 00 00 d1 00 d0  >...$............<
0000f0 00 00 80 83 0c 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)    0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)    0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)    0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)    0x57(87°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)    0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)    0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)    0x46(70%)   0x8d(byte141)
0x46(70%)   0x76(byte118)    0x46(70%)   0x8e(byte142)
0x46(70%)   0x77(byte119)    0x50(80%)   0x8f(byte143)
0x4b(75%)   0x78(byte120)    0x6e(110%)  0x90(byte144)
```

YoyPa, 2019-06-24:
17G1EMS1.104 can be applied on GS75_8SG/8SF/8SE/9SG/9SF/9SE/9SD but I don't know if it apply same value for every model.
Which one do you have (GS75_8SG/8SF/8SE/9SG/9SF/9SE/9SD) ?

ssergio-ll, 2019-06-25:
Hello. Please, also i have MSI GS75 Stealth 8SE. Can you add support?

Also, i use KDE NEON but i don't know install it. How can i install and use?

Thanks!

YoyPa, 2019-06-25:
@ssergio-ll Take a look at the readme:
- /sys/kernel/debug/ec/ec0/io doesn't exist, is ec_sys module loaded ?
- You need to move isw.conf into /etc.

ssergio-ll, 2019-06-25:
Ok. Now when i execute "isw -cp MSI_ADDRESS_DEFAULT" this show:

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e0 06 7b 0f  >..............{.<
000030 03 01 00 0d 50 0a 05 00 a0 14 60 3b 71 02 c0 00  >....P.....`;q...<
000040 50 0a 3c 00 ff 14 00 00 82 0c 67 3d fd 0b f8 43  >P.<.......g=...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2f 00 37 40 46 4c 52 58  >......../.7@FLRX<
000070 64 2d 2d 32 3c 46 46 46 4b 00 08 03 03 03 03 03  >d--2<FFFK.......<
000080 2e 00 37 3d 41 47 4d 57 5e 00 00 32 3c 46 46 50  >..7=AGMW^..2<FFP<
000090 6e 00 08 03 03 03 03 05 02 00 7d 02 00 64 00 00  >n.........}..d..<
0000a0 31 37 47 31 45 4d 53 31 2e 31 30 34 30 33 31 34  >17G1EMS1.1040314<
0000b0 32 30 31 39 31 33 3a 35 36 3a 32 35 ff 14 00 00  >201913:56:25....<
0000c0 00 07 25 00 00 9e c0 00 00 4a 00 00 00 ab 14 00  >..%......J......<
0000d0 00 00 00 00 70 80 00 00 00 00 00 80 00 13 06 19  >....p...........<
0000e0 e2 02 00 05 10 00 00 00 00 00 00 00 00 9a 00 bc  >................<
0000f0 00 00 80 82 0c 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)    0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)    0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)    0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)    0x57(87°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)    0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)    0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)    0x46(70%)   0x8d(byte141)
0x46(70%)   0x76(byte118)    0x46(70%)   0x8e(byte142)
0x46(70%)   0x77(byte119)    0x50(80%)   0x8f(byte143)
0x4b(75%)   0x78(byte120)    0x6e(110%)  0x90(byte144)

Thanks!

jirislav, 2019-06-28:
> 17G1EMS1.104 can be applied on GS75_8SG/8SF/8SE/9SG/9SF/9SE/9SD but I don't know if it apply same value for every model.
> Which one do you have (GS75_8SG/8SF/8SE/9SG/9SF/9SE/9SD) ?

Hi, I have 9SF.

Btw, I'm pretty sure I've already replied, but GitHub apparently have different opinion :dagger: 

Thank you :)

YoyPa, 2019-07-06:
Seems different models with same EC version number get the same value, I modified how config file work to consider the ec_version and avoid redundancy. https://github.com/YoyPa/isw/commit/2711d42390d8ff3db93d2ecc9663f278b9b6925a https://github.com/YoyPa/isw/commit/b4cca4088469dbbb2624b90ea66b7f4907173e24
Added GS75_8SG/8SF/8SE/9SG/9SF/9SE/9SD with https://github.com/YoyPa/isw/commit/a17abafa90a14c0962f86582315fc89fdcc8f685

closed 2019-07-06 by YoyPa