YoyPa/isw #19: What is option -s S S

ssergio-ll, 2019-07-06:
Hello.

I have a question. What is option "iws -s S S"? What functionality does it have? I don't understand.

Thanks.

YoyPa, 2019-07-06:
From the help message (-h):
```
-s S S      ┬ set a single value into EC
              ├ replace 1st S with ADDRESS in hexadecimal (0x00)
              └ replace 2nd S with VALUE   in decimal     (00)
```
It's mainly for testing purpose, it allow you to replace any value from any address.
For example:
```
$ sudo isw -c

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
[...]
000060 00 00 00 00 00 00 00 00 31 00 3a 3f 44 49 4e 53  >........1.:?DINS<
[...]
```
I I want to replace the value 0x00@0x60 (first of the line 000060) by 128 (0x80) I can type:
```
$ sudo isw -s 0x60 128

Value       set @ address
0x80(128)   0x60(byte96)

$ sudo isw -c         

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
[...]
000060 80 00 00 00 00 00 00 00 31 00 3a 3f 44 49 4e 53  >........1.:?DINS<
[...]

YoyPa, 2019-07-06:
An example of its usefulness -> https://github.com/YoyPa/isw/issues/3#issuecomment-497349034

closed 2019-07-14 by YoyPa