YoyPa/isw #20: Can't make 2 config for a same computer

DadouLapous, 2019-07-09:
If I make 2 config settings for the same computer, the program fails to run : 
``` 
sudo isw -r GL63_8RD
Traceback (most recent call last):
  File "/usr/bin/isw", line 314, in <module>
    main()
  File "/usr/bin/isw", line 311, in main
    parser.parse_args()
  File "/usr/lib/python3.7/argparse.py", line 1749, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1781, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1987, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.7/argparse.py", line 1927, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.7/argparse.py", line 1839, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.7/argparse.py", line 2369, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.7/argparse.py", line 2402, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 98, in ec_read
    cfgp.read_file(cfgfile)
  File "/usr/lib/python3.7/configparser.py", line 717, in read_file
    self._read(f, source)
  File "/usr/lib/python3.7/configparser.py", line 1065, in _read
    lineno)
configparser.DuplicateSectionError: While reading from '/etc/isw.conf' [line 171]: section '16P6EMS1.104_DEFAULT' already exists
```

YoyPa, 2019-07-10:
If you want to make a second config change the **section name**, and the **ec_version key**
For example I replace DEFAULT with YO in my config file:
```
$ cat /etc/isw.conf | grep -i _yo
ec_version = 14A1EMS1.106_YO
[14A1EMS1.106_YO]
```

DadouLapous, 2019-07-10:
Okay thanks, it works

closed 2019-07-10 by DadouLapous