YoyPa/isw #22: Support for GS65 9SG

wallentx, 2019-07-17:
Here is the value of `sudo isw -cp MSI_ADDRESS_DEFAULT`

> EC dump 
>        00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
> 000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
> 000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
> 000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 49 0b  >..............I.<
> 000030 03 03 01 0d 50 0a 05 00 a0 14 60 3b 71 02 80 00  >....P.....`;q...<
> 000040 50 0a 51 00 ec 15 a2 09 a9 11 98 42 da 0b f8 43  >P.Q........B...C<
> 000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
> 000060 00 00 00 00 00 00 00 00 23 00 36 3f 45 4b 51 57  >........#.6?EKQW<
> 000070 64 2e 2e 33 3d 47 4c 51 52 00 08 03 03 03 03 03  >d..3=GLQR.......<
> 000080 24 00 36 3c 40 46 4c 55 5e 01 01 33 3d 47 51 5b  >$.6<@FLU^..3=GQ[<
> 000090 5b 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >[.........d..n..<
> 0000a0 31 36 51 34 45 4d 53 31 2e 31 30 38 30 33 31 35  >16Q4EMS1.1080315<
> 0000b0 32 30 31 39 31 31 3a 32 34 3a 35 38 ec 15 00 00  >201911:24:58....<
> 0000c0 00 07 23 00 00 9e c1 00 01 04 00 ae 00 ae 00 00  >..#.............<
> 0000d0 00 00 40 00 70 80 00 00 05 00 00 00 00 00 00 00  >..@.p...........<
> 0000e0 e2 02 00 23 10 00 00 00 00 00 00 00 00 9a 00 80  >...#............<
> 0000f0 00 00 80 00 8c 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
> 000100
> 
> Profile dump 
> Value       set @ address    Fan mode
> 0x8c(140)   0xf4(byte244)    Advanced
> 
> -----------CPU-----------    -----------GPU-----------
> Value       set @ address    Value       set @ address
> 0x36(54°C)  0x6a(byte106)    0x36(54°C)  0x82(byte130)
> 0x3f(63°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
> 0x45(69°C)  0x6c(byte108)    0x40(64°C)  0x84(byte132)
> 0x4b(75°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
> 0x51(81°C)  0x6e(byte110)    0x4c(76°C)  0x86(byte134)
> 0x57(87°C)  0x6f(byte111)    0x55(85°C)  0x87(byte135)
> 
> 0x2e(46%)   0x72(byte114)    0x1(1%)     0x8a(byte138)
> 0x33(51%)   0x73(byte115)    0x33(51%)   0x8b(byte139)
> 0x3d(61%)   0x74(byte116)    0x3d(61%)   0x8c(byte140)
> 0x47(71%)   0x75(byte117)    0x47(71%)   0x8d(byte141)
> 0x4c(76%)   0x76(byte118)    0x51(81%)   0x8e(byte142)
> 0x51(81%)   0x77(byte119)    0x5b(91%)   0x8f(byte143)
> 0x52(82%)   0x78(byte120)    0x5b(91%)   0x90(byte144)
>

YoyPa, 2019-07-17:
```0000a0 31 36 51 34 45 4d 53 31 2e 31 30 38 30 33 31 35  >16Q4EMS1.1080315<```

Your EC is ```16Q4EMS1.108```, latest version distributed by MSI is ```16Q4EMS1.109``` can you update and resubmit ? I prefer to only support one version (currently 106, I will update to 109).

```download.msi.com/archive/frm_exe/nb/16Q4EMS1.109.zip```
```
$ isw -f 16Q4EMS1.109

Profile 1 dump
-----------CPU---------------    -----------GPU---------------
Value       set @ address        Value       set @ address
0x37(55°C)  0xf801(byte63489)    0x37(55°C)  0xf821(byte63521)
0x40(64°C)  0xf802(byte63490)    0x3d(61°C)  0xf822(byte63522)
0x46(70°C)  0xf803(byte63491)    0x41(65°C)  0xf823(byte63523)
0x4c(76°C)  0xf804(byte63492)    0x47(71°C)  0xf824(byte63524)
0x52(82°C)  0xf805(byte63493)    0x4d(77°C)  0xf825(byte63525)
0x58(88°C)  0xf806(byte63494)    0x56(86°C)  0xf826(byte63526)

0x2d(45%)   0xf80b(byte63499)    0x0(0%)     0xf82b(byte63531)
0x32(50%)   0xf80c(byte63500)    0x32(50%)   0xf82c(byte63532)
0x3c(60%)   0xf80d(byte63501)    0x3c(60%)   0xf82d(byte63533)
0x46(70%)   0xf80e(byte63502)    0x46(70%)   0xf82e(byte63534)
0x4b(75%)   0xf80f(byte63503)    0x50(80%)   0xf82f(byte63535)
0x50(80%)   0xf810(byte63504)    0x5a(90%)   0xf830(byte63536)
0x50(80%)   0xf811(byte63505)    0x5a(90%)   0xf831(byte63537)
```

YoyPa, 2019-07-17:
Updated to 109 with https://github.com/YoyPa/isw/commit/5747f2aecfd0a8eab45786fe8f9a7a753aa0e371

wallentx, 2019-07-17:
I thought my laptop was 8SG and was using/modifying those settings, but when I saw ISW returning errors, I investigated and realized my model is the newer 9SG. I avoided updating the bios due to some known issues mentioned in the Arch wiki, but those should not impact me. I'll upgrade and re-share.

YoyPa, 2019-07-18:
> I thought my laptop was 8SG and was using/modifying those settings, but when I saw ISW returning errors, I investigated and realized my model is the newer 9SG.

ISW doesn't use marketing laptop name since commit https://github.com/YoyPa/isw/commit/5a9c9e2bd94412cfb86a3b0ed09a3192cfaad8ec (3days), it now use motherboard/EC version name. I did that because MSI often reuse motherboard/EC on different laptops.
In your case it's 16Q4EMS1 which is used on: GS65_8SE GS65_8SF GS65_8SG GS65_9SD GS65_9SE GS65_9SF GS65_9SG

> I avoided updating the bios due to some known issues mentioned in the Arch wiki, but those should not impact me. I'll upgrade and re-share.

It's not BIOS but EC, the file is under the firmware section:
https://www.msi.com/Laptop/support/GS65-STEALTH-9SG#down-firmware
I checked 16Q4EMS1.106 108 and 109 and fan profiles doesn't change between version so you can avoid updating if u want.

closed 2019-08-12 by YoyPa

wallentx, 2019-08-20:
Hi, sorry this took so long. I had to get a windows dual boot working to install the EC, and then that killed the speakers on both Arch and Windows :sob:

Anyway, here's the dump with the latest EC.

> EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 5b 0b  >..............[.<
000030 03 01 00 0d 50 0a 05 00 a0 14 60 3b 71 02 c0 00  >....P.....`;q...<
000040 50 0a 51 00 19 15 00 00 e2 10 24 40 d0 0b f8 43  >P.Q.......$@...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 28 00 37 40 46 4c 52 58  >........(.7@FLRX<
000070 64 2d 2d 32 3c 46 4b 50 50 00 08 03 03 03 03 03  >d--2<FKPP.......<
000080 2a 00 37 3d 41 47 4d 56 5e 00 00 32 3c 46 50 5a  >*.7=AGMV^..2<FPZ<
000090 5a 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >Z.........d..n..<
0000a0 31 36 51 34 45 4d 53 31 2e 31 30 39 30 35 31 33  >16Q4EMS1.1090513<
0000b0 32 30 31 39 31 30 3a 34 34 3a 32 39 19 15 00 00  >201910:44:29....<
0000c0 00 07 23 00 00 9e c1 00 00 6a 00 00 00 b0 14 00  >..#......j......<
0000d0 00 00 00 00 70 80 00 00 05 00 00 00 00 13 08 14  >....p...........<
0000e0 e2 02 00 23 10 00 00 00 00 00 00 00 00 9a 00 d0  >...#............<
0000f0 00 00 80 00 0c 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)    0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)    0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)    0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)    0x56(86°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)    0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)    0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)    0x46(70%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)    0x50(80%)   0x8e(byte142)
0x50(80%)   0x77(byte119)    0x5a(90%)   0x8f(byte143)
0x50(80%)   0x78(byte120)    0x5a(90%)   0x90(byte144)

wallentx, 2019-09-03:
@YoyPa when we write to EC, does this only take effect during the running process? I.E. if we were to set a fan profile and write that to EC, are these settings in place when I boot to Windows instead of Linux?

Reason why I ask- When I updated my EC to 16Q4EMS1.109, I lost sound output to my speakers. Only headphone jack works. I'm trying to eliminate possibilities to figure out what happened. I had written changes to the EC where Advanced mode was set, as you pointed out above. I'm just wondering if I messed up somehow by having custom values written to EC when I did the EC upgrade.