YoyPa/isw #24: Support for GE72M (VR) 7RG  Apache Pro

zephir2008, 2019-08-06:
Here is the output of isw -c:
`
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 01 80 c0 06 48 4b  >..............HK<
000030 03 09 01 0d 51 0a 05 00 24 13 6c 2a ea 01 e0 00  >....Q...$.l*....<
000040 22 0b 5e 00 ac 10 00 00 93 0f 47 2f 2e 0c 0c 30  >".^.......G/...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 31 0a 37 3d 43 49 4f 55  >........1.7=CIOU<
000070 64 30 28 30 38 42 4c 56 5f 00 0a 08 08 08 08 08  >d0(08BLV_.......<
000080 34 00 37 3c 41 46 4b 4b 61 36 00 36 3e 46 4e 56  >4.7<AFKKa6.6>FNV<
000090 56 00 0a 07 07 07 07 07 82 16 7d 02 16 66 00 00  >V.........}..f..<
0000a0 31 37 39 43 45 4d 53 31 2e 31 30 36 30 32 32 30  >179CEMS1.1060220<
0000b0 32 30 31 37 31 36 3a 31 31 3a 35 36 a1 10 00 00  >201716:11:56....<
0000c0 00 01 24 00 00 00 00 00 00 62 00 51 00 50 00 00  >..$......b.Q.P..<
0000d0 00 00 20 00 7f 00 01 00 00 00 00 80 01 00 00 00  >.. .............<
0000e0 e2 02 00 81 10 00 00 00 00 00 00 00 00 d1 00 00  >................<
0000f0 00 00 80 01 0c 00 0c 00 00 00 00 00 00 00 00 00  >................<
000100
`

zephir2008, 2019-08-06:
Nice work and good luck! :)

YoyPa, 2019-08-11:
Hello,
can you provide the output of this command instead ? It make it a bit faster for me :).
```isw -cp MSI_ADDRESS_DEFAULT```

zephir2008, 2019-08-12:
Hi,
Here it is!
Thanks a lot!

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 48 4b  >..............HK<
000030 02 05 00 0d 51 0a 05 00 24 13 6c 2a ea 01 c0 00  >....Q...$.l*....<
000040 22 0b 5a 00 0e 0c c9 f0 ca 0a 67 2d b5 0b 0c 30  >".Z.......g-...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2b 0a 37 3d 43 49 4f 55  >........+.7=CIOU<
000070 64 28 28 30 38 42 4c 56 5f 00 0a 08 08 08 08 08  >d((08BLV_.......<
000080 2e 00 37 3c 41 46 4b 4b 61 00 00 36 3e 46 4e 56  >..7<AFKKa..6>FNV<
000090 56 00 0a 07 07 07 07 07 02 16 7d 02 16 66 00 00  >V.........}..f..<
0000a0 31 37 39 43 45 4d 53 31 2e 31 30 36 30 32 32 30  >179CEMS1.1060220<
0000b0 32 30 31 37 31 36 3a 31 31 3a 35 36 0e 0c 00 00  >201716:11:56....<
0000c0 00 01 25 00 00 00 00 00 00 02 00 00 00 cb 00 00  >..%.............<
0000d0 00 00 00 00 7f 01 01 00 00 00 00 80 01 00 00 00  >................<
0000e0 e2 02 00 81 10 00 00 00 00 00 00 00 00 82 00 00  >................<
0000f0 00 00 80 01 0c 00 0c 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x37(55°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)    0x4b(75°C)  0x87(byte135)

0x28(40%)   0x72(byte114)    0x0(0%)     0x8a(byte138)
0x30(48%)   0x73(byte115)    0x36(54%)   0x8b(byte139)
0x38(56%)   0x74(byte116)    0x3e(62%)   0x8c(byte140)
0x42(66%)   0x75(byte117)    0x46(70%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)    0x4e(78%)   0x8e(byte142)
0x56(86%)   0x77(byte119)    0x56(86%)   0x8f(byte143)
0x5f(95%)   0x78(byte120)    0x56(86%)   0x90(byte144)
```

YoyPa, 2019-10-08:
added with https://github.com/YoyPa/isw/commit/ca64bf83c7f0382310a6da4874e33de404419a87

closed 2019-10-08 by YoyPa