YoyPa/isw #25: Support for "Shift"?

jirislav, 2019-08-11:
Hello,

I really like that I can control fan speed according to the actual CPU/GPU temperature. But it would be totally awesome to have the ability to also control the "Shift", as shown in this snap from the Dragon Control Center:
[image]

You can choose from several Shift modes:
- Turbo
- Sport
- Comfort
- ECO (I think this one restricts the power of CPU/GPU so that it doesn't have to cool at all, which is really cool if you need to be quiet on a meeting)

[image]

---

Is there any way that I could help you with this? I must admit I don't have experience with tracing system calls on Windows, but I'm willing to learn this. Could you point me in some direction of how to reverse-engineer the communication with the EC?

YoyPa, 2019-08-12:
NB:
```
Sport Mode: 
    Overclocking of both CPU and GPU             <- I don't know if it's EC related (low hope)
    activates Cooler Boost                       <- Already supported
Comfort Mode:
    CPU turbo is enabled                         <- I don't know if it's EC related (low hope)
    GPU boost frequencies are prohibited.        <- I don't know if it's EC related (low hope)
Green Mode:
    CPU turbo disabled                           <- I don't know if it's EC related (low hope)
    lower frequencies/voltage on CPU and GPU     <- I don't know if it's EC related (low hope)
    Lower revving fans.                          <- ...
```
ECO and GREEN seems to be related. I don't really know what TURBO mode is, seems to be a manual overclock.
Everything related to cpu frequency/voltage is probably related to intel XTU kind of software while GPU will probably be related to MSI afterburner behavior.

PS: ISW only mess with EC

> how to reverse-engineer the communication with the EC

You can watch EC in realtime with RWEverything http://rweverything.com/screenshots/
And you have the wiki for a quick, lacking info and outdated explanation of some EC values.

jirislav, 2019-08-12:
> ECO and GREEN seems to be related. I don't really know what TURBO mode is, seems to be a manual overclock.

Yes, Turbo mode let's you overclock manually both CPU & GPU.

> Everything related to cpu frequency/voltage is probably related to intel XTU kind of software while GPU will probably be related to MSI afterburner behavior.

I noticed that when I change the mode, DragonCenter launches program called `gpuControl.exe`, but couldn't find out more about what it does.

> PS: ISW only mess with EC
> 
> > how to reverse-engineer the communication with the EC
> 
> You can watch EC in realtime with RWEverything http://rweverything.com/screenshots/
> And you have the wiki for a quick, lacking info and outdated explanation of some EC values.

I'll definitely check it out and report here in the case that the mode is EC related.

Thank you for such a quick response!

jirislav, 2019-08-12:
Okay, I've been playing with the Shift a little bit and I have some great news! All the settings seems to be EC related.

The Shift seems to live in `F0x02` byte. I've been waiting a while for it to change, but really, nothing can change it except of switching the Shift. See yourself:

### Sport shift
[image]

### Comfort shift
[image]

### ECO shift
[image]

### Turbo shift (with defaults)
[image]

----

## Bonus

I've also found out that you can change the battery settings as shown on this snip from the Dragon Center:
[image]

The battery settings "lives" in `E0x0F` byte. Changing battery settings is also the only way to change this value.

### Best for mobility
[image]

### Balanced
[image]

### Best for battery
[image]

---

I think that not all versions of EC support this, so it would be wise to implement such a functionality only for certain EC versions. However, it would be ultra cool to have the ability to control not only fan speed, but also the performance itself using built-in EC registers, so I'm really looking forward to see this feature implemented! Would the implementation be complex?

If you want, I could help and send a pull request? I have ~3 years of experience with Python in corporate. But I can't say when I'm going to have the time for this.

jirislav, 2019-08-12:
If you wanted to investigate further on the values, here I'm adding zipped file of the measurements.

I've measured each shift several times so that it is possible to extract static registers, which would hold the value of some sort of settings (time invariant bytes).

[dragon_control_EC_dumps.zip](https://github.com/YoyPa/isw/files/3494107/dragon_control_EC_dumps.zip)

jirislav, 2019-08-16:
I've played with the implementation a little bit and have achieved setting up both Shift & Battery modes! :)

Please see each commit of [the PR](https://github.com/YoyPa/isw/pull/28) separately to check the differences made:
- first commit just reformat whole script according to the PEP8 standard
- second commit adds features such as Batttery & Shift modes settings and does basic code refactor

DadouLapous, 2019-08-23:
Hi,

My laptop (GL63-8RD) uses exactly the same values and the same addresses with dragon center on windows.

Unfortunately, on linux, changing theses values is not enough to change shift, for example, 
`sudo isw -s 0xf2 194 ` ((int)194 = (hex)c2 = ECO shift)  
still lets my cpu go to 4Ghz...

On windows changing the value of the 0xf2 address is enough to change the shift profile with or without DC launched ( I did not tried with DC not installed) but since DC is launching on startup there might be another settings involved. 
I definitely have to try to change theses values without DC installed to check if it still works.

I could not get the battery settings to work on windows even with a fresh install of dragoncenter (EC values are changing when dragoncenter applies a profile, but my computer still charging above thersholds ...), so i suppose there is more than EC involved  for the battery settings ?

Is changing the 0xf2 value on your laptop enough to apply a shift profile on linux for you @jirislav ?

jirislav, 2019-08-23:
> Is changing the `0xf2` value on your laptop enough to apply a shift profile on linux for you @jirislav ?

Well, after thorough examination, you are right, even my laptop doesn't respond to the EC configuration :/ .. at first, it seemed like it worked, because he got more quiet, but that could be just some coincidence :sob:

DadouLapous, 2019-08-23:
Well, when you quit / run DC, à bunch of EC values gets modified, and I just tested, on windows without DC, setting `0xf2` is not enough to change shift mode, so there is definitely something whe're missing.

ssergio-ll, 2019-11-19:
> battery settings as shown on this snip from the Dragon Center:

Hello. Have you can run the different modes of battery? I am looking for the same. I have MSI GS75 8SE. Thanks.