YoyPa/isw #26: Service not starting

BruynoogheMatthias, 2019-08-11:
I tried to install the service to start it automatically on startup but when I try to start it it gives me an error.
This is the output of 

` systemctl status isw@16J9EMS1.service 
`

`● isw@16J9EMS1.service - ISW fan control service
   Loaded: loaded (/usr/lib/systemd/system/isw@.service; indirect; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-08-11 17:04:34 CEST; 12s ago
  Process: 20463 ExecStart=/usr/bin/isw -w 16J9EMS1 (code=exited, status=2)
 Main PID: 20463 (code=exited, status=2)

Aug 11 17:04:33 Pheonix systemd[1]: Starting ISW fan control service...
Aug 11 17:04:34 Pheonix isw[20463]: usage: isw [-h] [-b B B] [-c] [-f F] [-p P] [-r R] [-s S S] [-w W]
Aug 11 17:04:34 Pheonix isw[20463]: isw: error: argument -w: invalid ec_write value: '16J9EMS1'
Aug 11 17:04:34 Pheonix isw[20463]: Writing config to EC...
Aug 11 17:04:34 Pheonix systemd[1]: isw@16J9EMS1.service: Main process exited, code=exited, status=2/INV
Aug 11 17:04:34 Pheonix systemd[1]: isw@16J9EMS1.service: Failed with result 'exit-code'.
Aug 11 17:04:34 Pheonix systemd[1]: Failed to start ISW fan control service.
`

YoyPa, 2019-08-12:
Which version are your using ? Git or AUR ?
Does isw -w 16J9EMS1 work outside of systemd ?

BruynoogheMatthias, 2019-08-12:
I'm using the git version, the command works outside of systemd.

divineo, 2020-01-10:
I've got the same issue, and there are two services related to it. When I check it in systemctl.

> 
> isw@x5b16JBEMS1x5d.service
> isw@\x5b16JBEMS1\x5d.service

Although it's not a big of hassle to write `isw -w 16JBEMS1` every time on startup or after sleep. But it'll be great if it starts up on its own.

```
sudo systemctl status isw@\x5b16JBEMS1\x5d.service
● isw@x5b16JBEMS1x5d.service - ISW fan control service
   Loaded: loaded (/usr/lib/systemd/system/isw@.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since Fri 2020-01-10 14:54:16 IST; 14min ago
 Main PID: 809 (code=exited, status=1/FAILURE)

Jan 10 14:54:16 killi-manjaro isw[809]:   File "/usr/bin/isw", line 290, in ec_write
Jan 10 14:54:16 killi-manjaro isw[809]:     ap = cfgp.get(value, 'address_profile')
Jan 10 14:54:16 killi-manjaro isw[809]:   File "/usr/lib/python3.8/configparser.py", line 781, in get
Jan 10 14:54:16 killi-manjaro isw[809]:     d = self._unify_values(section, vars)
Jan 10 14:54:16 killi-manjaro isw[809]:   File "/usr/lib/python3.8/configparser.py", line 1149, in _unify_values
Jan 10 14:54:16 killi-manjaro isw[809]:     raise NoSectionError(section) from None
Jan 10 14:54:16 killi-manjaro isw[809]: configparser.NoSectionError: No section: 'x5b16JBEMS1x5d'
Jan 10 14:54:16 killi-manjaro systemd[1]: isw@x5b16JBEMS1x5d.service: Main process exited, code=exited, status=1/FAILURE
Jan 10 14:54:16 killi-manjaro systemd[1]: isw@x5b16JBEMS1x5d.service: Failed with result 'exit-code'.
Jan 10 14:54:16 killi-manjaro systemd[1]: Failed to start ISW fan control service.
```

divineo, 2020-01-10:
```
sudo systemctl status isw@16JBEMS1.service
● isw@16JBEMS1.service - ISW fan control service
   Loaded: loaded (/usr/lib/systemd/system/isw@.service; enabled; vendor preset: disabled)
   Active: inactive (dead) since Fri 2020-01-10 15:36:03 IST; 25s ago
  Process: 3339 ExecStartPre=/usr/bin/sleep 2 (code=exited, status=0/SUCCESS)
  Process: 3340 ExecStart=/usr/bin/isw -w 16JBEMS1 (code=exited, status=0/SUCCESS)
 Main PID: 3340 (code=exited, status=0/SUCCESS)

Jan 10 15:36:03 killi-manjaro isw[3340]: [1B blob data]
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x5(5%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x5(5%)     0x73(byte115)        0x36(54%)   0x8b(byte139)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x5(5%)     0x74(byte116)        0x3e(62%)   0x8c(byte140)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x42(66%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x4c(76%)   0x76(byte118)        0x4e(78%)   0x8e(byte142)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x56(86%)   0x77(byte119)        0x56(86%)   0x8f(byte143)
Jan 10 15:36:03 killi-manjaro isw[3340]: 0x56(86%)   0x78(byte120)        0x56(86%)   0x90(byte144)
Jan 10 15:36:03 killi-manjaro systemd[1]: isw@16JBEMS1.service: Succeeded.
Jan 10 15:36:03 killi-manjaro systemd[1]: Started ISW fan control service.

But, after sleep or reboot it still does not work.
```

divineo, 2020-01-10:
Never mind it works now. I don't know what changed but it works perfectly fine now

YoyPa, 2020-01-10:
@divineo You've made two services:
```
isw@16JBEMS1.service
isw@[16JBEMS1].service # this one is not correct, you should not use square bracket "[]"
```
So you should ```systemctl disable isw@[16JBEMS1].service```

PS: This is not the same issue :-1:.

divineo, 2020-01-11:
Yeah, I figured that out right after I commented. Thanks though

AnonJose, 2020-05-06:
Hey, I am also facing the same issue, I am using the AUR package
...
sudo systemctl status isw@166Q4EM1.service
● isw@166Q4EM1.service - ISW fan control service
     Loaded: loaded (/usr/lib/systemd/system/isw@.service; enabled; vendor preset: disabled)
     Active: inactive (dead)
...

Any idea on how to fix this. I am able  use isw -r outside of systemmd.