YoyPa/isw #27: Extra functionality (keybinding?)

66darkmatter99, 2019-08-16:
Hi there and thanks so much for isw, its working great so far. That being said do you know how to possibly keybind the coolerbooster to a special key? ie (Fn+F7 in my case as that is the unused turbo button). Im using gnome and ive been searching through gnome tweaks for keybinding and online for help and im just not finding how to assign these.

Also have you considered creating a gnome shell extension for something of the like so users can control the cooler booster or switch between profiles and the click of a button? 

Just a suggestion that i think would be pretty cool.
Thanks again for any help.

YoyPa, 2019-10-22:
Hello and sorry for the late response,
ISW need administrative rights to write into EC so for coolerboost you can try to bind a keyboard shortcut to ```pkexec isw -b on``` (latest git version), pkexec will show a password prompt (not ideal).

@mibofra mentionned this kde widget today: https://github.com/jsalatas/plasma-pstate/issues/35

I've seen nothing on gnome and don't plan to do something about it as I don't use gnome. I'm also not a programmer, I've very low knowledge about writing something working outside of the terminal.

66darkmatter99, 2019-10-23:
Not a problem and thanks for taking the time to get back to me. Thanks for the tips on admin privileges as Ive been having quite the time trying to figure out how to bypass that. Ill check out pkexec and plasma-pstate soon. Looking forward to your next release.

closed 2019-11-18 by YoyPa