YoyPa/isw #32: No difference between fan speed 01...55

akoidan, 2019-08-29:
It seems like:
 - cpu_fan_speed is off if I set the value to `0`
 - cpu_fan_speed becomes a bit annoying when I set the value to `01`
 - there's no difference in noise between `01`, `02`, ... ~`60`. (So I guess the fan speed as well)
 - thus I cant set the fans to low speed, it's either off or it's like 30 db which is irritating 
 - After cpu_fan_speed 60+ it seems the speed is increasing, but 60 is already loud

Is there any way to set the fans to low speed?

kmBlaine, 2019-09-17:
See comments header in [isw.conf](https://github.com/YoyPa/isw/blob/master/etc/isw.conf). There is a certain minimum speed that the fan controller will use instead of any non-zero value which is less than it. As I understand, what you are seeing is expected behavior and appears to be a limitation in the hardware itself (frustrating I know).

YoyPa, 2019-10-08:
@akoidan I have the exact same laptop and I started this to mitigate this annoyance.
I don't know how to affect the minimum rpm speed and as @kmBlaine said: It's told in isw.conf comment.
You can set higher temp threshold so it stay at 0 longer, and also undervold the cpu with:
https://github.com/hedgepigdaniel/linux-intel-undervolt-tool/
OR
https://github.com/erpalma/throttled
I achieved -178mV with mine.

akoidan, 2019-10-08:
> @akoidan I have the exact same laptop and I started this to mitigate this annoyance.
> I don't know how to affect the minimum rpm speed and as @kmBlaine said: It's told in isw.conf comment.
> You can [...]

Yeah I undervolted it already, i went with -150mv, as folks suggested for 6700hq. Passive cooling really sucks in my case, the temperature just goes up and never stops until fans go on. 

p.s. BTW I don't know why but I tend not to trust on soft that may be non easily revertible on non Mac/Windows systems.

closed 2019-10-22 by YoyPa