YoyPa/isw #34: OSError: [Errno 22] Invalid argument

xlith, 2019-09-24:
Hi,
I don't know if I am doing it right. 
I am trying to start using application.
When I write into the terminal:

`sudo isw -b GS65_8RF on`

It gives an error:

```
Value       set @ address
0x80(128)   0x98(byte152)
Traceback (most recent call last):
  File "/usr/bin/isw", line 314, in <module>
    main()
  File "/usr/bin/isw", line 311, in main
    parser.parse_args()
  File "/usr/lib/python3.7/argparse.py", line 1749, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1781, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1987, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.7/argparse.py", line 1927, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.7/argparse.py", line 1839, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.7/argparse.py", line 2387, in _get_values
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.7/argparse.py", line 2387, in <listcomp>
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.7/argparse.py", line 2402, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 263, in ec_boost
    print('Error: Only \033[4mon\033[0m and \033[4moff\033[0m are valid.')
OSError: [Errno 22] Invalid argument
```

66darkmatter99, 2019-09-24:
Don't forget to do this first: 

Builtin or not builtin ?
It need ec_sys module with option write_support=1, there are two scenario to set that:
ec_sys is a builtin kernel module:
add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
then update your grub and reboot.
ec_sys is not a builtin kernel module:
copy both isw-ec_sys.conf files provided (/etc/mod[...]) with same path (Arch AUR package will do it for you).
then reboot OR modprobe ec_sys write_support=1.

he58, 2019-09-25:
i gave up on running this a while ago. I have the exact same error as the guy posted above. Kernel 5.2.11-1-MANJARO which i believe doesn't have ec_sys built in. I installed via AUR which means i should have the isw-ec_sys.conf files in the correct place.

YoyPa, 2019-10-08:
@xlith Can you give me the output of:
```
cat /sys/module/ec_sys/parameters/write_support
```
> Kernel 5.2.11-1-MANJARO which i believe doesn't have ec_sys built in.

@he58 It's builtin, you have to edit /etc/default/grub, update grub and reboot.
```
$ uname -a
Linux yo-gs 5.3.4-1-MANJARO
$ modprobe -r ec_sys
modprobe: FATAL: Module ec_sys is builtin.
```

YoyPa, 2019-11-18:
Closing old issue, open a new if problem still exist.

closed 2019-11-18 by YoyPa