YoyPa/isw #35: GE75-9SG

thebeerbaron2018, 2019-09-27:
Hi there.  Can you assist me with fan control on my brand new GE75-9SG.  I have checked online and 1.108 is the latest EC update for this machine as far as I can find.

```EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 5b 0b  >..............[.<
000030 03 09 01 0d 51 0a 05 00 c0 17 30 2a 60 02 e7 00  >....Q.....0*`...<
000040 00 00 64 00 6a 17 00 00 6a 17 78 30 0f 0c a2 30  >..d.j...j.x0...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 31 0a 32 38 3e 46 4b 50  >........1.28>FKP<
000070 64 28 00 28 32 3c 46 50 5a 00 0a 07 07 09 09 09  >d(.(2<FPZ.......<
000080 32 36 37 3c 41 46 4b 50 61 2d 00 2d 34 3b 45 4f  >267<AFKPa-.-4;EO<
000090 59 00 0a 07 07 07 07 07 02 16 7d 02 16 66 00 00  >Y.........}..f..<
0000a0 31 37 45 32 45 4d 53 31 2e 31 30 38 30 34 30 32  >17E2EMS1.1080402<
0000b0 32 30 31 39 31 36 3a 31 37 3a 31 35 64 17 00 40  >201916:17:15d..@<
0000c0 00 06 22 00 00 9e c0 00 00 1c 00 b2 00 c9 14 00  >..".............<
0000d0 00 00 10 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 31 10 00 00 00 00 00 00 00 00 9a 00 e4  >...1............<
0000f0 00 00 82 00 0c 8f 05 80 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address    Fan mode
0xc(12)     0xf4(byte244)    Auto

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x32(50°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x38(56°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
0x3e(62°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
0x4b(75°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
0x50(80°C)  0x6f(byte111)    0x50(80°C)  0x87(byte135)

0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
0x28(40%)   0x73(byte115)    0x2d(45%)   0x8b(byte139)
0x32(50%)   0x74(byte116)    0x34(52%)   0x8c(byte140)
0x3c(60%)   0x75(byte117)    0x3b(59%)   0x8d(byte141)
0x46(70%)   0x76(byte118)    0x45(69%)   0x8e(byte142)
0x50(80%)   0x77(byte119)    0x4f(79%)   0x8f(byte143)
0x5a(90%)   0x78(byte120)    0x59(89%)   0x90(byte144)```

thebeerbaron2018, 2019-09-27:
Great work on the script by the way!  Another question.... this model has LED's in the USB ports - they light up red under Windows - but I can't get them to light up under Linux either.  Wonder if they may be controlled in the same way?  I have a test Windows setup if you need me to run software to try and sniff the commands?

thebeerbaron2018, 2019-09-29:
OK after a bit of probing the USB's can be accessed at 0xF7 writing 0x80 for off, 0xC1 for half brightness and 0x81 for full brightness

thebeerbaron2018, 2019-09-29:
It seems the GPU fan speeds are at 0x8a - 0x8f  (not like 0x90 in the other configs) and can be set from 0 to 150 but the fans don't start spinning until the value is 20 or above

thebeerbaron2018, 2019-09-29:
CPU fan speeds are at 0x72 - 0x77 (not like 0x78 in the other configs) and can also be set from 0 to 150 and also doesn't start spinning until value is 20 or above.

thebeerbaron2018, 2019-10-02:
More digging... one fan speed (set by 0x71) seems to be at 0xCA and 0xCB and ranges from the values 325 (1480rpm) down to 90 (5300rpm) - yes - it's reversed.

3820 rpms in this range, 239 steps in this range therefore each step = 16 rpms

So using the following pseudo code gets the fan speed:

if value > 325 then value = 325
fanspeed = ((325-value) * 16) + 1480

The other fan is at 0x89 and can be monitored in the same way using 0xCC and 0xCD

Hope this helps

thebeerbaron2018, 2019-10-04:
So I wrote this, it turns on the USB lights, sets the fan profiles to silent (off) until over 50 degrees or whatever step 3 is in the profile.  Activates the advanced fan profile and sets the CPU to performance mode.  It then polls the fan registers every two seconds and reports the correct fan speeds.  Please make sure the script is run as root.

```#!/usr/bin/env python
import os
import sys
import time
EC_IO_FILE="/sys/kernel/debug/ec/ec0/io"
if not os.path.exists(EC_IO_FILE):
        os.system("sudo modprobe ec_sys write_support=1")
def ec_write(addr,value):
    with open(EC_IO_FILE,"wb") as f:
        f.seek(addr)
        f.write(bytearray([value]))
def ec_read(addr):
    with open(EC_IO_FILE,"rb") as f:
        f.seek(addr)
        return int(f.read(2).hex(),16)

#usb lights on
ec_write(0xf7, 0x81)

#fans Advanced
ec_write(0xF4,0x8C)

#fan speeds
ec_write(0x71, 0)
ec_write(0x72, 0)
ec_write(0x73, 0)
ec_write(0x74, 25)
ec_write(0x75, 50)
ec_write(0x76, 75)
ec_write(0x77, 100)

ec_write(0x8a, 0)
ec_write(0x8b, 0)
ec_write(0x8c, 25)
ec_write(0x8d, 50)
ec_write(0x8e, 75)
ec_write(0x8f, 100)
ec_write(0x90, 100)

#performance CPU
ec_write(0xf2, 0xc4)
#powersave CPU
#ec_write(0xf2, 0xc2)

while True:
	CPUspeed=ec_read(0xCC)
	if CPUspeed>325:
		CPUspeed=325
	if CPUspeed==0:
		cpufs=0
	else:
		cpufs=((325-CPUspeed)*16)+1480	

	GPUspeed=ec_read(0xCA)
	if GPUspeed>325:
		GPUspeed=325
	if GPUspeed==0:
		gpufs=0
	else:
		gpufs=((325-GPUspeed)*16)+1480	

	print('CPU fan : %s rpm  GPU fan : %s rpm' % (str(cpufs),str(gpufs)))
	time.sleep(2)
	```

YoyPa, 2019-10-08:
That's a really nice finding <3.
I changed the RPM formula to fit both your and my laptop:
y=480000/x
y being rpm & x being the value in EC(decimal)
You can check the graph by typing the formula here: https://www.desmos.com/calculator

Example:
```
             result   expected
480000/90  = 5333     5300      your max
480000/94  = 5106     5100      my max
480000/133 = 3609     3600      my min
480000/325 = 1477     1480      your min
```
So we stay off by less than 50 rpm which seems ok while working with multiple laptops (if not all).

thebeerbaron2018, 2019-10-08:
I added a bit of extra code which now makes my laptop awesome!  It checks CPU use and switches to performance mode while under load and when the load drops - switches back to cool and quiet mode.  

```#!/usr/bin/env python
import os
import sys
import time
import psutil
EC_IO_FILE="/sys/kernel/debug/ec/ec0/io"
if not os.path.exists(EC_IO_FILE):
        os.system("sudo modprobe ec_sys write_support=1")
def ec_write(addr,value):
    with open(EC_IO_FILE,"wb") as f:
        f.seek(addr)
        f.write(bytearray([value]))
def ec_read(addr):
    with open(EC_IO_FILE,"rb") as f:
        f.seek(addr)
        return int(f.read(2).hex(),16)

#usb lights on
ec_write(0xf7, 0x81)

#fans Advanced
ec_write(0xF4,0x8C)

ec_write(0x71, 0)
ec_write(0x72, 0)
ec_write(0x73, 0)
ec_write(0x74, 0)
ec_write(0x75, 33)
ec_write(0x76, 66)
ec_write(0x77, 100)

ec_write(0x89, 0)
ec_write(0x8a, 0)
ec_write(0x8b, 0)
ec_write(0x8c, 25)
ec_write(0x8d, 50)
ec_write(0x8e, 75)
ec_write(0x8f, 100)

#performance
#ec_write(0xf2, 0xc4)

#powersave
ec_write(0xf2, 0xc2)
cpumode='powersave'
change=0

while True:
	CPUspeed=ec_read(0xCC)
	if CPUspeed>325:
		CPUspeed=325
	if CPUspeed==0:
		cpufs=0
	else:
		cpufs=((325-CPUspeed)*16)+1480	

	GPUspeed=ec_read(0xCA)
	if GPUspeed>325:
		GPUspeed=325
	if GPUspeed==0:
		gpufs=0
	else:
		gpufs=((325-GPUspeed)*16)+1480	

	for cpuloop in range (5):
		cpu=float(psutil.cpu_percent())
		if cpu>25:
			if cpumode=='powersave':
				#performance
				ec_write(0xf2, 0xc4)
				cpumode='performance'
				change=5
		else:
			if cpumode=='performance' and change>0:
				change-=1
				if change==0:
					#powersave
					ec_write(0xf2, 0xc2)
					cpumode='powersave'
		print('\033[FCPU fan : %s rpm  GPU fan : %s rpm  CPUuse : %s (%s) Change %s       ' % (str(cpufs),str(gpufs),str(cpu),cpumode,str(change)))
		time.sleep(2)
	print("")

thebeerbaron2018, 2019-10-08:
As for fan RPM's since they are different between laptops it might be worth including the min and max values of the fan registers in each profile - then it's just some simple math to do the conversion.  (Still loving your work though - I would never have got this far without it - but I was determined never to go back to Windows!)

YoyPa, 2019-10-22:
A weird thing about your laptop, when I look inside 17E2EMS1.108 (with option -f) from:
http://download.msi.com/archive/frm_exe/nb/17E2EMS1.108.zip

The majority of laptops use the first profile.
WS65_8SK is using the second one.
Old laptops like GT60 use the third one.
But this one use the 4th for CPU and the 1st for GPU :trollface:.

PS: I have no idea why there is random number (1 to 4) of profile inside those files.

YoyPa, 2019-10-22:
added with https://github.com/YoyPa/isw/commit/693d18221819c9799a6e9dcfc2c0ef6218223816

YoyPa, 2019-10-24:
usb backlight added with https://github.com/YoyPa/isw/commit/f4d0f484ddc165ab75cb8d36b38709988a1b25e5

YoyPa, 2019-10-24:
>  It checks CPU use and switches to performance mode while under load and when the load drops - switches back to cool and quiet mode.

I benched with Eco, Comfort and Sport mode and I see no difference at all (Unigine superposition):
Nothing = 4493
Eco = 4504
Comfort = 4505
Sport = 4504

And when using ```stress -c 1``` I'm able to turbo @3200MHz even in Eco mode. :trollface:

>  it might be worth including the min and max values of the fan registers in each profile

I only have one laptop I'm too lazy to ask for every motherboard already "supported". :sob:
*Good enough mode ON*

closed 2019-10-24 by YoyPa