YoyPa/isw #37: Support for MSI GE62-2QF

StormPooper, 2019-10-07:
First off, thanks for creating isw, so cool to see support for this kind of thing.

I've installed isw from the AUR in Arch and followed the setup steps in the README, but when I run `isw -cp MSI_ADDRESS_DEFAULT` I get an error for the profile dump section:
```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
*
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 0b 4b  >...............K<
000030 03 09 01 05 51 0a 05 00 26 13 6c 2a ea 01 e0 00  >....Q...&.l*....<
000040 00 00 64 00 59 10 00 00 59 10 e0 2f e1 0b 0c 30  >..d.Y...Y../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 32 3b 3c 41 46 4b 50 50  >........2;<AFKPP<
000070 64 32 00 32 38 42 4c 56 56 00 0f 08 08 08 08 08  >d2.28BLVV.......<
000080 2e 36 37 3c 41 46 4b 4b 5e 00 00 35 3c 44 4c 54  >.67<AFKK^..5<DLT<
000090 54 00 0a 07 07 07 07 07 02 16 64 02 16 5e 01 00  >T.........d..^..<
0000a0 31 36 4a 31 45 4d 53 31 20 56 65 72 35 2e 30 36  >16J1EMS1 Ver5.06<
0000b0 2c 20 30 37 2f 30 33 2f 32 30 31 35 59 10 00 50  >, 07/03/2015Y..P<
0000c0 00 01 00 00 00 9e c0 00 00 32 00 00 00 9e 14 00  >.........2......<
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00  >....p...........<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 80 00 00  >................<
0000f0 00 00 81 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address    Fan mode
Traceback (most recent call last):
  File "/usr/bin/isw", line 314, in <module>
    main()
  File "/usr/bin/isw", line 311, in main
    parser.parse_args()
  File "/usr/lib/python3.7/argparse.py", line 1749, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1781, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.7/argparse.py", line 1987, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.7/argparse.py", line 1927, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.7/argparse.py", line 1839, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.7/argparse.py", line 2369, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.7/argparse.py", line 2402, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 59, in ec_profile
    sfm+'\n'
UnboundLocalError: local variable 'sfm' referenced before assignment
```

Any help would be greatly appreciated.

thebeerbaron2018, 2019-10-08:
This is due to the code not managing to work out your fan profile....  I would suggest go back into Windows and use the Dragon Center to set the fan profile to "auto" and then come back to Linux and re-run the script.

```
		fm = int(file.read(1).hex(),16)
		if fm==int(140):
			sfm='Advanced'
		elif fm==int(76):
			sfm='Basic'
		elif fm==int(12):
			sfm='Auto'
```

The EC retains the values from whatever is set in Windows.  You might want to look at my script on issue #35  as I think all the GE/GS machines use the same registers.

YoyPa, 2019-10-08:
> This is due to the code not managing to work out your fan profile...
> ```
> 		fm = int(file.read(1).hex(),16)
> 		if fm==int(140):
> 			sfm='Advanced'
> 		elif fm==int(76):
> 			sfm='Basic'
> 		elif fm==int(12):
> 			sfm='Auto'
> ```
Yep, I added a "else" for unknown fan_mode with https://github.com/YoyPa/isw/commit/96dd5ee60f193e8bfed5c3a6f269b85cc4de5d78
I will make a new release when I've finished looking at every issue :sob:.
NB: config file structure changed btw :trollface:.

StormPooper, 2019-10-08:
Interesting, the modes I  get aren't in that list at all, which is probably why I'm getting that error then (the latest Dragon Gaming Center for my laptop is quite old):
[image]
I've tried all 3 but no difference. I've also tried MSI's [Silent Option](https://forum-en.msi.com/index.php?topic=255972.0) utility, but those profiles only appear to be an option on the Skylake version, whereas I'm running on a Haswell. Interestingly, Silent Option gives me these temperature settings by default when switching to 'Advanced' mode:
```
FanCPUT1Percentage:0
FanCPUT2Percentage:50
FanCPUT3Percentage:56
FanCPUT4Percentage:66
FanCPUT5Percentage:76
FanCPUT6Percentage:86
FanVGAT1Percentage:0
FanVGAT2Percentage:53
FanVGAT3Percentage:60
FanVGAT4Percentage:68
FanVGAT5Percentage:76
FanVGAT6Percentage:84
```
I'll try copying the settings from 16J9EMS1 (closest to my firmware ID from the list) and mapping these speeds to it, see how it goes.

StormPooper, 2019-10-08:
OK, an update: now I've switched to 'Advanced' in Silent Option the Profile dump works:
```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
*
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 0b 4b  >...............K<
000030 03 09 01 05 51 0a 05 00 26 13 6c 2a ea 01 e0 00  >....Q...&.l*....<
000040 00 00 63 00 59 10 00 00 2e 10 cd 2f df 0b 0c 30  >..c.Y....../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 4b 3b 3c 41 46 4b 50 50  >........K;<AFKPP<
000070 64 42 00 32 38 42 4c 56 56 00 0f 08 08 08 08 08  >dB.28BLVV.......<
000080 30 36 37 3c 41 46 4b 4b 5e 00 00 35 3c 44 4c 54  >067<AFKK^..5<DLT<
000090 56 00 0a 07 07 07 07 07 02 16 64 02 16 5e 01 00  >V.........d..^..<
0000a0 31 36 4a 31 45 4d 53 31 20 56 65 72 35 2e 30 36  >16J1EMS1 Ver5.06<
0000b0 2c 20 30 37 2f 30 33 2f 32 30 31 35 59 10 00 50  >, 07/03/2015Y..P<
0000c0 00 01 25 00 00 9e c0 00 00 64 00 00 00 93 14 00  >..%......d......<
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00  >....p...........<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 80 00 00  >................<
0000f0 00 00 80 00 8c 00 00 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address    Fan mode
0x8c(140)   0xf4(byte244)    Advanced

-----------CPU-----------    -----------GPU-----------
Value       set @ address    Value       set @ address
0x3c(60°C)  0x6a(byte106)    0x37(55°C)  0x82(byte130)
0x41(65°C)  0x6b(byte107)    0x3c(60°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)    0x41(65°C)  0x84(byte132)
0x4b(75°C)  0x6d(byte109)    0x46(70°C)  0x85(byte133)
0x50(80°C)  0x6e(byte110)    0x4b(75°C)  0x86(byte134)
0x50(80°C)  0x6f(byte111)    0x4b(75°C)  0x87(byte135)

0x0(0%)     0x72(byte114)    0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)    0x35(53%)   0x8b(byte139)
0x38(56%)   0x74(byte116)    0x3c(60%)   0x8c(byte140)
0x42(66%)   0x75(byte117)    0x44(68%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)    0x4c(76%)   0x8e(byte142)
0x56(86%)   0x77(byte119)    0x54(84%)   0x8f(byte143)
0x56(86%)   0x78(byte120)    0x56(86%)   0x90(byte144)
```

thebeerbaron2018, 2019-10-08:
Looks as though your registers are the same as mine - did you try my python 3 script on #35 ?  - run it as root  - it's set up to keep the fan off on the first 2-3 steps which allows the laptop to get warmer before requiring the fans to turn on - this saves a ton of battery over the fans being always on even if spinning at <2000rpm and near silent.  The new part I added was to also monitor the CPU use and ramp up the machine to performance mode when needed and return to powersave when idle - I am not sure if EC register 0xf2 corresponds to the "performance level" like my GE75-9SG so be careful with writing randomly to the EC - you might want to edit out the parts from the script relating to that.  The way I found them was to go into Windows and using "ec_probe monitor -c" switch back and forth between the modes and watch the values until you see which one changes with each option change on the Dragon Center.  Laborious but if it is 0xf2 then you can go ahead and leave the code as is.

StormPooper, 2019-10-08:
Sorry, meant to mention, I did give your script a go with the mode and USB lighting code taken out and it does seem to work. I might give the ec_probe a go when I have some time to experiment and give your full script a go, as it sounds like a cool feature (might be nice to build something like that into isw too).

YoyPa, 2019-10-22:
added with https://github.com/YoyPa/isw/commit/b77c310cba30b0636333943eed0c4e559239fbb6

closed 2019-10-22 by YoyPa