YoyPa/isw #38: Support for GL62MVR 7RFX

norelissme, 2019-10-10:
Hi ! Thanks a lot for making that tool. I've been looking for this for a while.

I'm on Ubuntu and I followed the your instruction to set it up. I use the profil 16J9EMS1 as there is also some GL62 in there. But it's seems that it's not working as expected because my fan is almost always on meanwhile my CPUs temperature is low and my GPU is switched off.

I drop you my dump here. Thanks in advance for having a look at it ;)

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 4b  >..............IK<
000030 03 09 01 0d 50 0a 05 00 54 0f 30 2a 88 01 e0 00  >....P...T.0*....<
000040 00 00 62 00 e0 0d 00 00 90 0d 9d 2f c3 0b 0c 30  >..b......../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2a 0a 37 3d 43 49 4f 4f  >........*.7=CIOO<
000070 64 28 28 30 38 42 4c 56 56 00 0a 08 08 08 08 08  >d((08BLVV.......<
000080 2f 00 37 3c 41 46 4b 4b 66 36 00 36 3e 46 4e 56  >/.7<AFKKf6.6>FNV<
000090 56 00 0a 07 07 07 07 07 02 16 64 02 16 66 00 00  >V.........d..f..<
0000a0 31 36 4a 42 45 4d 53 31 2e 31 30 43 30 36 30 36  >16JBEMS1.10C0606<
0000b0 32 30 31 37 31 31 3a 31 38 3a 35 35 e0 0d 00 40  >201711:18:55...@<
0000c0 00 01 25 00 00 00 00 00 00 27 00 93 00 cc 14 00  >..%......'......<
0000d0 00 00 40 00 7f 00 01 00 00 00 00 80 01 00 00 00  >..@.............<
0000e0 e2 02 00 81 10 00 00 00 00 00 02 00 00 81 00 00  >................<
0000f0 00 00 82 02 8c 00 0c 81 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x4f(79°C)  0x6f(byte111)        0x4b(75°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x30(48%)   0x73(byte115)        0x36(54%)   0x8b(byte139)
0x38(56%)   0x74(byte116)        0x3e(62%)   0x8c(byte140)
0x42(66%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)        0x4e(78%)   0x8e(byte142)
0x56(86%)   0x77(byte119)        0x56(86%)   0x8f(byte143)
0x56(86%)   0x78(byte120)        0x56(86%)   0x90(byte144)

YoyPa, 2019-10-22:
added with https://github.com/YoyPa/isw/commit/b418cab04adbc56991af8ee26792c8e31a9e3e90

> But it's seems that it's not working as expected because my fan is almost always on meanwhile my CPUs temperature is low and my GPU is switched off.

You have an update available and the changelog is:
```
New EC Firmware: 16JBEMS1.10D					
ROM CheckSum   : 3F66
Release date   : 2017/11/15	

;--------------------------- Description ------------------------------------;

1. Fix the issue of unnormal fan speed.
```
And the default CPU fan spin is 40%, you need to change it to 0% in the config file if you want it to stop.

YoyPa, 2019-10-22:
download.msi.com/archive/frm_exe/nb/16JBEMS1.10D.zip

YoyPa, 2019-11-18:
Closing old issue, open a new if problem still exist.

closed 2019-11-18 by YoyPa