YoyPa/isw #39: Make documentation more readable

mibofra, 2019-10-22:
I was trying to implement isw controls on https://github.com/jsalatas/plasma-pstate/ , to use in my plasma desktop (and in other ones if merged).

Anyway, if I can more or less read and understand everything, because I know what I'm reading, I guess is pretty difficult to understand something.

Reading the values is pretty clear, setting the coolerboost too, but for example, is there the possibility to set a particular speed on a particular fans or some fans or on all fans?

I guess the Readme can be improved.

YoyPa, 2019-10-22:
Yep, I need to update the wiki, I barely touched it in 2 years:
https://github.com/YoyPa/isw/wiki/How-EC-work-(for-GS40-6QE-at-least)
There is also a bit of explanation in the config file:
```
# cpu fan will spin at cpu_fan_speed_0:
# 		if realtime_cpu_temp is above cpu_temp_0
#			cpu_fan_speed_1 is applied
# 		if realtime_cpu_temp is above cpu_temp_1
#			cpu_fan_speed_2 is applied
#		[...]
# gpu fan will spin at gpu_fan_speed_0:
# 		if realtime_gpu_temp is above gpu_temp_0
#			gpu_fan_speed_1 is applied
# 		if realtime_gpu_temp is above gpu_temp_1
#			gpu_fan_speed_2 is applied
#		[...]
```
> Reading the values is pretty clear, setting the coolerboost too, but for example, is there the possibility to set a particular speed on a particular fans or some fans or on all fans?

I'm not aware of a clean way to do that, the EC set a fan speed depending on temperature.
But if you set the same fan speed for every temperature step I guess your goal is achieved.

YoyPa, 2019-10-24:
Is this ok ? https://github.com/YoyPa/isw/blob/master/wiki/msi%20ec.png

mibofra, 2019-10-24:
Well, I guess is surely readable. But a good doc is good when it is redable for a non technical user. If you don't want to isolate the community of your application, this question must be answerd. (Even if it's a cli app) can an avarage desktop user, who maybe can only login/logout/power on and off the pc, surf on the net and use an office suite, read and understand my doc, without issues?

66darkmatter99, 2019-10-25:
@mibofra 
> Well, I guess is surely readable. But a good doc is good when it is readable for a non technical user. If you don't want to isolate the community of your application, this question must be answerd. (Even if it's a cli app) can an avarage desktop user, who maybe can only login/logout/power on and off the pc, surf on the net and use an office suite, read and understand my doc, without issues?

With all due respect dude, its kind of up to you to become a bit more technically inclined, most people with varying degrees of experience have had no issues getting this program to work.  We should consider ourselves lucky (especially us Optimus users) that Yoypa has shared his skills and made this little pet project. Hes already mentioned that hes working on a new release taking into account everyones latest info and suggestions so lets give him some time and be easy on the criticisms. ISW has gained alot of attention lately and Im sure at some point a developer might work with him to package this as a user friendly app.

mibofra, 2019-10-25:
@66darkmatter99 just to be clear. I'm not making criticisms to @YoyPa or the project. I really like it and I'm a grateful user of the app. And that is why, as you can see, I've submitted the dump of the EC of both my MSI laptops.

I'm just poiting out that the docs is really too much technical for an average user, and we are not the only owners of MSI laptops, with the EC controlling most of the pm system (fans in primis).

That is why, as you can see, as I've added selectable online support for the https://github.com/jsalatas/plasma-pstate/pull/42 plasma-pstate plasmoid, I've want to make possible for the plasmoid to setup the fans, and the modalities https://github.com/jsalatas/plasma-pstate/issues/35 .

So I've made this issue request, just to trace the issue itself. But I don't expect the issue to be solved directly by @YoyPa or in two hours. Maybe someone will complete and rewrite the docs to be more readable.

YoyPa, 2019-10-25:
[image]

mibofra, 2019-10-25:
@YoyPa I love this image ahaha. Anyway, I simply was trying to say to @66darkmatter99 that I don't have anything against you. I love your work too :D . That's all.

66darkmatter99, 2019-10-25:
@mibofra get dude if pimento apologize, for some reason when I read this the first time the page didn't load correctly and it didn't show your guys first correspondence. Without that context the second comment just looked like a flat out criticism lol. Ya I'm the dick here.