YoyPa/isw #40: Make the install easier for non-arch users

haydenhughes, 2019-10-27:
It would be helpful if the project could use a more distro agnostic install method. Could possibly look into creating a `setup.py` and/or an install script.

Bu4ak, 2019-12-09:
Does anyone have instructions how to install and use it on ubuntu for noobs?

YoyPa, 2019-12-09:
1. Check if ec_sys module is builtin or not with rmmod:
```
$ rmmod ec_sys
rmmod: ERROR: Module ec_sys is builtin.
```
2. Configure ec_sys accordingly 
from readme:
```
It need ec_sys module with option write_support=1, there are two scenario to set that:

    ec_sys is a builtin kernel module:
        add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
        then update your grub with update-grub and reboot.
    ec_sys is not a builtin kernel module:
        copy both isw-ec_sys.conf files provided (/etc/mod[...]) with same path (Arch AUR package will do it for you).
        then reboot OR modprobe ec_sys write_support=1.
```
3. Put isw.conf in /etc
4. Run isw

silver-mx, 2019-12-29:
__NOTE: These steps were tested in __Ubuntu 19.10__ for a MSI GS65 (16Q2EMS1)__.

1. Clone the isw repository and move to the folder:

```
git clone https://github.com/YoyPa/isw
cd isw
```

2. Check if module is built-in in the kernel

```
sudo rmmod ec_sys

rmmod: ERROR: Module ec_sys is not currently loaded
```

__NOTE:__ The response above says that the module is not loaded but I assume built-in anyway. See __step 4__ to load the module.

3. To enable write support to ec_sys:
* Edit the grub configuration (/etc/default/grub):

`sudo vim /etc/default/grub`
    
* Add to GRUB_CMDLINE_LINUX __ec_sys.write_support=1__

```
GRUB_CMDLINE_LINUX="ec_sys.write_support=1"
```
* Save the file and update grub:

```
sudo update-grub
```

4. Load the ec_sys module when Ubuntu starts:

* Edit /etc/modules

```
sudo vim /etc/modules
```

* Add __ec_sys__ in a new line and save the file. This will cause the module to be loaded after reboot.

5. Copy required isw files:
 
```
# Copy isw so it is found in the system PATH
sudo cp isw /usr/bin/

# Copy the isw configuration
cp path_isw_git_repo/etc/isw.conf /etc
```

6. Reboot or run `sudo modprobe ec_sys` to load the module

7. Run isw `sudo isw -w [SECTION_NAME for your MSI model]`

__NOTE:__ The value below(__16Q2EMS1__) is for __MSI GS65 8RF__, change it according to your model.

```
sudo isw -w 16Q2EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)        0x57(87°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x37(55%)   0x8c(byte140)
0x4b(75%)   0x75(byte117)        0x3c(60%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)        0x41(65%)   0x8e(byte142)
0x4b(75%)   0x77(byte119)        0x44(68%)   0x8f(byte143)
0x4b(75%)   0x78(byte120)        0x4e(78%)   0x90(byte144)

```

7. __OPTIONAL:__ Read the fan values to check everything is working `sudo isw -r`

8. __OPTIONAL:__ To start isw as a service after reboot:

```
# Copy systemd service launcher
sudo cp path_isw_git_repo/usr/lib/systemd/system/isw@.service /etc/systemd/system

# Enable the service at startup
sudo systemctl enable isw@16Q2EMS1.service
Created symlink /etc/systemd/system/multi-user.target.wants/isw@16Q2EMS1.service → /etc/systemd/system/isw@.service.
Created symlink /etc/systemd/system/sleep.target.wants/isw@16Q2EMS1.service → /etc/systemd/system/isw@.service.

# Reboot
sudo reboot now

```

9. __OPTIONAL:__ Check the status of the isw service:

```
sudo systemctl status isw@16Q2EMS1.service
 
● isw@16Q2EMS1.service - ISW fan control service
   Loaded: loaded (/etc/systemd/system/isw@.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Sun 2019-12-29 10:10:02 CET; 20s ago
  Process: 1137 ExecStartPre=/usr/bin/sleep 2 (code=exited, status=0/SUCCESS)
  Process: 1463 ExecStart=/usr/bin/isw -w 16Q2EMS1 (code=exited, status=0/SUCCESS)
 Main PID: 1463 (code=exited, status=0/SUCCESS)

dec 29 10:10:02 diego-msi isw[1463]: [1B blob data]
dec 29 10:10:02 diego-msi isw[1463]: 0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
dec 29 10:10:02 diego-msi isw[1463]: 0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
dec 29 10:10:02 diego-msi isw[1463]: 0x3c(60%)   0x74(byte116)        0x37(55%)   0x8c(byte140)
dec 29 10:10:02 diego-msi isw[1463]: 0x4b(75%)   0x75(byte117)        0x3c(60%)   0x8d(byte141)
dec 29 10:10:02 diego-msi isw[1463]: 0x4b(75%)   0x76(byte118)        0x41(65%)   0x8e(byte142)
dec 29 10:10:02 diego-msi isw[1463]: 0x4b(75%)   0x77(byte119)        0x44(68%)   0x8f(byte143)
dec 29 10:10:02 diego-msi isw[1463]: 0x4b(75%)   0x78(byte120)        0x4e(78%)   0x90(byte144)
dec 29 10:10:02 diego-msi systemd[1]: isw@16Q2EMS1.service: Succeeded.
dec 29 10:10:02 diego-msi systemd[1]: Started ISW fan control service.

```

silver-mx, 2019-12-29:
The wiki can be found in a cacher snippet for future reference https://snippets.cacher.io/snippet/4c829db1515eaab4879b.

closed 2020-02-26 by YoyPa

JimishF, 2021-04-20:
How can I find section name for my model?
I have `GF65 Thin 9SD`

stefanjaro, 2021-12-07:
The solution @silver-mx provided is fantastic. Here are the steps I followed to get it working on my MSI GF63 9SC running Pop!_OS. It's slightly different because Pop!_OS doesn't have an `/etc/default/grub` file.

1. Clone the repository

```
git clone https://github.com/YoyPa/isw
```

2. Run all of the commands you'll find in the (PKGBUILD script)[https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=isw]

```
cd isw
sudo cp ./etc/isw.conf /etc/
sudo cp ./etc/modprobe.d/isw-ec_sys.conf etc/modprobe.d/
sudo cp ./etc/modules-load.d/isw-ec_sys.conf /etc/modules-load.d/
sudo cp ./usr/lib/systemd/system/isw@.service /usr/lib/systemd/system/
sudo cp ./usr/bin
sudo cp isw /usr/bin/
```

3. Now run the following commands and reboot

```
modprobe ec_sys write_support=1
```

4. Open the configuration file using `vim` or `gedit`, look for the section associated with your laptop (see @JimishF 's comment) and start adjusting the values to your liking (read the top of the config file for instructions).

```
sudo gedit /etc/isw.conf
```

5. Apply and enforce your changes

```
sudo isw -w [MSI_MODEL_SECTION_NAME]
```

6. To make sure this always runs at startup use

```
sudo systemctl enable isw@[MSI_MODEL_SECTION_NAME].service
```

You can check to make sure your settings are working by using: `sudo isw -r` and you can check to make sure your changes have been enforced by using `sudo isw -p [MSI_MODEL_SECTION_NAME]`

PR #65 (2020-02-26): Add debian packaging
Adds the required configuration files to be able to build `.deb` packages of `isw`. Closes #40 

## Basic how to

(assumes that `isw` is cloned and is your current directory)

```
debuild -us -uc
```

You should now have `../isw_10.1-1_amd64.deb`

#### Install package

```
sudo apt install ../isw_10.1-1_amd64.deb
```

---

I'm willing to maintain the package if required; Otherwise feel free to swap out my details in `debian/control` and `debian/changelog` for your own.