YoyPa/isw #41: Battery

ssergio-ll, 2019-11-18:
This that i go to comment is not related to this project, but I suggest if it were possible if you want and can.

MSI have a tool called "MSI Dragon". This utility have a option for battery care that set limit charging to battery. For example: loading between 40% and 60%. 

Can you do a script that can limit values min and max of ours batteries for increase its useful life.

TLP can do but i have seen that it only run with thinkpad notebook.

I have been thinking that if you can do all or some of utilities that MSI Dragon have, i could do a GUI (QT) for yours scripts and so we can have a "MSI Dragon" amateur for Linux.

YoyPa, 2019-11-18:
I had to remove the battery of my laptop (swollen). Can't help on this one :innocent:.

ssergio-ll, 2019-11-21:
> I had to remove the battery of my laptop (swollen). Can't help on this one innocent.

Hello YoyPa. What is your model Laptop? Maybe i could buy you a battery for that you can do this feature. 

Thanks.

ssergio-ll, 2019-12-02:
> I had to remove the battery of my laptop (swollen). Can't help on this one innocent.

Hello YoyPa. Are you interested in my proposal? I can buy to you a battery for that you can add this feature. 

Thanks.

YoyPa, 2019-12-04:
Hello,
That's really kind, but I installed multiple version of Dragon center and i don't see anything related to battery in it. I've also seen this statement (relevant?) "Battery Master is only for RTX and GTX16 series". My laptop is GTX970m (GS40_6QE), maybe too old for the feature.

ssergio-ll, 2019-12-04:
> Hello,
> That's really kind, but I installed multiple version of Dragon center and i don't see anything related to battery in it. I've also seen this statement (relevant?) "Battery Master is only for RTX and GTX16 series". My laptop is GTX970m (GS40_6QE), maybe too old for the feature.

Hello Yoypa.
Then If you wish, i can give you access remote to my computer for that you can add this feature. In exchange of you work, i can buy a battery for you or I can make you an economic contribution. How about?

Thanks!

YoyPa, 2019-12-04:
>  i can give you access remote to my computer

Might make it possible, but it will be a pain (reboot between linux/windows to test) and I could fail :trollface:.

ssergio-ll, 2019-12-04:
> > i can give you access remote to my computer
> 
> Might make it possible, but it will be a pain (reboot between linux/windows to test + charge discharge battery) and I could fail :trollface:.

I will be with you for give access with teamviewer. About what could fail, do you mean damage the battery? 

Do we speak privately when you can and particularize?

Thanks!

YoyPa, 2019-12-04:
> About what could fail, do you mean damage the battery?

I mean not finding anything but there is always a risk, even if I never broke anything until now. I don't think EC is able to damage the battery (battery contain it's own circuitry).

> Do we speak privately when you can and particularize?

You can find my @mail here -> https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=isw

ssergio-ll, 2019-12-17:
> I mean not finding anything but there is always a risk, even if I never broke anything until now. I don't think EC is able to damage the battery (battery contain it's own circuitry).
>
>
> You can find [...]

Sorry. I don't see your email nowhere.

garis, 2019-12-27:
> This that i go to comment is not related to this project, but I suggest if it were possible if you want and can.
>
> MSI have a tool called "MSI Dragon". This utility have a option for battery care [...]

Maybe I can help.
I was looking to get a little bit of control of the battery in Linux, long story short I have a MSI Prestige 15 exactly like TLP on my Lenovo (that was awesome). Maybe I found something but I didn't test these results yet.
The options in Dragon Center for the battery are:

1. Best for mobility: charge the battery to 100% all the time.
2. Balanced: charge the battery when under 70% stop at 80%
3. Best form battery: charge the battery when under 50% stop at 60%.

These options are directly writing in the EC at address EF with the following values:
1. E4
2. D0
3. BC

These are 3 screenshots of the 3 modes.
[image]
[image]
[image]

I was trying to explore the code of the application to look for an explanation of the values but I can't understand the logic behind the code I found (if someone is interested I can share what I found).

Like I was saying I didn't tests these results in Linux, hopefully in the next few days I will be able to do so.

YoyPa, 2019-12-27:
Nice :), here are the corresponding isw commands:
```
0xE4@0xEF (Mobility(100-100)) = sudo isw -s 0xef 228 
0xD0@0xEF (Balanced(70-80)) = sudo isw -s 0xef 208 
0xBC@0xEF (Battery(50-60)) = sudo isw -s 0xef 188
```

> Sorry. I don't see your email nowhere.

Just look at the first line: # Maintainer: yoann dot p dot public at gmail dot com

garis, 2019-12-27:
They seems to work well in Linux.
The value in 0xEF is probably the threshold that the system uses to stop charging the battery. It makes sense if the value is considered as signed (228-128=100, 208-128=80 and 188-128=60).
Probably all the value from 128 to 228 are possible (corresponding to 0% - 100% charge)
I didn't found anything regarding the lower threshold so I supposed that the charge always start at minus 10% of the threshold (and 100% is a special case).

ssergio-ll, 2020-01-03:
H

> Nice :), here are the corresponding isw commands:
>
> ```
> 0xE4@0xEF (Mobility(100-100)) = sudo isw -s 0xef 228
> 0xD0@0xEF (Balanced(70-80)) = sudo isw -s 0xef 208
> 0xBC@0xEF (Battery(50-60)) = sudo isw [...]

Hello YoyPa.

I have tested this on my notebook MSI GS75 and it works perfectly. Now if you could add alias command for this commands it would be perfect.

I go to begin to work in a graphics tools for KDE based in your work.

Also I appreciate your work and i wish do a contribution economic for you. If you wish could set a link for donations via paypal for example.

Very thanks!

YoyPa, 2020-01-05:
> (Comment that for it to work you have to restart the computer once the command has been executed)

Weird, EC should be reset after reboot :japanese_goblin:

YoyPa, 2020-01-05:
Hmmm, how should it be added ?
- A) a new option, ```isw -t 80``` = treshold at 80%
need to make a new systemd unit for auto apply at startup.

- B) add a line in every profile for battery treshold, default at 100%
set via option -w and no new systemd unit to make/enable

garis, 2020-01-05:
Regarding the reboot, in my case I was setting the battery threshold in windows via dual boot and then use Linux, the option was persistent after multiple reboots.

Regarding the implementation, I really like both implemented at the same time since it was I am currently using (sort of).
I would use the default at 80% or 60% at home with normal use and then I would use the -t 100 to charge the battery before leaving so the battery is full when I am not sure that a a power source would be near me.

ssergio-ll, 2020-01-05:
For me (MSI GS75) this options are permanent after of execute commanns and reboot so i think that is not necessary add a new systemd unit for auto apply at startup. 

0xE4@0xEF (Mobility(100-100)) = sudo isw -s 0xef 228 
0xD0@0xEF (Balanced(70-80)) = sudo isw -s 0xef 208 
0xBC@0xEF (Battery(50-60)) = sudo isw -s 0xef 188

YoyPa, 2020-01-07:
Added with https://github.com/YoyPa/isw/commit/047ca4fa64b9c0bf38fb812fe8c8487503de1c44
As @garis  suggested I added both:
- ```battery_charging_threshold``` in isw.conf is applied via option -w or via systemd unit @startup
- option -t is available to set it manually without touching isw.conf

ISW will check if the threshold is between 20% and 100% (arbitrary)
And I just noticed I wrote treshold instead of threshold everywhere ... :sob:

andreiboyanov, 2020-04-09:
@YoyPa, Do you plan to update the AUR package to include this precious feature ? :)

andreiboyanov, 2020-04-09:
I have the following issue when trying to set the threshold with the -s option (I'm with the version 1.10 of isw from AUR):

`[andrei@stealth bin]$ sudo isw -s 0xef 208

Value       set @ address
0xd0(208)   0xef(byte239)
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2407, in _get_values
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.8/argparse.py", line 2407, in <listcomp>
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 252, in ec_set
    single_write(int(Store.list_s[0],16), int(Store.list_s[1]))
  File "/usr/bin/isw", line 362, in single_write
    print(Text.TWO_COLUMN %(
OSError: [Errno 22] Invalid argument
`

Alphamineron, 2020-10-16:
@ssergio-ll Hello, I just stumbled upon this amazing project by @YoyPa... I'll have to check if this works with my MSI notebook as it doesn't seem that this has been extensively tested, however, I wanted to ask if there's any progress on the GUI concept of a linux-dragon-center...?

Looking a the help output of isw, it seems that it's implementing many things from the Dragon Center including Battery and Fan control... adding support for CPU Power Management and profiles would essentially cover most of the Dragon Center requirements.

stef204, 2022-02-05:
> @YoyPa, Do you plan to update the AUR package to include this precious feature ? :)

https://aur.archlinux.org/packages/isw-git