YoyPa/isw #44: Error reading temps or trying to flash ec

thgn, 2019-12-03:
While reading I get: isw -r  16Q4EMS1

I'm using MSI GS65

isw -cp 16Q4EMS1

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 1b 0b  >................<
000030 03 03 00 0d 30 0a 05 00 a0 14 60 3b 71 02 80 00  >....0.....`;q...<
000040 50 0a 3c 00 3b 10 af 09 a3 09 14 3f d7 0b f8 43  >P.<.;......?...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 24 00 37 40 46 4c 52 58  >........$.7@FLRX<
000070 64 2d 2d 32 3c 46 4b 50 50 00 08 03 03 03 03 03  >d--2<FKPP.......<
000080 00 00 37 3d 41 47 4d 56 5e 00 00 32 3c 46 50 5a  >..7=AGMV^..2<FPZ<
000090 5a 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >Z.........d..n..<
0000a0 31 36 51 34 45 4d 53 31 2e 31 30 39 30 35 31 33  >16Q4EMS1.1090513<
0000b0 32 30 31 39 31 30 3a 34 34 3a 32 39 3b 10 00 40  >201910:44:29;..@<
0000c0 00 01 25 00 00 9e 41 00 00 dd 00 00 00 ae 00 00  >..%...A.........<
0000d0 00 00 20 00 70 80 00 00 05 00 00 00 00 00 00 00  >.. .p...........<
0000e0 e2 02 00 05 10 00 00 00 00 00 00 00 00 9a 00 80  >................<
0000f0 00 00 80 00 0c 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

Profile dump 
Value       set @ address        Fan mode
0xc(12)     0xf4(byte244)        Auto

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)        0x56(86°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x50(80%)   0x77(byte119)        0x5a(90%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x5a(90%)   0x90(byte144)

YoyPa, 2019-12-04:
I don't see any error with reading. Can you give me the error when writing ?

thgn, 2019-12-04:
My bad with the example:

sudo isw -r 16Q4EMS1
-----------CPU-----------        -----------GPU-----------
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
│ 28°C │ 45%   2655RPM  │        │ 29°C │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘

I can't actually modify in any way the fan or even the EC controller at all.

YoyPa, 2019-12-04:
Hmmmm, I still don't get it. What are you trying to write and with which command? ISW throw an error?
You can't just say "it doesn't work", it's not enough for me to help :innocent:.

thgn, 2019-12-04:
Again my bad, sry: 
in previous message I was trying to pint that i can't see GPU fan speed.
```

sudo isw -b off

Value       set @ address
0x0(0)      0x98(byte152)
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2388, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2421, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 45, in ec_boost
    single_write(cba, cb_off)
  File "/usr/bin/isw", line 362, in single_write
    print(Text.TWO_COLUMN %(
OSError: [Errno 22] Invalid argument
```

Another example: 
```

sudo isw -w 16Q4EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 323, in ec_write
    file.seek(list_cta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2388, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2421, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 346, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument
```

YoyPa, 2019-12-04:
Can you give me output of:
```
ls -l /sys/kernel/debug/ec/ec0/io
```

thgn, 2019-12-04:
ls -l /sys/kernel/debug/ec/ec0/io
-r-------- 1 root root 0 dic  4 17:38 /sys/kernel/debug/ec/ec0/io

YoyPa, 2019-12-04:
Should be rw, ec_sys is not loaded with write_support.
Test is ec_sys is builtin, if it is you will get an error like:
````
$ rmmod ec_sys
rmmod: ERROR: Module ec_sys is builtin.
```

thgn, 2019-12-04:
> rmmod ec_sys
> rmmod: ERROR: Module ec_sys is builtin.

Added to /etc/default/grub and:

> grub-mkconfig -o /boot/grub/grub.cfg

thgn, 2019-12-04:
> rmmod ec_sys
> rmmod: ERROR: Module ec_sys is builtin.

Added to /etc/default/grub and:

> grub-mkconfig -o /boot/grub/grub.cfg

After rebooting:

> -rw------- 1 root root 0 dic  4 19:56 /sys/kernel/debug/ec/ec0/io

YoyPa, 2019-12-04:
ISW work as intended now ?

thgn, 2019-12-04:
Tried isw -b on/offf and definitly worked but reading gpu fan outputs this:

isw -r  16Q4EMS1
-----------CPU-----------        -----------GPU-----------
```
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
│ 31°C │ 45%   2685RPM  │        │ 32°C │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
```

0 rpm, is there any way to change the GPU fan to control rpms? (I have switched between MSI ec controllers version and some of them are quieter/louder so I guess this is somehow possible.

About the ec_sys totally my fault for not reading help properly... :/

YoyPa, 2019-12-04:
Look into /etc/isw.conf your section is:
```
[16Q4EMS1]
# GS65_8SE GS65_8SF GS65_8SG GS65_9SD GS65_9SE GS65_9SF GS65_9SG
# 16Q4EMS1.109
address_profile = MSI_ADDRESS_DEFAULT
fan_mode = 140
# CPU
cpu_temp_0 = 55
cpu_temp_1 = 64
cpu_temp_2 = 70
cpu_temp_3 = 76
cpu_temp_4 = 82
cpu_temp_5 = 88
cpu_fan_speed_0 = 45
cpu_fan_speed_1 = 50
cpu_fan_speed_2 = 60
cpu_fan_speed_3 = 70
cpu_fan_speed_4 = 75
cpu_fan_speed_5 = 80
cpu_fan_speed_6 = 80
# GPU
gpu_temp_0 = 55
gpu_temp_1 = 61
gpu_temp_2 = 65
gpu_temp_3 = 71
gpu_temp_4 = 77
gpu_temp_5 = 86
gpu_fan_speed_0 = 0
gpu_fan_speed_1 = 50
gpu_fan_speed_2 = 60
gpu_fan_speed_3 = 70
gpu_fan_speed_4 = 80
gpu_fan_speed_5 = 90
gpu_fan_speed_6 = 90
```

YoyPa, 2019-12-04:
So currently fan stay at 0RPM until 55°C

thgn, 2019-12-04:
mmmm, then is it possible than another fan exists? As I said switching MSI official EC versions provides different noise results.

About isw everything seems OK but my own stupidity for not reading/understanding...

YoyPa, 2019-12-04:
Three:
https://www.notebookcheck.net/fileadmin/_processed_/8/7/csm_gs65_20_774bf64ff7.jpg

YoyPa, 2019-12-04:
Ec update file provided by MSI are longer (around 4300 lines) than what we see via /sys/kernel/debug/ec/ec0/io.
You can read them with:
```
od -A x -t x1z FILENAME
```
```
isw -f FILENAME
```

thgn, 2019-12-05:
Made a diff between different ec drivers and noticed this:
```
12c12
< 0x1e(30%)   0xf80b(byte63499)    0x0(0%)     0xf82b(byte63531)
---
> 0x2d(45%)   0xf80b(byte63499)    0x0(0%)     0xf82b(byte63531)
30c30
< 0x1e(30%)   0xf84b(byte63563)    0x0(0%)     0xf86b(byte63595)
---
> 0x2d(45%)   0xf84b(byte63563)    0x0(0%)     0xf86b(byte63595)
```

It doesn't say too much to me :S Another thing is that I remembered when making some test with windows using dragon center it was not possible to control that fan.

Any chance of implementing it?

YoyPa, 2019-12-08:
Are you talking about the minimum fan speed being 30% on an update and 45% on an other ? I already looked into that (my GS40 is 50% fan speed minimum :hear_no_evil:), and I don't know how to change minimum speed, maybe by flashing a modified EC but there is a 2048bit checksum at the end and no idea how it work (it's not 16x128bit, the checksum globally change even if a single value is modified).

If it's the third fan you are talking about, I only see two profile in EC so no idea.

closed 2020-01-05 by YoyPa