YoyPa/isw #49: Realtime monitoring

wpc3001, 2019-12-11:
This may not be the right place to post this but i have been banging my head around a way to get a command applet in MATE to realtime fan speed/temp in the system tray.   what i basically need is a way to output the results of -r to a log file, one that updates every 1-3 second or so.... can anyone assist me with this?

YoyPa, 2019-12-11:
I don't know if you are at ease with modifying code but in def ec_read if you remove(line243):
+Text.CURSOR_UP+Text.CURSOR_UP
you will get an output like that instead of rewritten lines:
```
-----------CPU-----------        -----------GPU-----------
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
│ 35°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
│ 34°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
│ 35°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
│ 35°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
│ 35°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘
```

YoyPa, 2019-12-11:
You can also use dd(with caution) instead of ISW, the EC file is "/sys/kernel/debug/ec/ec0/io", and you have the Rosetta here: https://github.com/YoyPa/isw/wiki/MSI-G-laptop-EC---Rosetta

wpc3001, 2019-12-11:
maybe it would be simpler to ask if there is a way to set up -r so that it runs once, checks the speed and temp, reports it back, and then stops running. I know this may seem simple and pointless but it would really help me.

YoyPa, 2019-12-11:
Which output do you prefer:
```
${cpu_temp} ${cpu_pct} ${cpu_rpm} ${gpu_temp} ${gpu_pct} ${gpu_rpm}
```
OR
```
${cpu_temp} ${cpu_pct} ${cpu_rpm}
${gpu_temp} ${gpu_pct} ${gpu_rpm}
```
OR
```
${cpu_temp}
${cpu_pct}
${cpu_rpm}
${gpu_temp}
${gpu_pct}
${gpu_rpm}
```
Should I keep it minimal (only value) or give a tiny bit of context with symbol (% RPM °C CPU GPU) ?

wpc3001, 2019-12-11:
the first output should suffice, and i think context and symbols would be nice

YoyPa, 2019-12-11:
I will change how option -r work, now you give it a number and it will loop accordingly.
number is optional.
nothing = 0.
0 = infinite.

Is this output ok ?
```
# yo @ yo-gs in ~/Documents/git/isw on git:master x [17:22:21] 
$ sudo ./isw -r 1
-----------CPU-----------        -----------GPU-----------
│ 55°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │

# yo @ yo-gs in ~/Documents/git/isw on git:master x [17:23:38] 
$ sudo ./isw -r 2
-----------CPU-----------        -----------GPU-----------
│ 55°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
│ 55°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
```

wpc3001, 2019-12-11:
that will be perfect

YoyPa, 2019-12-11:
done :D -> https://github.com/YoyPa/isw/commit/80163ce0dcd49bf6db8c1acea38c8f797f43b710

closed 2020-01-05 by YoyPa