YoyPa/isw #52: Setting GPU fan speed while the GPU is off.

gramos93, 2019-12-17:
Is there a way to set the GPU fans to whatever the CPU fan speed is when the dedicated GPU is off (using Intel Integrated)? 

-----------GPU-----------
┌─Temp─┬─Fan Speed──────┐     
│     0°C    │    0%           0RPM     │

I know that when the GPU is off, it's temp can't be read, is there a way to bypass this?

gramos93, 2019-12-19:
<h2>GS63-6RF</h2>
<pre>EC dump 
<font color="#8AE234"><b>       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F</b></font>
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  &gt;................&lt;
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  &gt;................&lt;
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 4b 0b  &gt;..............K.&lt;
000030 02 05 00 0d 50 0a 05 00 44 16 88 2c 3a 02 c0 00  &gt;....P...D..,:...&lt;
000040 22 0b 4e 00 fe 13 5d fc 85 0f b4 2d 73 0b fa 32  &gt;&quot;.N...]....-s..2&lt;
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  &gt;................&lt;
000060 00 00 00 00 00 00 00 00 1f 00 37 3d 43 49 4f 55  &gt;..........7=CIOU&lt;
000070 64 1e 1e 2d 36 40 46 4b 55 00 08 03 03 03 03 03  &gt;d..-6@FKU.......&lt;
000080 00 00 37 3d 43 49 4f 55 66 00 00 32 3c 46 50 50  &gt;..7=CIOUf..2&lt;FPP&lt;
000090 55 00 08 03 03 03 03 03 02 0a 6e 02 0a 70 00 00  &gt;U.........n..p..&lt;
0000a0 31 36 4b 32 45 4d 53 31 2e 31 30 36 30 38 31 39  &gt;16K2EMS1.1060819&lt;
0000b0 32 30 31 36 31 33 3a 35 37 3a 32 37 f9 13 00 00  &gt;201613:57:27....&lt;
0000c0 00 07 22 00 00 9e 40 00 00 00 00 00 01 0a 14 00  &gt;..&quot;...@.........&lt;
0000d0 00 00 00 00 70 00 01 00 00 00 00 80 01 00 00 00  &gt;....p...........&lt;
0000e0 e2 02 00 8f 10 00 00 00 00 00 00 00 00 83 00 00  &gt;................&lt;
0000f0 00 00 82 00 0c 80 03 00 00 00 00 00 00 00 00 00  &gt;................&lt;
000100

Profile dump 
<font color="#8AE234"><b>Value       set @ address        Fan mode</b></font>
0xc(12)     0xf4(byte244)        Auto

-----------CPU-----------        -----------GPU-----------
<font color="#8AE234"><b>Value       set @ address        Value       set @ address</b></font>
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)        0x43(67°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x49(73°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4f(79°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x55(85°C)  0x87(byte135)

0x1e(30%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x36(54%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x40(64%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x46(70%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x4b(75%)   0x77(byte119)        0x50(80%)   0x8f(byte143)
0x55(85%)   0x78(byte120)        0x55(85%)   0x90(byte144)</pre>

YoyPa, 2020-01-05:
I don't know, the closest behavior I've seen is (on my laptop) if I configure CPU fan to stay at 0RPM over 70°C GPU fan turn ON.

gramos93, 2020-01-12:
Ok I'll try that, maybe it'll work for my laptop too. Thank you :)

closed 2024-07-04 by gramos93