YoyPa/isw #53: Can't write into EC

sid350, 2019-12-24:
```
$ sudo isw -w 16Q4EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 323, in ec_write
    file.seek(list_cta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2388, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2421, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 346, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument

```

I'm using the latest Manjaro linux, isw-git from AUR. MSI GS65-8RE

```

$ sudo ls -l /sys/kernel/debug/ec/ec0/io
-r-------- 1 root root 0 дек 23 23:46 /sys/kernel/debug/ec/ec0/io

```

```

$ sudo isw -c

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e0 06 09 0b  >................<
000030 03 09 00 0d 30 0a 05 00 a0 14 60 3b 71 02 e0 00  >....0.....`;q...<
000040 00 00 5e 00 3e 13 00 00 06 12 0f 42 86 0b 00 00  >..^.>......B....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 25 00 37 40 46 4c 52 58  >........%.7@FLRX<
000070 64 1e 1e 2d 32 3c 46 4b 50 00 08 03 03 03 03 03  >d..-2<FKP.......<
000080 00 00 37 3d 46 4c 51 55 66 00 00 32 37 3c 4b 55  >..7=FLQUf..27<KU<
000090 58 00 08 03 03 03 03 03 02 00 64 02 00 6e 00 00  >X.........d..n..<
0000a0 31 36 51 32 45 4d 53 31 2e 31 30 36 30 37 32 33  >16Q2EMS1.1060723<
0000b0 32 30 31 38 31 34 3a 33 39 3a 32 35 42 13 00 40  >201814:39:25B..@<
0000c0 00 07 23 00 00 9e c0 00 00 48 00 00 01 0d 14 00  >..#......H......<
0000d0 00 00 10 00 70 00 00 00 00 00 00 81 01 00 00 00  >....p...........<
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 b8 00 00  >................<
0000f0 00 00 80 00 0c 80 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

```

YoyPa, 2020-01-05:
$ sudo ls -l /sys/kernel/debug/ec/ec0/io
-r-------- 1 root root 0 дек 23 23:46 /sys/kernel/debug/ec/ec0/io

ec_sys does'nt have write_support, should be -rw-------

divineo, 2020-01-08:
I have the same issue.

`sudo isw -w 16JBEMS1
Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 323, in ec_write
    file.seek(list_cta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 346, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument`

`cat /sys/module/ec_sys/parameters/write_support
N`

Using the latest Manjaro with Linux 5.4.6-2.
Also, I use only the Intel GPU. And don't really have any need for cooler boost. Just need to slow down the fans since they keep running around 70% of the max. speed (That's just my guess).

YoyPa, 2020-01-08:
Have you checked the readme about ec_sys write_support ?

divineo, 2020-01-09:
> 
> 
> Have you checked the readme about ec_sys write_support ?

Yes I did. It does not work.

YoyPa, 2020-01-09:
can you show me your grub configfile ?
```
< /etc/default/grub
```

divineo, 2020-01-09:
GRUB_DEFAULT=saved
GRUB_TIMEOUT=10
GRUB_TIMEOUT_STYLE=hidden
GRUB_DISTRIBUTOR="Manjaro"
GRUB_CMDLINE_LINUX_DEFAULT="quiet apparmor=1 security=apparmor udev.log_priority=3"
GRUB_CMDLINE_LINUX="i915.modeset=1"
GRUB_DISABLE_SUBMENU=y

# If you want to enable the save default function, uncomment the following
# line, and set GRUB_DEFAULT to saved.
GRUB_SAVEDEFAULT=true

# Preload both GPT and MBR modules so that they are not missed
GRUB_PRELOAD_MODULES="part_gpt part_msdos"

# Uncomment to enable booting from LUKS encrypted devices
#GRUB_ENABLE_CRYPTODISK=y

# Uncomment to use basic console
GRUB_TERMINAL_INPUT=console

# Uncomment to disable graphical terminal
#GRUB_TERMINAL_OUTPUT=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command 'videoinfo'
GRUB_GFXMODE=auto

# Uncomment to allow the kernel use the same resolution used by grub
GRUB_GFXPAYLOAD_LINUX=keep

# Uncomment if you want GRUB to pass to the Linux kernel the old parameter
# format "root=/dev/xxx" instead of "root=/dev/disk/by-uuid/xxx"
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
GRUB_DISABLE_RECOVERY=true

# Uncomment and set to the desired menu colors.  Used by normal and wallpaper
# modes only.  Entries specified as foreground/background.
GRUB_COLOR_NORMAL="light-gray/black"
GRUB_COLOR_HIGHLIGHT="green/black"

# Uncomment one of them for the gfx desired, a image background or a gfxtheme
#GRUB_BACKGROUND="/usr/share/grub/background.png"
GRUB_THEME="/usr/share/grub/themes/manjaro/theme.txt"

# Uncomment to get a beep at GRUB start
#GRUB_INIT_TUNE="480 440 1"

divineo, 2020-01-09:
So, I was going around AUR and found this
`rmmod ec_sys write_support=1
rmmod: ERROR: Module ec_sys is builtin.`
 But, I cannot load the module with 
`insmod ec_sys write_support=1`

YoyPa, 2020-01-09:
When a module is builtin you need to use grub config file:
from the readme:
```
ec_sys is a builtin kernel module:
    add ec_sys.write_support=1 in /etc/default/grub
    then update your grub with update-grub and reboot.
```
mine look like that for example:
```
GRUB_CMDLINE_LINUX_DEFAULT="quiet acpi_osi=! acpi_osi=\"Windows 2009\" video.report_key_events=0 acpi_backlight=vendor ec_sys.write_support=1 udev.log_priority=3 audit=0 i915.enable_psr=1 intel_iommu=igfx_off 
bbswitch.load_state=0"
```

YoyPa, 2020-01-09:
I've just made a wiki page: https://github.com/YoyPa/isw/wiki/How-to-configure-ec_sys-with-write_support=1

divineo, 2020-01-10:
Thanks that did! I was editing the grub file in the wrong way. Pardon me I'm new to linux.
Everything works perfectly now. Thank you so much!!

Misioka, 2020-11-10:
@divineo  Can you please help me? 
I still have an issue with permissions, but file is `-rw------- 1 root root 0 lis 10 22:23 /sys/kernel/debug/ec/ec0/io` and
write support returns Y
[image]

[image]

shanu2698, 2021-04-24:
Heyy hi there @YoyPa  i found your work very nice but I'm not able to implement in my kali linux
here's a screenshot please help brother

[image]

Bloody-Badboy, 2021-06-27:
Disable Secure Boot in BIOS and try