YoyPa/isw #57: EC Dump for gp62mvr 6rf-242ca

clarjon1, 2020-01-10:
sudo isw -cp MSI_ADDRESS_DEFAULT 

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 4b  >..............IK<
000030 03 09 01 0d 50 0a 05 00 a0 0f 30 2a 90 01 e0 00  >....P.....0*....<
000040 00 00 5b 00 78 0e 00 00 0d 0d c3 2e c6 0b 0c 30  >..[.x..........0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2d 0a 37 3d 43 49 4f 4f  >........-.7=CIOO<
000070 64 56 28 30 38 42 4c 56 56 00 0a 08 08 08 08 08  >dV(08BLVV.......<
000080 ff 00 37 3c 41 46 4b 4b 66 36 00 36 3e 46 4e 56  >..7<AFKKf6.6>FNV<
000090 56 00 0a 07 07 07 07 07 00 16 64 02 16 66 00 00  >V.........d..f..<
0000a0 31 36 4a 42 45 4d 53 31 2e 31 30 34 30 38 33 31  >16JBEMS1.1040831<
0000b0 32 30 31 36 31 36 3a 34 35 3a 31 32 74 0e 00 40  >201616:45:12t..@<
0000c0 00 07 25 00 00 00 00 00 00 2e 00 93 00 61 00 00  >..%..........a..<
0000d0 00 00 00 00 7f 00 01 00 00 00 00 80 00 00 00 00  >................<
0000e0 e2 02 00 81 10 00 00 00 00 00 02 00 00 82 00 00  >................<
0000f0 00 00 80 02 0c 00 0c 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xc(12)     0xf4(byte244)        Auto

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x4f(79°C)  0x6f(byte111)        0x4b(75°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x30(48%)   0x73(byte115)        0x36(54%)   0x8b(byte139)
0x38(56%)   0x74(byte116)        0x3e(62%)   0x8c(byte140)
0x42(66%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)        0x4e(78%)   0x8e(byte142)
0x56(86%)   0x77(byte119)        0x56(86%)   0x8f(byte143)
0x56(86%)   0x78(byte120)        0x56(86%)   0x90(byte144)

YoyPa, 2020-01-10:
16JBEMS1 is already supported

closed 2020-01-10 by YoyPa

clarjon1, 2020-01-10:
ah, thank you. I did not find the exact motherboard model in the config by searching for it, so was unsure.