YoyPa/isw #58: Support for GV62 8RD [16JFEMS1]

NathanIkola, 2020-01-10:
[isw-cp-dump.txt](https://github.com/YoyPa/isw/files/4047909/isw-cp-dump.txt)

Here is the dump from my laptop. I'm currently seeing some odd issues with fan control, and I'm trying to quiet the thing down; at idle, the CPU temp will be sitting at about 32°C, and the fan will be stuck at 75%. I have the discrete GPU disabled in the BIOS, and it reads the temperature at 255°C, fan RPM at 55% (I thought it would read at 0°C, but that's a non-issue) -- all readings according to isw -r.

I tried to enable the profile set up for the GV62-7RD and the fans just got stuck at 86% and wouldn't budge no matter what the temperature was. Is there any way you can get this added to the configuration? Does this sound like a problem that ISW can even solve?

Thank you!

NathanIkola, 2020-01-13:
I poked around in the configuration files, and I copied the profile for the GV62_7RD and it set the fans up to max speed set in the profile. I edited my new profile to set the max speed to a lower speed, and when I used isw to write it to the EC, the fan target speed slowed down to the 10% that I set it to.

For some reason, even when I am below the minimum temperature in the profile, it always uses the maximum speed, no matter the temperature (I ran a virtual machine and got the temps up to about 70°C, and the target fan speed never changed, even though the rest of the target fan speeds were untouched).

xatusbetazx17, 2020-03-31:
maybe is the fan mode is on 140 maybe you need to changge to 180