YoyPa/isw #68: debian based distro

jack-dod, 2020-03-03:
Hello, i am trying run isw on a debian based distro but the ec_sys is not a builtin module (sys/kernel/debug/ec/ec0/io does not exist).
Can you help?
If i run isw i will give you the ec dump because i have updated bios and ec on a gl62m 7rd (ec ver. 16j9ems1.112)

hugglesfox, 2020-03-03:
Duplicate of #61 

Out of interest, what Debian based distribute you using specifically?

jack-dod, 2020-03-08:
ok i tested it on ubuntu live and ec_sys does exist on ubuntu
i have a GL62M-7RD with **updated** bios (ver E16J9IMS.324) and ec (ver **16J9EMS1.112**)
it works on coolerboost and cpu fan but not on gpu fan

here my dump:

000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 49 4b  >..............IK<
000030 03 09 00 0d 51 0a 05 00 54 0f 30 2a 88 01 e0 00  >....Q...T.0*....<
000040 00 00 60 00 5f 0e 00 00 b8 0d 45 2f 74 0b 0c 30  >..`._.....E/t..0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 27 0a 37 3d 43 49 4f 4f  >........'.7=CIOO<
000070 64 28 28 30 38 42 4c 56 56 00 0a 08 08 08 08 08  >d((08BLVV.......<
000080 22 36 37 3c 41 46 4b 4b 63 00 00 36 3e 46 4e 56  >"67<AFKKc..6>FNV<
000090 56 00 0a 07 07 07 07 07 02 16 7d 02 16 63 00 00  >V.........}..c..<
0000a0 31 36 4a 39 45 4d 53 31 2e 31 31 32 31 31 30 39  >16J9EMS1.1121109<
0000b0 32 30 31 37 31 35 3a 30 34 3a 32 38 5f 0e 00 00  >201715:04:28_...<
0000c0 00 07 25 00 00 9e c0 00 00 5b 00 00 00 ce 14 00  >..%......[......<
0000d0 00 00 10 00 73 00 01 00 00 00 00 80 01 00 00 00  >....s...........<
0000e0 e2 02 00 80 10 00 00 00 00 00 05 00 00 d1 00 05  >................<
0000f0 00 00 80 02 0c 00 0c 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
[32;1mValue       set @ address        Fan mode[0m
0xc(12)     0xf4(byte244)        Auto

[32;1mValue       set @ address        Charging below - stop @[0m
0x5(5)      0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
[32;1mValue       set @ address        Value       set @ address[0m
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x3d(61°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x43(67°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x4f(79°C)  0x6f(byte111)        0x4b(75°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x30(48%)   0x73(byte115)        0x36(54%)   0x8b(byte139)
0x38(56%)   0x74(byte116)        0x3e(62%)   0x8c(byte140)
0x42(66%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4c(76%)   0x76(byte118)        0x4e(78%)   0x8e(byte142)
0x56(86%)   0x77(byte119)        0x56(86%)   0x8f(byte143)
0x56(86%)   0x78(byte120)        0x56(86%)   0x90(byte144)

closed 2020-06-09 by jack-dod