YoyPa/isw #70: Support for MSI Prestige 15 A10SC

Fau46, 2020-03-15:
Thank a lot for this amazing tool!
Here the output of the sudo isw -cp MSI_ADDRESS_DEFAULT command:

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 20 80 e0 06 0a 0b  >.......... .....<
000030 03 03 01 0d 51 0a 05 00 a0 14 60 3b 71 02 80 00  >....Q.....`;q...<
000040 18 15 63 00 9b 14 e4 01 48 14 be 43 bc 0b f8 43  >..c.....H..C...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 30 00 32 3a 41 46 5a 5f  >........0.2:AFZ_<
000070 64 2d 2d 3a 41 48 50 55 64 00 03 05 05 05 05 03  >d--:AHPUd.......<
000080 00 00 32 3c 46 52 5a 5d 64 2d 2d 32 41 48 50 55  >..2<FRZ]d--2AHPU<
000090 64 00 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >d.........n..f..<
0000a0 31 36 53 33 45 4d 53 31 2e 31 30 33 31 31 30 37  >16S3EMS1.1031107<
0000b0 32 30 31 39 31 37 3a 30 34 3a 33 30 9b 14 00 00  >201917:04:30....<
0000c0 00 07 25 00 00 00 00 00 00 ab 00 ad 00 a0 14 00  >..%.............<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 14 03 0f  >....p...........<
0000e0 e2 02 00 03 10 00 00 00 00 00 00 00 00 79 00 e4  >.............y..<
0000f0 00 00 80 83 0c 8f 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xc(12)     0xf4(byte244)        Auto

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x2d(45%)   0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

ricardgf, 2020-04-10:
I think it is already implemented ...
See #47 and  #41

CarloLucibello, 2020-04-22:
a new release should be tagged though