YoyPa/isw #78: P15 A10SC: use gpu fan for cpu dissipation with nvidia primus/bumblebee

effeffe, 2020-04-05:
Dear all,

I have had this laptop for about a week now, and I need to say it's impressive. I am running it with Manjaro KDE and it's very fluid, but the CPU get very hot when using it.

I have modified the default profile to be more aggressive and keep it around or below 80°C, but I would liek to improove this fact. Looking at the cooling system in [1] and [2], the CPU and GPU share one pipe for cooling: while the CPU fan gets only one pipe, the GPU gets two pipe, of which one is shared with the CPU.

Now, the entire system can be easily setted to dissipate more if the GPU is under load, but in case I only use the CPU, the GPU heat is not recognised, which could cause some problems:
a) the gpu could overheat
b) the heat dissipation is not as efficient as it could be.

Would it be possible to keep the GPU temperature monitored without having to keep running a program on it (eg. glxgears)?

Thank you
Best regards

effeffe

[1]: https://www.msi.com/Content-creation/Prestige-15-A10X/Overview
[2]: https://www.notebookcheck.net/MSI-Prestige-15-A10SC-Laptop-Review-One-of-the-Best-Dell-XPS-15-Alternatives.441280.0.html#Emissions

garis, 2020-04-05:
I don't know if I understood correctly but if you have nvidia-smi you can run `nvidia-smi --query-gpu=driver_version,pstate,temperature.gpu,utilization.gpu,power.draw --format=csv` periodically.

effeffe, 2020-04-05:
Thank you for giving me that command: it works, but only when the GPU is on.

I use bumblebee for the primus optimization, which means that bbswitch manages the power supply to the GPU and turns it on and off, unloading the driver when needed.

This means that the laptop cannot see the temperature of the GPU, and consequently isw keeps the GPU fan at the lower speed possible, returning a 0°C value for it.

What I want to do is running the GPU fan at the same speed of the CPU fan when the GPU is off, i.e. the temperature of that is 0 and the nvidia driver is unloaded.

garis, 2020-04-06:
In that case I don't any idea.
Maybe lowering all the temperature reference points to 0 degree, but then you will have to change again if the GPU comes back online.

Personally, I configured my laptop with prime so the GPU is in standby basically all the time and the temp are always showing up.
Of course this is not to say that your setup is wrong :)

effeffe, 2020-04-13:
Yay, I have looked at keeping the gpu online, something I would personally avoid as it uses more power than the usual.

I am trying now to set the speed fan of the gpu as the cpu while the gpu is offline. I have forked it, but I still have to submit changes. If anyone is available trying this, it would be nice.

maybe, adding a true/false flag to the config file would be nice, this one enabling or disabling the gpu speed. This would require to see when /proc/acpi/bbswitch is not OFF (i.e. ON) without a continous loop which would overfill the memory.
Does anyone know how to do it as an invocation?

Thank you

catalyst17, 2020-05-21:
> If anyone is available trying this, it would be nice.
Heeey! Have you figured out any solutions? I'm using Pop OS, usually in a hybrid mode (the light on the power button is always amber, but it's said that in this mode gpu is consuming a really small amount of energy and, to be honest, the battery life is fine for me). However, even in this hybrid mode, when gpu is not being used (but still consuming some energy - amber light), ISW can't detect the temperature of the it. So I am also seeking for some solutions here!)

effeffe, 2020-06-17:
Hey, I still have to inspect if anythong changed with the new EC update, but here's what happens.

Basically, the gpu fan only works accordingly to the gpu temp, even if it dissipates the cpu as well.
Right now I am thinking to write a script that writes the gpu fan speed equal to the cpu one if T_gpu<T_cpu.
But, I don't know if the ec gets damaged by many scriptures... In that case I'd rather set a more aggressive profile, with a higher minimum gpu speed, which runs even when the gpu is off

effeffe, 2020-06-17:
> I don't know if I understood correctly but if you have nvidia-smi you can run `nvidia-smi --query-gpu=driver_version,pstate,temperature.gpu,utilization.gpu,power.draw --format=csv` periodically.

I have tried this, and also `intel-undervolt measure` to understand the power draw on battery with the gpu enabled. Well, it doubles, going from 12W to 25W. The Gpu draws about 12W itself, and under very intense power saver settings, I have 7W consumption on Cpu only. That would mean ~20W power draw, just to turn on the gpu and do nothing but measure the temperature...

Or we can do a double profile thing, like off on battery and on on AC, or the best solution currently remains the cooler boost...

garis, 2020-06-17:
> I have tried this, and also `intel-undervolt measure` to understand the power draw on battery with the gpu enabled. Well, it doubles, going from 12W to 25W. The Gpu draws about 12W itself, and under [...]

Maybe there is something off.
I am measuring under 2 W of GPU power in idle (orange light and the value is reported by nvidia-smi). Total is around 11-15W in idle.

effeffe, 2020-06-18:
Lol... I honestly don't know how I could reduce that power consumption. Also because on the Nvidia card nothing is running atm, unless I invoke it with bumblebee

effeffe, 2020-07-21:
So, I've tried again and realised that the power consumption I was measuring was the peak one because the invocation was turning on the GPU for an instant. Now, running `primusrun glxgears` and then running the `nvidia-smi` command I get around 3.5-3.6W from GPU and between 11 and 17W overall.

I am seriously considering a double configuration, one with bumblebee and one with nvidia-xrun, the former for mobility and the latter for production/heavy usage. Does anyone know how to do this or can help me?