YoyPa/isw #79: Support MSI Prestige 14 A10SC

kanstudio, 2020-04-16:
I compared the dump with the Prestige 15 model and found they are differ. So here is the dump for the P14 version. Thanks in advance!

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 20 80 e0 06 0b 0a  >.......... .....<
000030 03 01 00 05 51 0a 05 00 f8 11 88 2c cc 01 c0 00  >....Q......,....<
000040 f8 11 3d 00 e6 11 00 00 be 0a d0 2d fb 0b fa 32  >..=........-...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 33 00 38 3e 44 4a 50 5a  >........3.8>DJPZ<
000070 64 00 00 32 41 49 50 5a 64 50 04 03 03 03 03 03  >d..2AIPZdP......<
000080 00 00 35 3a 3f 44 50 5a 69 00 00 3c 46 50 5a 64  >..5:?DPZi..<FPZd<
000090 69 4b 03 03 03 03 05 05 06 0f 78 06 0f 78 00 00  >iK........x..x..<
0000a0 31 34 43 31 45 4d 53 31 2e 31 30 31 31 30 32 39  >14C1EMS1.1011029<
0000b0 32 30 31 39 31 36 3a 32 37 3a 34 36 e6 11 00 00  >201916:27:46....<
0000c0 00 07 25 00 00 00 00 00 00 4f 00 00 00 00 14 00  >..%......O......<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 14 04 10  >....p...........<
0000e0 e2 02 00 03 10 00 00 00 00 00 00 00 00 79 00 bc  >.............y..<
0000f0 00 00 80 81 0d 0f 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x38(56°C)  0x6a(byte106)        0x35(53°C)  0x82(byte130)
0x3e(62°C)  0x6b(byte107)        0x3a(58°C)  0x83(byte131)
0x44(68°C)  0x6c(byte108)        0x3f(63°C)  0x84(byte132)
0x4a(74°C)  0x6d(byte109)        0x44(68°C)  0x85(byte133)
0x50(80°C)  0x6e(byte110)        0x50(80°C)  0x86(byte134)
0x5a(90°C)  0x6f(byte111)        0x5a(90°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x3c(60%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x46(70%)   0x8c(byte140)
0x49(73%)   0x75(byte117)        0x50(80%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x5a(90%)   0x8e(byte142)
0x5a(90%)   0x77(byte119)        0x64(100%)  0x8f(byte143)
0x64(100%)  0x78(byte120)        0x69(105%)  0x90(byte144)
```

dmirubtsov, 2020-06-07:
looks like the same for a10rb

dmirubtsov, 2020-06-08:
@kanstudio I tested values from A10SC-15 model, and looks like all working fine with temperature values adjustments:

```
[16S3EMS1]
# P15_A10RC P15_A10SC
# 16S3EMS1.103
address_profile = MSI_ADDRESS_DEFAULT
fan_mode = 140
battery_charging_threshold = 100
# CPU
cpu_temp_0 = 56
cpu_temp_1 = 62
cpu_temp_2 = 68
cpu_temp_3 = 74
cpu_temp_4 = 80
cpu_temp_5 = 90
cpu_fan_speed_0 = 45
cpu_fan_speed_1 = 58
cpu_fan_speed_2 = 65
cpu_fan_speed_3 = 73
cpu_fan_speed_4 = 80
cpu_fan_speed_5 = 90
cpu_fan_speed_6 = 100
# GPU
gpu_temp_0 = 53
gpu_temp_1 = 58
gpu_temp_2 = 63
gpu_temp_3 = 68
gpu_temp_4 = 80
gpu_temp_5 = 90
gpu_fan_speed_0 = 45
gpu_fan_speed_1 = 50
gpu_fan_speed_2 = 70
gpu_fan_speed_3 = 80
gpu_fan_speed_4 = 90
gpu_fan_speed_5 = 100
gpu_fan_speed_6 = 105

```

dmirubtsov, 2020-06-11:
related request: https://github.com/YoyPa/isw/pull/97