YoyPa/isw #92: EC dump from GS66 10SFS

ZZZXXX0110, 2020-06-06:
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c8 06 0b 4b  >...............K<
000030 02 0d 00 05 50 0a 05 00 6a 18 60 3b 71 02 e0 00  >....P...j.`;q...<
000040 00 00 63 00 5d 18 00 fd 1c 18 39 42 a6 0b 00 00  >..c.].....9B....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 22 00 37 3c 41 46 5a 5f  >........".7<AFZ_<
000070 64 00 00 32 3c 48 50 64 64 00 08 07 05 05 05 03  >d..2<HPdd.......<
000080 00 00 32 3c 46 52 5a 5d 64 00 00 32 3c 48 50 55  >..2<FRZ]d..2<HPU<
000090 64 00 03 05 05 06 03 03 02 00 7d 02 00 64 00 00  >d.........}..d..<
0000a0 31 36 56 31 45 4d 53 31 2e 31 31 32 30 35 30 35  >16V1EMS1.1120505<
0000b0 32 30 32 30 31 34 3a 30 35 3a 34 37 5d 18 00 00  >202014:05:47]...<
0000c0 00 01 25 00 00 9e 40 00 00 02 00 00 00 00 14 00  >..%...@.........<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 14 06 05  >....p...........<
0000e0 e2 02 00 4c 10 00 00 00 00 00 00 00 00 b7 00 e4  >...L............<
0000f0 00 00 80 83 0d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0xe4(228)   0xef(byte239)        90% - 100%

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3c(60°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x64(100%)  0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

TurboGraphxBeige, 2020-06-11:
#96

liubenyuan, 2020-08-19:
My GS66 is 10SFS 097CN (10980HK, 2070 Super Max-Q), and the EC dump is a little bit different. I am updating the latest EC firmware from MSI, `16V1EMS1.115`, The EC firmware are dump using `isw -cp MSI_ADDRESS_DEFAULT`,

```

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c8 06 4b 4b  >..............KK<
000030 03 09 01 0d 30 0a 05 00 6a 18 60 3b 71 02 e0 00  >....0...j.`;q...<
000040 00 00 64 00 cf 18 00 00 cf 18 69 43 e9 0b 00 00  >..d.......iC....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 33 00 37 3c 41 46 5a 5f  >........3.7<AFZ_<
000070 64 32 2d 32 3c 48 50 55 64 00 08 07 05 05 05 03  >d2-2<HPUd.......<
000080 32 00 32 3c 46 52 5a 5d 64 32 00 32 3c 48 50 55  >2.2<FRZ]d2.2<HPU<
000090 64 00 03 05 05 06 03 03 02 00 7d 02 00 64 00 00  >d.........}..d..<
0000a0 31 36 56 31 45 4d 53 31 2e 31 31 35 30 36 31 30  >16V1EMS1.1150610<
0000b0 32 30 32 30 31 37 3a 33 38 3a 32 33 cf 18 00 00  >202017:38:23....<
0000c0 00 07 22 00 00 9e c0 00 00 c6 00 ba 00 b1 00 00  >..".............<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 14 08 13  >....p...........<
0000e0 e2 02 00 4c 10 00 00 00 00 00 00 00 00 9a 00 80  >...L............<
0000f0 00 00 80 83 0d 00 05 00 00 bd 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3c(60°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

```