YoyPa/isw #98: MSI P65-Creator-9SE

westandskif, 2020-06-14:
Hi! Thank you very much for your tool!

```
~ ❯❯❯ sudo isw -cp MSI_ADDRESS_DEFAULT

EC dump
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 b0 e4 06 4b 0b  >..............K.<
000030 03 09 00 0d 50 0a 05 00 a0 14 60 3b 71 02 e0 00  >....P.....`;q...<
000040 00 00 5d 00 3e 15 00 00 ae 13 3a 42 fc 0b 00 00  >..].>.....:B....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2c 00 37 40 46 4c 52 58  >........,.7@FLRX<
000070 64 2d 2d 32 3c 46 4b 50 50 00 08 03 03 03 03 03  >d--2<FKPP.......<
000080 2a 00 37 3d 41 47 4d 56 5e 00 00 32 3c 46 50 5a  >*.7=AGMV^..2<FPZ<
000090 5a 00 08 03 03 03 03 05 02 00 64 02 00 6e 00 00  >Z.........d..n..<
0000a0 31 36 51 34 45 4d 53 32 2e 31 30 37 30 35 31 33  >16Q4EMS2.1070513<
0000b0 32 30 31 39 31 30 3a 34 35 3a 31 36 3e 15 00 00  >201910:45:16>...<
0000c0 00 01 31 00 00 9e c1 00 00 4c 00 00 00 ae 14 00  >..1......L......<
0000d0 00 00 00 00 70 80 00 00 05 00 00 00 00 00 00 00  >....p...........<
0000e0 e2 02 00 25 10 00 00 00 00 00 00 00 00 83 00 e4  >...%............<
0000f0 00 00 80 80 0d 00 05 00 00 00 32 00 00 00 00 00  >..........2.....<
000100

Profile dump
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x37(55°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x40(64°C)  0x6b(byte107)        0x3d(61°C)  0x83(byte131)
0x46(70°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x4c(76°C)  0x6d(byte109)        0x47(71°C)  0x85(byte133)
0x52(82°C)  0x6e(byte110)        0x4d(77°C)  0x86(byte134)
0x58(88°C)  0x6f(byte111)        0x56(86°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x32(50%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x3c(60%)   0x74(byte116)        0x3c(60%)   0x8c(byte140)
0x46(70%)   0x75(byte117)        0x46(70%)   0x8d(byte141)
0x4b(75%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x50(80%)   0x77(byte119)        0x5a(90%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x5a(90%)   0x90(byte144)
```
p.s. I expected to be able to change one value at a time, but:
```
~ ❯❯❯ sudo isw -s 0x78 80                                                         ✘ 1

Value       set @ address
0x50(80)    0x78(byte120)
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2407, in _get_values
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.8/argparse.py", line 2407, in <listcomp>
    value = [self._get_value(action, v) for v in arg_strings]
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 252, in ec_set
    single_write(int(Store.list_s[0],16), int(Store.list_s[1]))
  File "/usr/bin/isw", line 362, in single_write
    print(Text.TWO_COLUMN %(
OSError: [Errno 22] Invalid argument
```

liubenyuan, 2020-08-19:
I am facing the same problem running `sudo isw -b on` on `MSI GS66 SFS 097CN`

```
~$ sudo isw -b on

Value       set @ address
0x80(128)   0x98(byte152)
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 47, in ec_boost
    single_write(cba, cb_on)
  File "/usr/bin/isw", line 362, in single_write
    print(Text.TWO_COLUMN %(
OSError: [Errno 22] Invalid argument

```

liubenyuan, 2020-08-19:
Hi, you may try https://github.com/YoyPa/isw/wiki/How-to-configure-ec_sys-with-write_support=1
and it worls.

closed 2023-03-24 by westandskif