YoyPa/isw #101: OSError: [Errno 22] Invalid argument

guocongyun, 2020-06-22:
Hi, I'm using MSI GS 65 and I have been trying to setting up isw for a few days, but it still doesn't seem to work. I am using it on manjaro with builtin ec_sys module, and the AUR version of the program. Below are the error message using AUR version. 

~ [1] $ sudo isw -w 16Q4EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 323, in ec_write
    file.seek(list_cta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 346, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument

~ [1] $ sudo isw -w COOLER_BOOST

Writing config to EC...
Traceback (most recent call last):
  File "/usr/lib/python3.8/configparser.py", line 789, in get
    value = d[option]
  File "/usr/lib/python3.8/collections/__init__.py", line 898, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.8/collections/__init__.py", line 890, in __missing__
    raise KeyError(key)
KeyError: 'fan_mode'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 292, in ec_write
    fm = int(cfgp.get(value, 'fan_mode'))
  File "/usr/lib/python3.8/configparser.py", line 792, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'fan_mode' in section: 'COOLER_BOOST'

and below is my grub config

GRUB_DEFAULT=saved
GRUB_TIMEOUT=10
GRUB_TIMEOUT_STYLE=hidden
GRUB_DISTRIBUTOR="Manjaro"
GRUB_CMDLINE_LINUX_DEFAULT="quiet udev.log_priority=3 ec_sys.write_support=1"
GRUB_CMDLINE_LINUX=""

# If you want to enable the save default function, uncomment the following
# line, and set GRUB_DEFAULT to saved.
GRUB_SAVEDEFAULT=true

# Preload both GPT and MBR modules so that they are not missed
GRUB_PRELOAD_MODULES="part_gpt part_msdos"

# Uncomment to enable booting from LUKS encrypted devices
#GRUB_ENABLE_CRYPTODISK=y

# Uncomment to use basic console
GRUB_TERMINAL_INPUT=console

# Uncomment to disable graphical terminal
#GRUB_TERMINAL_OUTPUT=console

# The resolution used on graphical terminal
# note that you can use only modes which your graphic card supports via VBE
# you can see them in real GRUB with the command 'videoinfo'
GRUB_GFXMODE=auto

# Uncomment to allow the kernel use the same resolution used by grub
GRUB_GFXPAYLOAD_LINUX=keep

# Uncomment if you want GRUB to pass to the Linux kernel the old parameter 
# format "root=/dev/xxx" instead of "root=/dev/disk/by-uuid/xxx" 
#GRUB_DISABLE_LINUX_UUID=true

# Uncomment to disable generation of recovery mode menu entries
GRUB_DISABLE_RECOVERY=true

# Uncomment and set to the desired menu colors.  Used by normal and wallpaper 
# modes only.  Entries specified as foreground/background.
GRUB_COLOR_NORMAL="light-gray/black"
GRUB_COLOR_HIGHLIGHT="green/black"

# Uncomment one of them for the gfx desired, a image background or a gfxtheme
#GRUB_BACKGROUND="/usr/share/grub/background.png"
#GRUB_THEME="/path/to/gfxtheme"

# Uncomment to get a beep at GRUB start
#GRUB_INIT_TUNE="480 440 1"

Thank you for your time

closed 2020-06-22 by guocongyun

infinito84, 2020-08-04:
@guocongyun I'm facing the same error after a fresh install how do you solve it?

infinito84, 2020-08-04:
For not reading the readme:
```
add ec_sys.write_support=1 in /etc/default/grub (Arch AUR package can't do it for you).
```