YoyPa/isw #106: I need help with my MSI Modern 14 A10M-615 (14B3EMS1.104)

JoeGold12, 2020-07-04:
Hi I own an MSI Modern 14 A10M-615 with 14B3EMS1 main board using Linux Mint 19.3 with 5.4 Kernel. I set up the ec_sys write support in grub like in the wiki. Secure boot is disabled in the bios. I'm trying to read the EC (isw -cp MSI_ADDRESS_DEFAULT) but  only getting following output:

.@Modern-14-A10M:~$ sudo isw -cp MSI_ADDRESS_DEFAULT
           
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
od: /sys/kernel/debug/ec/ec0/io:  No such file or directory

Profile dump 
Traceback (most recent call last):
  File "/usr/bin/isw", line 499, in <module>
    main()
  File "/usr/bin/isw", line 496, in main
    parser.parse_args()
  File "/usr/lib/python3.6/argparse.py", line 1743, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.6/argparse.py", line 1775, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.6/argparse.py", line 1981, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.6/argparse.py", line 1921, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.6/argparse.py", line 1833, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.6/argparse.py", line 2274, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.6/argparse.py", line 2303, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 155, in ec_profile
    with open(EC_IO_FILE,'r+b') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/debug/ec/ec0/io'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 154, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_isw.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/isw", line 499, in <module>
    main()
  File "/usr/bin/isw", line 496, in main
    parser.parse_args()
  File "/usr/lib/python3.6/argparse.py", line 1743, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.6/argparse.py", line 1775, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.6/argparse.py", line 1981, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.6/argparse.py", line 1921, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.6/argparse.py", line 1833, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.6/argparse.py", line 2274, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.6/argparse.py", line 2303, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 155, in ec_profile
    with open(EC_IO_FILE,'r+b') as file:
FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/debug/ec/ec0/io'

1. Do I have to read the EC myself? Because Issue #83 (PS42 Modern 8MO) is using the same EC (like "Model    : MS-14B3                    
MKT Name : PS42 MODERN 8MO/PS42 MODERN 8RA/Modern 14 A10RB/Modern 14 A10M" written in release notes of  EC patch 104 by MSI)

2. Can I read the profile stored on EC update file (option -f) and put the values myself into the isw.conf or do I have to post it here for further support?

jsanchez91, 2020-07-06:
I @JoeGold12,

You need to activate ec_sys write, try this command:

```
sudo modprobe ec_sys write_support=1
```

JoeGold12, 2020-07-06:
I managed to get the EC dump work. I found a solution in the NBFC-project wiki (https://github.com/hirschmann/nbfc/wiki/First-steps):

For Secure Boot users

NBFC normally uses /dev/port, which is not available with Secure Boot enabled. The current solutions to workaround this are:

    Disabling Secure Boot.
    You can do this at boot, by switching a parameter in the BIOS, depending on your computer.

    Enabling ec_sys. You can enable a kernel module which NBFC can use, by running these commands:

sudo sh -c "echo ec_sys >> /etc/modules-load.d/ec_sys.conf"
sudo sh -c "echo 'options ec_sys write_support=1' >> /etc/modprobe.d/ec_sys-write-support.conf"
sudo systemctl restart systemd-modules-load.service