YoyPa/isw #107: EC dump MSI Modern 14 A10M-615 (14B3EMS1.104)

JoeGold12, 2020-07-06:
EC dump 
             00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 10 80 c0 06 0a 0b  >................<
000030 02 0d 00 0d 51 0a 05 00 da 0c 60 3b 71 02 e0 00  >....Q.....`;q...<
000040 00 00 61 00 4e 0d ce fe cd 0c 33 42 96 0b 00 00  >..a.N.....3B....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 1d 00 32 3a 41 46 50 64  >..........2:AFPd<
000070 64 2d 2d 3a 41 48 50 64 64 00 03 05 05 05 05 05  >d--:AHPdd.......<
000080 00 00 32 37 3e 45 4c 5f 64 00 2d 32 3a 41 50 64  >..27>EL_d.-2:APd<
000090 64 00 03 05 04 04 06 03 02 00 64 02 00 64 00 00  >d.........d..d..<
0000a0 31 34 42 33 45 4d 53 31 2e 31 30 34 30 36 31 31  >14B3EMS1.1040611<
0000b0 32 30 32 30 31 32 3a 35 38 3a 33 32 4d 0d 00 00  >202012:58:32M...<
0000c0 00 01 25 00 00 9e 40 00 00 00 00 00 00 af 14 00  >..%...@.........<
0000d0 00 00 40 00 70 80 00 00 00 00 00 80 00 00 00 00  >..@.p...........<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 81 00 e4  >................<
0000f0 00 00 80 83 1d 00 0d 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0x1d(29)    0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0xe4(228)   0xef(byte239)        90% - 100%

-----------CPU-----------                    -----------GPU-----------
Value       set @ address              Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x37(55°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x3e(62°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x45(69°C)  0x85(byte133)
0x50(80°C)  0x6e(byte110)        0x4c(76°C)  0x86(byte134)
0x64(100°C) 0x6f(byte111)        0x5f(95°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x2d(45%)   0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x3a(58%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x41(65%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x64(100%)  0x77(byte119)       0x64(100%)  0x8f(byte143)
0x64(100%)  0x78(byte120)       0x64(100%)  0x90(byte144)

JoeGold12, 2020-07-15:
I tried to add my own config to the isw.conf like this, but the fan still does spin at 2700 rpm (at 36 Celsius):

sudo isw -w 14B3EMS1

[14B3EMS1]
# 14 A10M-615
# 14B3EMS1.1040611
address_profile = MSI_ADDRESS_DEFAULT
fan_mode = 140
battery_charging_threshold = 100
# CPU
cpu_temp_0 = 50
cpu_temp_1 = 58
cpu_temp_2 = 65
cpu_temp_3 = 70
cpu_temp_4 = 80
cpu_temp_5 = 100
cpu_fan_speed_0 = 0
cpu_fan_speed_1 = 58 
cpu_fan_speed_2 = 65
cpu_fan_speed_3 = 72
cpu_fan_speed_4 = 80
cpu_fan_speed_5 = 100
cpu_fan_speed_6 = 100
# GPU
gpu_temp_0 = 50
gpu_temp_1 = 55
gpu_temp_2 = 62
gpu_temp_3 = 69
gpu_temp_4 = 76
gpu_temp_5 = 95
gpu_fan_speed_0 = 0
gpu_fan_speed_1 = 50
gpu_fan_speed_2 = 58
gpu_fan_speed_3 = 65
gpu_fan_speed_4 = 80
gpu_fan_speed_5 = 100
gpu_fan_speed_6 = 100

How can I troubleshoot?

elzuzax, 2020-09-13:
Hi Joe,

I'm running a modern 15 A10M-098,  added a section for my EC, and was able to get the fans to turn off at lower temps. Did you try fiddling with the values much? My current config looks like what you see below. Note that the lowest fan speed seems to be about 2700-2800 RPM, regardless of how low I set the speed (greater than 0).

There is no gfx card in mine, and the second fan speed seems to be tied to the CPU fan speed (GPU temp always 0, but spins at same speed as CPU fan and seems to be tied to CPU temp)

[1551EMS1]
# M15_AM10-098
# 1551EMS1.103
address_profile = MSI_ADDRESS_DEFAULT
fan_mode = 140
battery_charging_threshold = 65
# CPU
cpu_temp_0 = 0
cpu_temp_1 = 43
cpu_temp_2 = 48
cpu_temp_3 = 60
cpu_temp_4 = 70
cpu_temp_5 = 95
cpu_fan_speed_0 = 0
cpu_fan_speed_1 = 0
cpu_fan_speed_2 = 20
cpu_fan_speed_3 = 50
cpu_fan_speed_4 = 70
cpu_fan_speed_5 = 90
cpu_fan_speed_6 = 100
# GPU
gpu_temp_0 = 0
gpu_temp_1 = 43
gpu_temp_2 = 48
gpu_temp_3 = 60
gpu_temp_4 = 70
gpu_temp_5 = 95
#This model has no discrete GPU and temp should always be 0
gpu_fan_speed_0 = 0
gpu_fan_speed_1 = 0
gpu_fan_speed_2 = 0
gpu_fan_speed_3 = 50
gpu_fan_speed_4 = 70
gpu_fan_speed_5 = 90

mkaddts, 2021-04-18:
Send 0xC2(194) to 0xF2 address - will STOP CPU fan spinning (in case low temperature):
$ sudo isw -s 0xF2 194

[image]

I checked this on my MSI Modern 14 A10M (14B3EMS1).

falke2, 2021-05-14:
> Hi Joe,
>
> I'm running a modern 15 A10M-098, added a section for my EC, and was able to get the fans to turn off at lower temps. Did you try fiddling with the values much? My current config looks like [...]

Hello. You've missed `gpu_fan_speed_6 = 100` in your config.
Anyway i can't do anything on the same laptop as yours, still getting `PermissionError: [Errno 1] Operation not permitted: '/sys/kernel/debug/ec/ec0/io'` on Ubuntu 21.04...

falke2, 2021-05-14:
Oh nvm i've had secure boot turned on. After turning it off all seemed to work fine! Thank you for the config @elzuzax

lolo120916, 2022-06-11:
Secure boot problems (when turned on) can be solved with acpi_ec module: https://github.com/musikid/acpi_ec
Just /sys/kernel/debug/ec/ec0/i0 with /dev/ec in isw python script (after having properly signed the acpi_ec module on your distro)
Tested on Ubuntu 22.04, secure boot on, and MSI Prestige