YoyPa/isw #118: Error "No section"

Nico-GS, 2020-08-09:
Hello,
I try to use your tool but I have a problem when I want to choose my GF65_9SD configuration

I use the command:
isw -w 16W1EMS1

But I have this error message :

```
[lorn@nico-pc isw]$ isw -cw [16W1EMS1]

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
od: /sys/kernel/debug/ec/ec0/io: Permission non accordée

Writing config to EC...
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 290, in ec_write
    ap = cfgp.get(value, 'address_profile')
  File "/usr/lib/python3.8/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.8/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: '[16W1EMS1]'

```

I must be doing it wrong, can you help me?

thank you so much

MdotO, 2020-08-09:
You dont have to use square brackets when writing the section name i.e [16W1EMS1]. simply write 'sudo isw -w 16W1EMSI'. Same applies if you plan to use it as a service. This is why the error is saying no section [16W1EMS1] because it expects 16W1EMSI as the input

Nico-GS, 2020-08-09:
I also tried without but it gives me the same error message

MdotO, 2020-08-09:
can you please give the whole trace when using **sudo**. Also did you make any changes to the etc/isw.config file at all?

Nico-GS, 2020-08-09:
Terminal : 

```
[lorn@nico-pc ~]$ sudo isw -cw 16W1EMSI
[sudo] Mot de passe de lorn : 

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 01 00 80 c0 06 48 0b  >..............H.<
000030 03 03 01 0d 50 0a 05 00 94 11 88 2c c2 01 80 00  >....P......,....<
000040 fc 08 33 00 5c 0d 94 06 b1 06 5c 2e 14 0c fa 32  >..3.\.....\....2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 5d 00 37 40 49 4c 52 58  >........].7@ILRX<
000070 64 55 26 2b 30 36 3c 46 55 00 08 03 03 03 03 03  >dU&+06<FU.......<
000080 4b 00 37 3d 43 49 4f 54 61 3c 00 2b 30 36 3c 46  >K.7=CIOTa<.+06<F<
000090 55 00 08 03 03 03 03 02 02 00 64 02 00 6e 00 00  >U.........d..n..<
0000a0 31 36 57 31 45 4d 53 31 2e 31 30 32 30 39 32 30  >16W1EMS1.1020920<
0000b0 32 30 31 39 30 39 3a 35 32 3a 35 32 5c 0d 00 40  >201909:52:52\..@<
0000c0 00 01 25 00 00 9e c0 00 01 cd 00 86 00 5e 14 00  >..%..........^..<
0000d0 00 00 01 00 7f 00 d1 00 00 00 00 00 00 00 00 00  >................<
0000e0 e2 02 00 f7 10 00 00 00 00 00 00 00 00 83 00 bc  >................<
0000f0 00 00 80 83 0d 00 03 00 00 00 00 00 00 00 00 00  >................<
000100

Writing config to EC...
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 290, in ec_write
    ap = cfgp.get(value, 'address_profile')
  File "/usr/lib/python3.8/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.8/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: '16W1EMSI'
[lorn@nico-pc ~]$ 

```

Nico-GS, 2020-08-09:
When i launch this command : 
systemctl enable isw@16W1EMS1.service

It seems to activate

But when i'm launch :  
sudo systemctl status isw@16W1EMS1.service

i have this error : 

```
[lorn@nico-pc ~]$ systemctl enable isw@16W1EMS1.service
[lorn@nico-pc ~]$ sudo systemctl status isw@16W1EMS1.service
● isw@16W1EMS1.service - ISW fan control service
     Loaded: loaded (/usr/lib/systemd/system/isw@.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2020-08-09 20:38:30 CEST; 2min 9s ago
   Main PID: 2714 (code=exited, status=1/FAILURE)

août 09 20:38:30 nico-pc isw[2714]:   File "/usr/bin/isw", line 322, in ec_write
août 09 20:38:30 nico-pc isw[2714]:     fma = int(cfgp.get(ap, 'fan_mode_address'),16)
août 09 20:38:30 nico-pc isw[2714]:   File "/usr/lib/python3.8/configparser.py", line 781, in get
août 09 20:38:30 nico-pc isw[2714]:     d = self._unify_values(section, vars)
août 09 20:38:30 nico-pc isw[2714]:   File "/usr/lib/python3.8/configparser.py", line 1149, in _unify_val>
août 09 20:38:30 nico-pc isw[2714]:     raise NoSectionError(section) from None
août 09 20:38:30 nico-pc isw[2714]: configparser.NoSectionError: No section: 'MSI_ADDRESS_DEFAULT'
août 09 20:38:30 nico-pc systemd[1]: isw@16W1EMS1.service: Main process exited, code=exited, status=1/FAI>
août 09 20:38:30 nico-pc systemd[1]: isw@16W1EMS1.service: Failed with result 'exit-code'.
août 09 20:38:30 nico-pc systemd[1]: Failed to start ISW fan control service.
lines 1-15/15 (END)

```

MdotO, 2020-08-09:
At a complete loss here but... A long shot, but can you confirm whether the **/etc/isw.conf** file exists in your systems directory. The ConfigParser doesn't give any logs if a file doesn't exist. 
If it does exist then try running this code in your python interpreter:
`
import configparser as cp
config = cp.ConfigParser()
config.read('/etc/isw.conf')
print(config.sections())
`
This is to see what sections it gives to you upon opening the file. You may also store it as a python file if you want to edit any possible file paths.

Nico-GS, 2020-08-09:
The isw.conf file is present in /etc/isw.conf and here is the result of the instructions in Python :

```
[lorn@nico-pc etc]$ python
Python 3.8.3 (default, May 17 2020, 18:15:42) 
[GCC 10.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import configparser as cp
>>> config = cp.ConfigParser()
>>> config.read('/etc/isw.conf')
['/etc/isw.conf']
>>> print(config.sections())
['16W1EMS1']

```

Nico-GS, 2020-08-09:
And now the command : isw -w 16W1EMS1 displays another message : No Section: MSI_ADRESS_DEFAULT

```
[lorn@nico-pc ~]$ isw -w 16W1EMS1

Writing config to EC...
Traceback (most recent call last):
  File "/usr/bin/isw", line 499, in <module>
    main()
  File "/usr/bin/isw", line 496, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 322, in ec_write
    fma = int(cfgp.get(ap, 'fan_mode_address'),16)
  File "/usr/lib/python3.8/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.8/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: 'MSI_ADDRESS_DEFAULT'
```

MdotO, 2020-08-09:
You have removed the MSI_ADDRESS_DEFAULT section which is used to look up addresses in the EC. You have also removed other sections like cooler boost which you may need later. Anyways, I recommend you copy the isw.conf from the source again and dont change anything from it.Your 'motherboard name' is already there so no need to do anything the isw.conf file. Then see whether the isw -w command works or not. Do this first

If it does work, then you can customize your gpu/cpu temps by altering the lines 
cpu/gpu temp/fan speed = X (**under your section i.e under [16W1EMSI] **) .

Nico-GS, 2020-08-09:
I copied the correct file to /etc/ and everything is working now!
Thank you very much for your help and your time.

Twix53791, 2022-01-26:
Hello, I got the same issue, which forbit the service to start : 

```
root@motherhome /etc $ isw -w 14A1EMSI

Writing config to EC...
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.10/argparse.py", line 1825, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.10/argparse.py", line 1858, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.10/argparse.py", line 2067, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.10/argparse.py", line 2007, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.10/argparse.py", line 1919, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.10/argparse.py", line 2450, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.10/argparse.py", line 2483, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 290, in ec_write
    ap = cfgp.get(value, 'address_profile')
  File "/usr/lib/python3.10/configparser.py", line 782, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.10/configparser.py", line 1153, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: '14A1EMSI'
```
I try here with 14A1EMSI as i know this section exist in the isw.conf file, but my MSI model is 16H7EMS1 (but I did'nt get still how to add a new config model to the isw.conf.

I reinstalled isw, I cheked all the files, I checked running isw -r as root, I cheked the python code you put before, which find the section (but, strangely, there are some others missing) : 
```
[archx@motherhome kernel]$ python
Python 3.10.2 (main, Jan 15 2022, 19:56:27) [GCC 11.1.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import configparser as cp
>>> config = cp.ConfigParser()
>>> config.read('/etc/isw.conf')
['/etc/isw.conf']
>>> print(config.sections())
['14A1EMS1', '14A3EMS1', '16F4EMS1', '16J1EMS1', '16J5EMS1', '16J9EMS1', '16JBEMS1', '16K2EMS1', '16K5EMS1', '16L1ED61', '16L2EMS1', '16P6EMS1', '16Q2EMS1', '16Q2EMS2', '16Q2EWS1', '16Q3EMS1', '16Q4EMS1', '16R1EMS1', '179CEMS1', '17A6EMS1', '17B1EMS1', '17E2EMS1', '17G1EMS1', 'COOLER_BOOST', 'USB_BACKLIGHT', 'MSI_ADDRESS_DEFAULT']
```
I checked if have have python 3 installed, python2, I installed some packages with argparser (but whitout understanding really what I was doing), but checked manually if I had all the python modules the bin/isw import installed, its ok.
The ec_sys works well, the /sys/kernel/debug/ex/ec0/io is present. So, everything seems fine, except I did'nt ad my model config from dump (as I don't know how to do it). 
Any clues?? Thanks

Twix53791, 2022-01-26:
PS : 
I realize the usr/bin/isw I got with the AUR installer (yay) was'nt the actual one of the git repo. So i manually copied all the files from the repo to my computer... but I still got the same result. I can try with all the sections, there are always not found.

Twix53791, 2022-01-26:
PPS: isw -r 14A1EMSI  works, its only isw -which don't work