YoyPa/isw #119: OSError: [Errno 22] Invalid argument when trying to write EC

RazorSiM, 2020-08-10:
Hello! Used this tool on pop_os without problems, now trying on arch.

When launching the isw -w [section] 
```isw -w 16S3EMS1```
I get this error:

```
[raz@razstige isw]$ sudo isw -w 16S3EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 323, in ec_write
    file.seek(list_cta[i])
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.8/argparse.py", line 1768, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 1800, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.8/argparse.py", line 2006, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.8/argparse.py", line 1946, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.8/argparse.py", line 1858, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.8/argparse.py", line 2389, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.8/argparse.py", line 2422, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 346, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument

```

I've updated the isw.conf with the 16S3EMS1 section (i think the aur package is not updated) and it's not working.

The ec dump:

```
[raz@razstige isw]$ sudo isw -c

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 20 80 e0 06 0a 0b  >.......... .....<
000030 03 01 01 0d 51 0a 05 00 a0 14 60 3b 71 02 c0 00  >....Q.....`;q...<
000040 18 15 31 00 f3 13 00 00 af 09 ca 3b f6 0b f8 43  >..1........;...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 32 00 32 3a 41 46 5a 5f  >........2.2:AFZ_<
000070 64 3a 2d 3a 41 48 50 55 64 00 03 05 05 05 05 03  >d:-:AHPUd.......<
000080 00 00 32 3c 46 52 5a 5d 64 2d 2d 32 41 48 50 55  >..2<FRZ]d--2AHPU<
000090 64 00 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >d.........n..f..<
0000a0 31 36 53 33 45 4d 53 31 2e 31 30 32 30 39 31 31  >16S3EMS1.1020911<
0000b0 32 30 31 39 31 30 3a 35 31 3a 35 33 f3 13 00 40  >201910:51:53...@<
0000c0 00 07 23 00 00 00 00 00 00 62 00 a5 00 9b 14 00  >..#......b......<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 03 10 00 00 00 00 00 00 00 00 79 00 b2  >.............y..<
0000f0 00 00 80 83 0d 83 05 81 00 00 00 00 00 00 00 00  >................<
000100
```

the isw -cp MSI_ADDRESS_DEFAULT :

```
[raz@razstige isw]$ sudo isw -cp MSI_ADDRESS_DEFAULT

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 20 80 e0 06 0a 0b  >.......... .....<
000030 03 01 01 0d 51 0a 05 00 a0 14 60 3b 71 02 c0 00  >....Q.....`;q...<
000040 18 15 31 00 f3 13 00 00 af 09 cc 3b f6 0b f8 43  >..1........;...C<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 32 00 32 3a 41 46 5a 5f  >........2.2:AFZ_<
000070 64 3a 2d 3a 41 48 50 55 64 00 03 05 05 05 05 03  >d:-:AHPUd.......<
000080 00 00 32 3c 46 52 5a 5d 64 2d 2d 32 41 48 50 55  >..2<FRZ]d--2AHPU<
000090 64 00 03 05 05 06 03 03 02 00 6e 02 00 66 00 00  >d.........n..f..<
0000a0 31 36 53 33 45 4d 53 31 2e 31 30 32 30 39 31 31  >16S3EMS1.1020911<
0000b0 32 30 31 39 31 30 3a 35 31 3a 35 33 f3 13 00 40  >201910:51:53...@<
0000c0 00 07 23 00 00 00 00 00 00 61 00 a5 00 9d 14 00  >..#......a......<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 03 10 00 00 00 00 00 00 00 00 79 00 b2  >.............y..<
0000f0 00 00 80 83 0d 83 05 81 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x32(50°C)  0x82(byte130)
0x3a(58°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x41(65°C)  0x6c(byte108)        0x46(70°C)  0x84(byte132)
0x46(70°C)  0x6d(byte109)        0x52(82°C)  0x85(byte133)
0x5a(90°C)  0x6e(byte110)        0x5a(90°C)  0x86(byte134)
0x5f(95°C)  0x6f(byte111)        0x5d(93°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x2d(45%)   0x8a(byte138)
0x3a(58%)   0x73(byte115)        0x32(50%)   0x8b(byte139)
0x41(65%)   0x74(byte116)        0x41(65%)   0x8c(byte140)
0x48(72%)   0x75(byte117)        0x48(72%)   0x8d(byte141)
0x50(80%)   0x76(byte118)        0x50(80%)   0x8e(byte142)
0x55(85%)   0x77(byte119)        0x55(85%)   0x8f(byte143)
0x64(100%)  0x78(byte120)        0x64(100%)  0x90(byte144)

```

Any ideas?

zlalama, 2020-08-17:
I guess that you did not run ec_sys with parameter write_support=1.

RazorSiM, 2020-08-24:
You are right! I had to setup the kernel option on refind to make it work :+1:

closed 2020-08-24 by RazorSiM

catalyst17, 2021-02-22:
> You are right! I had to setup the kernel option on refind to make it work +1

Yo! I have exactly the same problem and also use rEFInd instead of grub. How have you set up the kernel option?

RazorSiM, 2021-02-22:
> > You are right! I had to setup the kernel option on refind to make it work +1
> 
> Yo! I have exactly the same problem and also use rEFInd instead of grub. How have you set up the kernel option?

I just passed the kernel parameter:
https://wiki.archlinux.org/index.php/REFInd#Passing_kernel_parameters

catalyst17, 2021-02-22:
@RazorSiM Yes, was reading the exact same paragraph when saw your answer! Done, thank you very much for such a quick help)