YoyPa/isw #131: MSI Bravo 17 A4DDR MS-17FK

evil-god, 2020-11-17:
hi, here isw -cp MSI_ADDRESS_DEFAULT for the bravo 17 :-)

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 10 80 c0 07 0b 0b  >................<
000030 02 05 00 0d 51 0a 05 00 94 11 88 2c c2 01 c0 00  >....Q......,....<
000040 fc 08 47 00 a4 10 fa fb b7 0b 04 2d 86 0b fa 32  >..G........-...2<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 1b 00 00 00 1b 0a 34 3b 42 49 4f 55  >..........4;BIOU<
000070 6e 28 28 2d 32 37 41 46 50 00 07 04 04 04 04 04  >n((-27AFP.......<
000080 00 2d 34 3b 42 49 4f 55 6c 00 00 2d 32 37 41 46  >.-4;BIOUl..-27AF<
000090 50 00 07 04 04 04 04 04 02 16 64 02 16 64 00 00  >P.........d..d..<
0000a0 31 37 46 4b 45 4d 53 31 2e 31 30 38 30 34 32 33  >17FKEMS1.1080423<
0000b0 32 30 32 30 31 31 3a 34 33 3a 33 34 a4 10 00 40  >202011:43:34...@<
0000c0 00 06 22 00 00 00 00 00 00 00 00 00 00 c7 00 00  >..".............<
0000d0 00 00 00 00 7f a1 00 00 00 00 00 80 00 00 00 00  >................<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 a1 00 e4  >................<
0000f0 00 00 84 83 0c 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xc(12)     0xf4(byte244)        Auto

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x34(52°C)  0x6a(byte106)        0x34(52°C)  0x82(byte130)
0x3b(59°C)  0x6b(byte107)        0x3b(59°C)  0x83(byte131)
0x42(66°C)  0x6c(byte108)        0x42(66°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x49(73°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4f(79°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x55(85°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x2d(45%)   0x8b(byte139)
0x32(50%)   0x74(byte116)        0x32(50%)   0x8c(byte140)
0x37(55%)   0x75(byte117)        0x37(55%)   0x8d(byte141)
0x41(65%)   0x76(byte118)        0x41(65%)   0x8e(byte142)
0x46(70%)   0x77(byte119)        0x46(70%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x50(80%)   0x90(byte144)

Alfonsso786, 2020-12-14:
It works for MSI Bravo 15?

evil-god, 2020-12-15:
> It works for MSI Bravo 15?

it partly works on my bravo 17, 
fancontrol is perfekt working.
battery charge threshold is not working.
my guess is you get the same experience on your bravo 15.

FadaMatiik, 2021-07-23:
Hello, I don't know why, but my EC dump is different even after an EC reset, I noticed that the EC firmware version above was not the same as mine (mine is 17FKEMS1.10A) ( mine is reported as the latest firmware version according to this link after clicking on the firmware tab https://fr.msi.com/product/support_sku/59663), I hope this will provide support for my laptop, here is the return of the command isw -cp MSI_ADDRESS_DEFAULT for my bravo 17 :) 

EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 07 4b 0b  >..............K.<
000030 03 09 00 0d 51 0a 05 00 94 11 88 2c c2 01 e0 00  >....Q......,....<
000040 00 00 5b 00 22 10 00 00 88 0e 4b 30 f5 0b 00 00  >..[.".....K0....<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 26 00 00 00 26 0a 34 3b 42 49 4f 55  >....&...&.4;BIOU<
000070 6e 28 28 2d 32 37 41 46 50 00 07 04 04 04 04 04  >n((-27AFP.......<
000080 28 2d 34 3b 42 49 4f 55 6c 00 00 2d 32 37 41 46  >(-4;BIOUl..-27AF<
000090 50 00 07 04 04 04 04 04 02 16 64 02 16 64 00 00  >P.........d..d..<
0000a0 31 37 46 4b 45 4d 53 31 2e 31 30 41 30 39 31 36  >17FKEMS1.10A0916<
0000b0 32 30 32 30 31 33 3a 35 35 3a 32 32 22 10 00 40  >202013:55:22"..@<
0000c0 00 06 22 00 00 00 00 00 00 1a 00 00 00 c8 00 00  >..".............<
0000d0 00 00 00 00 7f a0 00 00 00 00 00 80 00 00 00 00  >................<
0000e0 e2 02 00 00 10 00 00 00 00 00 00 00 00 00 00 80  >................<
0000f0 00 00 80 83 0d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x34(52°C)  0x6a(byte106)        0x34(52°C)  0x82(byte130)
0x3b(59°C)  0x6b(byte107)        0x3b(59°C)  0x83(byte131)
0x42(66°C)  0x6c(byte108)        0x42(66°C)  0x84(byte132)
0x49(73°C)  0x6d(byte109)        0x49(73°C)  0x85(byte133)
0x4f(79°C)  0x6e(byte110)        0x4f(79°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x55(85°C)  0x87(byte135)

0x28(40%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x2d(45%)   0x8b(byte139)
0x32(50%)   0x74(byte116)        0x32(50%)   0x8c(byte140)
0x37(55%)   0x75(byte117)        0x37(55%)   0x8d(byte141)
0x41(65%)   0x76(byte118)        0x41(65%)   0x8e(byte142)
0x46(70%)   0x77(byte119)        0x46(70%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x50(80%)   0x90(byte144)

Aayan-ta24, 2022-03-05:
> It works for MSI Bravo 15?

No