YoyPa/isw #134: Startup service fails to run (Linux Mint 20)

ductyle, 2020-11-24:
I  followed your instructions to enable isw at startup however the systemctl service cannot start. When i checked the status of the service i got the error below.

● isw@16U4EMS1.service - ISW fan control service
     Loaded: loaded (/etc/systemd/system/isw@.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2020-11-24 13:31:59 IST; 2min 8s ago
    Process: 735 ExecStartPre=/usr/bin/sleep 2 (code=exited, status=203/EXEC)

Nov 24 13:31:59 home-PC systemd[1]: Starting ISW fan control service...
Nov 24 13:31:59 home-PC systemd[735]: isw@16U4EMS1.service: Failed to execute command: No such file or directory
Nov 24 13:31:59 home-PC systemd[735]: isw@16U4EMS1.service: Failed at step EXEC spawning /usr/bin/sleep: No such file or directory
Nov 24 13:31:59 home-PC systemd[1]: isw@16U4EMS1.service: Control process exited, code=exited, status=203/EXEC
Nov 24 13:31:59 home-PC systemd[1]: isw@16U4EMS1.service: Failed with result 'exit-code'.
Nov 24 13:31:59 home-PC systemd[1]: Failed to start ISW fan control service.
~

Do you have any solution for this?

Thankyou!

TurboGraphxBeige, 2020-11-24:
Which distro are you using?

ductyle, 2020-11-24:
> Which distro are you using?

Linux Mint 20

TurboGraphxBeige, 2020-11-24:
Here is my working service file (/etc/systemd/system/isw.service):
(normal-fans is my [section name] is /etc/isw.conf)
-------------------------------

[Unit]
Description=undervolt
After=suspend.target
After=hibernate.target
After=hybrid-sleep.target

[Service]
Type=oneshot
ExecStart=/usr/bin/isw -w normal-fans

[Install]
WantedBy=multi-user.target
WantedBy=suspend.target
WantedBy=hibernate.target
WantedBy=hybrid-sleep.target

-------------------------------

I don't remember exactly, but I think I enabled the serive with:
systemctl enable isw.service

ductyle, 2020-11-24:
> Here is my working service file (/etc/systemd/system/isw.service):
> (normal-fans is my [section name] is /etc/isw.conf)
> 
> [Unit]
> Description=undervolt
> After=suspend.target
> After=hibernate.target
> After=hybrid-sleep.target
> 
> [Service]
> Type=oneshot
> ExecStart=/usr/bin/isw -w normal-fans
> 
> [Install]
> WantedBy=multi-user.target
> WantedBy=suspend.target
> WantedBy=hibernate.target
> WantedBy=hybrid-sleep.target
> 
> I don't remember exactly, but I think I enabled the service with:
> systemctl enable isw.service

This worked, Thankyou! :)

closed 2020-11-24 by ductyle