YoyPa/isw #138: isw python errors

fawqsir, 2020-12-05:
I have a GL65_9SE and running garuda linux (arch).

000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 59 0b  >..............Y.<
000030 03 01 00 0d 51 0a 05 00 7a 12 6c 2a ea 01 c7 00  >....Q...z.l*....<
000040 22 0b 3b 00 3e 10 00 00 7d 09 0b 2c ed 0b 0c 30  >".;.>...}..,...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2f 00 32 38 3e 46 4b 50  >......../.28>FKP<
000070 64 28 00 28 30 38 42 4c 56 00 0a 07 07 09 09 09  >d(.(08BLV.......<
000080 31 00 37 3c 41 46 4b 50 61 00 00 2d 36 3e 46 4e  >1.7<AFKPa..-6>FN<
000090 4e 00 0a 07 07 07 07 07 02 16 7d 02 16 61 00 00  >N.........}..a..<
0000a0 31 36 55 35 45 4d 53 31 2e 31 30 30 30 36 32 35  >16U5EMS1.1000625<
0000b0 32 30 31 39 31 30 3a 32 34 3a 30 37 3e 10 00 00  >201910:24:07>...<
0000c0 00 07 23 00 00 9e 40 00 00 11 00 00 00 c5 14 00  >..#...@.........<
0000d0 00 00 10 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 25 10 00 00 00 00 00 02 00 00 9a 00 bc  >...%............<
0000f0 00 00 80 00 0c 80 05 80 00 00 00 00 00 00 00 00  >................<
000100

EC dump 
[32;1m       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F[0m

I get this error after install and a reboot

```
╰─λ sudo isw -w 16U5EMS1
[sudo] password for jeremy:

Writing config to EC...
Traceback (most recent call last):
  File "/usr/bin/isw", line 427, in <module>
    main()
  File "/usr/bin/isw", line 424, in main
    parser.parse_args()
  File "/usr/lib/python3.9/argparse.py", line 1818, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 1851, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 2060, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.9/argparse.py", line 2000, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.9/argparse.py", line 1912, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.9/argparse.py", line 2443, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.9/argparse.py", line 2476, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 290, in ec_write
    ap = cfgp.get(value, 'address_profile')
  File "/usr/lib/python3.9/configparser.py", line 781, in get
    d = self._unify_values(section, vars)
  File "/usr/lib/python3.9/configparser.py", line 1149, in _unify_values
    raise NoSectionError(section) from None
configparser.NoSectionError: No section: '16U5EMS1'
```

TurboGraphxBeige, 2020-12-05:
sudo sudo isw -cp 16U5EMS1

sudo sudo ??

fawqsir, 2020-12-05:
Typo my bad

fawqsir, 2020-12-05:
in the isw conf file for that profile it mentions a zip file

`# GL65_9SD GL65_9SDK GL65_9SE GL65_9SEK (using 2nd profile in 16U5EMS1.100.zip)`

fawqsir, 2020-12-05:
`╰─λ sudo isw -r
usage: isw [-h] [-b B] [-c] [-f FILE] [-p P] [-r R] [-s S S] [-u USB] [-w W]
isw: error: argument -r: expected one argument`

I thought this was supposed to work without another argument

`╰─λ sudo isw -r 16U5EMS1
-----------CPU-----------        -----------GPU-----------
┌─Temp─┬─Fan Speed──────┐        ┌─Temp─┬─Fan Speed──────┐
│ 44°C │ 0%    0RPM     │        │ 0°C  │ 0%    0RPM     │
└──────┴────────────────┘        └──────┴────────────────┘`

TurboGraphxBeige, 2020-12-05:
You don't seem to have the latest version. Did you clone the repo?(git clone https://github.com/YoyPa/isw.git)

According to the errors you have and the line numbers, we don't have the same code base.

fawqsir, 2020-12-05:
I installed with sudo pacman -Syu

fawqsir, 2020-12-05:
` sudo pacman -Qs isw
local/isw 1.10-1
    Fan control tool for MSI gaming series laptops`

fawqsir, 2020-12-05:
I think that you're right in pamac manager it shows that isw was last modified 2019-10-25.

fawqsir, 2020-12-05:
what are the chances of updating AUR?

TurboGraphxBeige, 2020-12-05:
Seems to be the latest release according to the repo here. 

Trying uninstalling from your package manager and clone the repo instead. That's what I did and everything works fine.

fawqsir, 2020-12-06:
Perhaps I didn't do this right.

I uninstalled with pamac. Then ran 
git clone https://github.com/YoyPa/isw.git

then I copied the file to the correct place and made sure their permissions were correct.

```
╰─λ sudo isw -cp 16U5EMS1

EC dump
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 06 19 0b  >................<
000030 03 01 00 05 51 0a 05 00 7a 12 6c 2a ea 01 c7 00  >....Q...z.l*....<
000040 22 0b 3b 00 41 10 00 00 78 09 08 2c e1 0b 0c 30  >".;.A...x..,...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 00 00 00 00 2b 00 32 39 40 47 4e 55  >........+.29@GNU<
000070 64 00 00 28 2f 36 3d 44 4b 00 0a 08 08 08 08 08  >d..(/6=DK.......<
000080 00 00 37 3c 41 46 4b 50 61 00 00 2d 36 3e 46 4e  >..7<AFKPa..-6>FN<
000090 4e 00 0a 07 07 07 07 07 02 16 7d 02 16 61 00 00  >N.........}..a..<
0000a0 31 36 55 35 45 4d 53 31 2e 31 30 30 30 36 32 35  >16U5EMS1.1000625<
0000b0 32 30 31 39 31 30 3a 32 34 3a 30 37 3e 10 00 00  >201910:24:07>...<
0000c0 00 02 23 00 00 9e 40 00 00 0d 00 00 00 00 14 00  >..#...@.........<
0000d0 00 00 00 00 70 00 00 00 00 00 00 80 00 00 00 00  >....p...........<
0000e0 e2 02 00 25 10 00 00 00 00 00 02 00 00 9a 00 bc  >...%............<
0000f0 00 00 80 00 0c 80 05 80 00 00 00 00 00 00 00 00  >................<
000100

Profile dump
Value       set @ address        Fan mode
0xc(12)     0xf4(byte244)        Auto

Value       set @ address        Charging below - stop @
0xbc(188)   0xef(byte239)        50% - 60%

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x39(57°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x40(64°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x47(71°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4e(78°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x50(80°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x28(40%)   0x73(byte115)        0x2d(45%)   0x8b(byte139)
0x2f(47%)   0x74(byte116)        0x36(54%)   0x8c(byte140)
0x36(54%)   0x75(byte117)        0x3e(62%)   0x8d(byte141)
0x3d(61%)   0x76(byte118)        0x46(70%)   0x8e(byte142)
0x44(68%)   0x77(byte119)        0x4e(78%)   0x8f(byte143)
0x4b(75%)   0x78(byte120)        0x4e(78%)   0x90(byte144)
```
```
╰─λ sudo isw -w 16U5EMS1

Writing config to EC...
Value       set @ address        Fan mode
0x8c(140)   0xf4(byte244)        Advanced

Traceback (most recent call last):
  File "/usr/bin/isw", line 356, in ec_write
    file.seek(bcta)
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/isw", line 499, in <module>
    main()
  File "/usr/bin/isw", line 496, in main
    parser.parse_args()
  File "/usr/lib/python3.9/argparse.py", line 1818, in parse_args
    args, argv = self.parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 1851, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 2060, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.9/argparse.py", line 2000, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.9/argparse.py", line 1912, in take_action
    argument_values = self._get_values(action, argument_strings)
  File "/usr/lib/python3.9/argparse.py", line 2443, in _get_values
    value = self._get_value(action, arg_string)
  File "/usr/lib/python3.9/argparse.py", line 2476, in _get_value
    result = type_func(arg_string)
  File "/usr/bin/isw", line 391, in ec_write
    print(Text.FOUR_COLUMN %(
OSError: [Errno 22] Invalid argument
```

TurboGraphxBeige, 2020-12-06:
Well, I'm out of ideas!

fawqsir, 2020-12-06:
@YoyPa do you have any ideas? I am willing to help further.

fawqsir, 2020-12-15:
I switched kernels to linux-tkg-pds and still getting the same errors.

VVlovsky, 2020-12-15:
Hey, i had the same problem on manjaro. Fixed it with a small script changing the symbol 'N" to 'Y' in /sys/module/ec_sys/parameters/write_support.

fawqsir, 2020-12-28:
`sudo nano /sys/module/ec_sys/parameters/write_support
and change N to Y and then sudo isw -w YOURMODEL`

This is the answer! Thanks so much

fawqsir, 2020-12-29:
It would be nice to add it to the isw@service.

I tried `sudo sed 's/N/Y/g' /sys/module/ec_sys/parameters/write_support` but that didn't work for me.