YoyPa/isw #144: MSI Alpha 15 AMD (A4DEK-005)

samuelmattjohnston, 2020-12-22:
`$ sudo isw -cp MSI_ADDRESS_DEFAULT`

Plugged in:

```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 07 0b 0b  >................<
000030 03 03 00 0d 51 0a 05 00 c0 17 30 2a 60 02 87 00  >....Q.....0*`...<
000040 16 0d 5c 00 55 17 41 03 50 15 70 30 d4 0b a2 30  >..\.U.A.P.p0...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 26 00 00 00 26 00 32 39 40 47 4e 55  >....&...&.29@GNU<
000070 6e 2d 2d 2d 30 38 40 48 50 00 0a 08 08 08 08 08  >n---08@HP.......<
000080 00 00 37 3c 41 46 4b 50 6c 00 00 2e 36 3e 46 50  >..7<AFKPl...6>FP<
000090 5a 00 0a 07 07 07 07 07 02 16 7d 02 16 66 00 00  >Z.........}..f..<
0000a0 31 36 55 4b 45 4d 53 31 2e 31 30 31 30 36 31 31  >16UKEMS1.1010611<
0000b0 32 30 32 30 31 31 3a 31 38 3a 34 37 55 17 00 40  >202011:18:47U..@<
0000c0 00 07 23 00 00 00 00 00 00 1d 00 00 00 b0 00 00  >..#.............<
0000d0 00 00 00 00 7f a0 5a 00 20 1f 20 80 00 00 00 00  >......Z. . .....<
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 a6 00 80  >................<
0000f0 00 00 80 81 0d 00 02 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0x80(128)   0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x39(57°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x40(64°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x47(71°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4e(78°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x50(80°C)  0x87(byte135)

0x2d(45%)   0x72(byte114)        0x0(0%)     0x8a(byte138)
0x2d(45%)   0x73(byte115)        0x2e(46%)   0x8b(byte139)
0x30(48%)   0x74(byte116)        0x36(54%)   0x8c(byte140)
0x38(56%)   0x75(byte117)        0x3e(62%)   0x8d(byte141)
0x40(64%)   0x76(byte118)        0x46(70%)   0x8e(byte142)
0x48(72%)   0x77(byte119)        0x50(80%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x5a(90%)   0x90(byte144)
```

unplugged:
```
EC dump 
       00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
000000 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000020 00 00 00 00 00 00 00 00 00 00 00 80 c0 07 0b 0b  >................<
000030 02 05 00 05 51 0a 05 00 c0 17 30 2a 60 02 c7 00  >....Q.....0*`...<
000040 16 0d 5d 00 51 17 9f fd 76 15 fe 2f ce 0b a2 30  >..].Q...v../...0<
000050 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  >................<
000060 00 00 00 00 1f 00 00 00 1f 00 32 39 40 47 4e 55  >..........29@GNU<
000070 6e 00 00 28 30 38 40 48 50 00 0a 08 08 08 08 08  >n..(08@HP.......<
000080 00 00 37 3c 41 46 4b 50 6c 00 00 2e 36 3e 46 50  >..7<AFKPl...6>FP<
000090 5a 00 0a 07 07 07 07 07 02 16 7d 02 16 66 00 00  >Z.........}..f..<
0000a0 31 36 55 4b 45 4d 53 31 2e 31 30 31 30 36 31 31  >16UKEMS1.1010611<
0000b0 32 30 32 30 31 31 3a 31 38 3a 34 37 55 17 00 40  >202011:18:47U..@<
0000c0 00 07 23 00 00 00 00 00 00 01 00 00 00 00 00 00  >..#.............<
0000d0 00 00 00 00 7f a1 5a 00 1e 1e 1e 80 00 00 00 00  >......Z.........<
0000e0 e2 02 00 01 10 00 00 00 00 00 00 00 00 a1 00 80  >................<
0000f0 00 00 80 81 0d 00 05 00 00 00 00 00 00 00 00 00  >................<
000100

Profile dump 
Value       set @ address        Fan mode
0xd(13)     0xf4(byte244)        Unknown

Value       set @ address        Charging below - stop @
0x80(128)   0xef(byte239)        Nothing is set

-----------CPU-----------        -----------GPU-----------
Value       set @ address        Value       set @ address
0x32(50°C)  0x6a(byte106)        0x37(55°C)  0x82(byte130)
0x39(57°C)  0x6b(byte107)        0x3c(60°C)  0x83(byte131)
0x40(64°C)  0x6c(byte108)        0x41(65°C)  0x84(byte132)
0x47(71°C)  0x6d(byte109)        0x46(70°C)  0x85(byte133)
0x4e(78°C)  0x6e(byte110)        0x4b(75°C)  0x86(byte134)
0x55(85°C)  0x6f(byte111)        0x50(80°C)  0x87(byte135)

0x0(0%)     0x72(byte114)        0x0(0%)     0x8a(byte138)
0x28(40%)   0x73(byte115)        0x2e(46%)   0x8b(byte139)
0x30(48%)   0x74(byte116)        0x36(54%)   0x8c(byte140)
0x38(56%)   0x75(byte117)        0x3e(62%)   0x8d(byte141)
0x40(64%)   0x76(byte118)        0x46(70%)   0x8e(byte142)
0x48(72%)   0x77(byte119)        0x50(80%)   0x8f(byte143)
0x50(80%)   0x78(byte120)        0x5a(90%)   0x90(byte144)
```

samuelmattjohnston, 2020-12-29:
Talked with MSI -- They made their firmware behave this way. there is no fix for it possible supposedly.

Sonickyle27, 2021-06-24:
Dragon Center on Windows comes with a "Silent" profile that turns the fans off even when plugged in. It may be worth seeing what that is changing to make the fan stop running.

samuelmattjohnston, 2021-06-27:
Maybe that is a new update? 
Can you supply version of software, firmware, and the bios? (I checked and I don't see new firmware/bios listed)

I've tried silent mode in the past on windows and did not see that behavior you described. Any chance you can get the EC values  dumped on your machine?

Sonickyle27, 2021-07-10:
Unfortunately I just checked again and I think I was misremembering. When selecting silent mode the fan stays at 2700 rpm :(

marshevms, 2021-08-23:
Hi can you download rweverything on windows and create screenshot of EC? And do it for all fan mode.

Sonickyle27, 2021-09-01:
@marshevms Would you be able to give some pointers as to how to actually use that software please to get what you need?

Sonickyle27, 2021-09-16:
> Hi can you download rweverything on windows and create screenshot of EC? And do it for all fan mode.

Here is a zip file containing .bin dumps for each of the modes. I saw some of the values changing several seconds after setting the modes in Dragon Center, so I waited about 10 seconds after changing the mode before creating each of these.

[a4dek-bins.zip](https://github.com/YoyPa/isw/files/7180342/a4dek-bins.zip)

marshevms, 2021-09-20:
You can try https://github.com/marshevms/mlfc/releases/tag/0.5.1 If you have debian based distr