YoyPa/isw #153: Nothing happen when i write into EC

dirompante, 2021-01-31:
I've installed isw correctely, if I try:
`sudo isw -b on`
actually  it enable CoolerBoost.
I've also checked with:
```
$ sudo ls -l /sys/kernel/debug/ec/ec0/io
-rw------- 1 root root 0 31 gen 00.18 /sys/kernel/debug/ec/ec0/io
```

If I write EC with:
`sudo isw -w MY_SECTION`
nothing change also if I've set `cpu_fan_speed_0 = 100` and previously was a low percentage.

If I change only a single value into EC with:
`sudo isw -s address_of_cpu_fan_speed_0 100`
as the previous case, I can't notice differences.

Laptop: MSI modern 15 A11M-217XIT (1552EMS1)
OS: Manjaro Nibia 20.2.1

ShiweiJin, 2021-05-22:
Have you sovled this problem? I truned off the secure boot but still cannot modify the config file.

dirompante, 2021-12-24:
I'm sorry for the late answer.
Yes, after some attempts I figured out how to fix this problem.
I have installed Windows and changed the fan profiles with the official MSI tool. Looking at the EC values was not difficult to check how MSI tool works. When a personalized profile is set, you can see that the value  `0xd4` switch to `64`. that's the only difference from [MSI G laptop EC Rosetta
](https://github.com/YoyPa/isw/wiki/MSI-G-laptop-EC---Rosetta) I was able to notice.
After some attempts, I convinced myself that it is enough to set the value of `0xd4` to make it works.
Now I have added to autostart the following lines:
```
isw -w 1552EMS1
isw -s 0xd4 64 
```
Where 1552EMS1 is the profile [here](https://github.com/YoyPa/isw/issues/152).
Now everything seems to work properly even if I have updated the BIOS a couple of times (since I have found this trick).
Tell me if this works properly also for you (please specify your BIOS update).